markniu/Bed_Distance_sensor #5: Great idea but instructions are unclear

Zchulski, 2023-02-22:
I bought one of these sensors but the directions I think are unclear enough for non linux based user to install klipper without kiauh after about 3 hours of dinking with the klipper install i'm about to give up.  Great idea but unless this get put into the main fork of klipper it is a losing battle i think

DarKingF0x, 2023-02-22:
The instructions are right there in the Klipper install wiki in this repository.
https://github.com/markniu/Bed_Distance_sensor/wiki/Installing-for-Klipper

Zchulski, 2023-02-22:
ohh trust i followed them to the letter and am still running to issues

Zchulski, 2023-02-22:
installed everything according to the directions and still have nothing working correctly mainsail talks to moonraker but moonraker and klipper are not working together

Zchulski, 2023-02-22:
[image]

this is what i have

closed 2023-02-22 by Zchulski