markniu/Bed_Distance_sensor #6: unable to flash new firmware to SKR3

Zchulski, 2023-02-22:
not sure if this is the the fault of mark here but I am unable to flash the bdsensor firmware to an SKR3.  i keep getting the following error

`MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: 256a0062-dirty
MCU(s) which should be updated:
mcu: Current version 256a0062-dirty-20230222_064949-mainsail
Up-to-date MCU(s):
<none>

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'mcu': Unknown command: I2C_BD_send`

I have tried putting the firmware on multiple different drives and none of them seem to want to accept the flash

markniu, 2023-02-22:
seems that the version of the MCU firmware and the klippy are different. make sure the /home/pi/klipper is download from https://github.com/markniu/klipper-for-pull.git not https://github.com/Klipper3d/klipper

Zchulski, 2023-02-22:
That's where I got it from

-------- Original Message --------

markniu, 2023-02-22:
please run command `grep "I2C_BD_send" -r *`in the /home/pi/klipper ,  there should be some I2C_BD_send information out there like this 
[image]

Zchulski, 2023-02-22:
Will try tomm I swapped out to my working os so I could print something overnight will update tomm afternoon

-------- Original Message --------

markniu, 2023-02-22:
message in the log `mcu 'mcu': Unknown command: I2C_BD_send` means there is no command code` I2C_BD_send` in the python code of pi or firmware in the mcu.

markniu, 2023-02-22:
I think I find the reason of this problem: the klipper will not compile the BDsensor.c into the fimware after we select the STM32H7 and STM32M0 in the `make menuconfig`.

Zchulski, 2023-02-22:
Do I have to reinstall klipper to get these commits loaded?

markniu, 2023-02-23:
> Do I have to reinstall klipper to get these commits loaded?

right, just delete the klipper folder and git clone again.

Zchulski, 2023-02-23:
still no good still getting an error whatever you did was not the fix

```pi@mainsail:~/klipper $ grep I2C_BD_send -r
out/compile_time_request.c:extern void command_I2C_BD_send(uint32_t*);
out/compile_time_request.c:    // I2C_BD_send oid=%c data=%*s
out/compile_time_request.c:    .func=command_I2C_BD_send
grep: out/compile_time_request.o: binary file matches
out/klipper.dict:{"build_versions":"gcc: (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027] binutils: (2.35.2-2+14+b2) 2.35.2","commands":{"I2C_BD_receive oid=%c data=%*s":54,"I2C_BD_receive2 oid=%c data=%*s":53,"I2C_BD_send oid=%c data=%*s":52,"allocate_oids count=%c":8,"buttons_ack oid=%c count=%c":61,"buttons_add oid=%c pos=%c pin=%u pull_up=%c":63,"buttons_query oid=%c clock=%u rest_ticks=%u retransmit_count=%c invert=%c":62,"clear_shutdown":2,"config_I2C_BD oid=%c sda_pin=%u scl_pin=%u delay=%u":51,"config_adxl345 oid=%c spi_oid=%c":47,"config_analog_in oid=%c pin=%u":32,"config_buttons oid=%c button_count=%c":64,"config_counter oid=%c pin=%u pull_up=%c":71,"config_digital_out oid=%c pin=%u value=%c default_value=%c max_duration=%u":17,"config_endstop oid=%c pin=%c pull_up=%c":26,"config_hd44780 oid=%c rs_pin=%u e_pin=%u d4_pin=%u d5_pin=%u d6_pin=%u d7_pin=%u delay_ticks=%u":60,"config_neopixel oid=%c pin=%u data_size=%hu bit_max_ticks=%u reset_min_ticks=%u":69,"config_pwm_out oid=%c pin=%u cycle_ticks=%u value=%hu default_value=%hu max_duration=%u":43,"config_spi oid=%c pin=%u cs_active_high=%c":38,"config_spi_angle oid=%c spi_oid=%c spi_angle_type=%c":50,"config_spi_shutdown oid=%c spi_oid=%c shutdown_msg=%*s":33,"config_spi_without_cs oid=%c":37,"config_st7920 oid=%c cs_pin=%u sclk_pin=%u sid_pin=%u sync_delay_ticks=%u cmd_delay_ticks=%u":57,"config_stepper oid=%c step_pin=%c dir_pin=%c invert_step=%c step_pulse_ticks=%u":23,"config_thermocouple oid=%c spi_oid=%c thermocouple_type=%c":40,"config_tmcuart oid=%c rx_pin=%u pull_up=%c tx_pin=%u bit_time=%u":66,"config_trsync oid=%c":30,"debug_nop":9,"debug_ping data=%*s":10,"debug_read order=%c addr=%u":12,"debug_write order=%c addr=%u val=%u":11,"emergency_stop":3,"endstop_home oid=%c clock=%u sample_ticks=%u sample_count=%c rest_ticks=%u pin_value=%c trsync_oid=%c trigger_reason=%c":25,"endstop_query_state oid=%c":24,"finalize_config crc=%u":6,"get_clock":5,"get_config":7,"get_uptime":4,"hd44780_send_cmds oid=%c cmds=%*s":59,"hd44780_send_data oid=%c data=%*s":58,"identify offset=%u count=%c":1,"neopixel_send oid=%c":67,"neopixel_update oid=%c pos=%hu data=%*s":68,"query_adxl345 oid=%c clock=%u rest_ticks=%u":46,"query_adxl345_status oid=%c":45,"query_analog_in oid=%c clock=%u sample_ticks=%u sample_count=%c rest_ticks=%u min_value=%hu max_value=%hu range_check_count=%c":31,"query_counter oid=%c clock=%u poll_ticks=%u sample_ticks=%u":70,"query_spi_angle oid=%c clock=%u rest_ticks=%u time_shift=%c":49,"query_thermocouple oid=%c clock=%u rest_ticks=%u min_value=%u max_value=%u max_invalid_count=%c":39,"queue_digital_out oid=%c clock=%u on_ticks=%u":15,"queue_pwm_out oid=%c clock=%u value=%hu":42,"queue_step oid=%c interval=%u count=%hu add=%hi":22,"reset":72,"reset_step_clock oid=%c clock=%u":20,"set_digital_out pin=%u value=%c":13,"set_digital_out_pwm_cycle oid=%c cycle_ticks=%u":16,"set_next_step_dir oid=%c dir=%c":21,"set_pwm_out pin=%u cycle_ticks=%u value=%hu":41,"spi_angle_transfer oid=%c data=%*s":48,"spi_send oid=%c data=%*s":34,"spi_set_bus oid=%c spi_bus=%u mode=%u rate=%u":36,"spi_set_software_bus oid=%c miso_pin=%u mosi_pin=%u sclk_pin=%u mode=%u rate=%u":44,"spi_transfer oid=%c data=%*s":35,"st7920_send_cmds oid=%c cmds=%*s":56,"st7920_send_data oid=%c data=%*s":55,"stepper_get_position oid=%c":19,"stepper_stop_on_trigger oid=%c trsync_oid=%c":18,"tmcuart_send oid=%c write=%*s read=%c":65,"trsync_set_timeout oid=%c clock=%u":28,"trsync_start oid=%c report_clock=%u report_ticks=%u expire_reason=%c":29,"trsync_trigger oid=%c reason=%c":27,"update_digital_out oid=%c value=%c":14},"config":{"ADC_MAX":4095,"BUS_PINS_spi1":"PA6,PA7,PA5","BUS_PINS_spi1a":"PB4,PB5,PB3","BUS_PINS_spi2":"PB14,PB15,PB13","BUS_PINS_spi2a":"PC2,PC3,PB10","BUS_PINS_spi2b":"PI2,PI3,PI1","BUS_PINS_spi3a":"PC11,PC12,PC10","BUS_PINS_spi4":"PE13,PE14,PE12","BUS_PINS_spi5":"PF8,PF9,PF7","BUS_PINS_spi5a":"PH7,PF11,PH6","BUS_PINS_spi6":"PG12,PG14,PG13","CLOCK_FREQ":400000000,"MCU":"stm32h743xx","PWM_MAX":255,"RESERVE_PINS_USB":"PA11,PA12","RESERVE_PINS_crystal":"PH0,PH1","STATS_SUMSQ_BASE":256,"STEPPER_BOTH_EDGE":1},"enumerations":{"pin":{"ADC_TEMPERATURE":254,"PA0":[0,16],"PB0":[16,16],"PC0":[32,16],"PD0":[48,16],"PE0":[64,16],"PF0":[80,16],"PG0":[96,16],"PH0":[112,16],"PI0":[128,16]},"spi_angle_type":{"a1333":0,"as5047d":1,"tle5012b":2},"spi_bus":{"spi1":1,"spi1a":2,"spi2":0,"spi2a":3,"spi2b":6,"spi3a":4,"spi4":5,"spi5":7,"spi5a":8,"spi6":9},"static_string_id":{"ADC out of range":26,"Already finalized":13,"Can not set soft pwm cycle ticks while updates pending":20,"Can't add signal that is already active":25,"Can't assign oid":11,"Can't reset time when stepper active":22,"Command parser error":7,"Command request":8,"Invalid PWM channel":45,"Invalid buttons retransmit count":34,"Invalid command":5,"Invalid count parameter":23,"Invalid move request size":14,"Invalid neopixel data_size":39,"Invalid neopixel update command":38,"Invalid oid type":12,"Invalid spi bus":44,"Invalid spi config":27,"Invalid spi_angle chip type":33,"Invalid thermocouple chip type":29,"Max of 8 buttons":36,"Messag [... cut ...]
klippy/extras/probe.py:            self.I2C_BD_send_cmd3 = self.mcu.lookup_command(
klippy/extras/probe.py:                "I2C_BD_send oid=%c data=%*s", cq=self.cmd_queue)
klippy/extras/probe.py:        if probe.I2C_BD_send_cmd3 is not None:
klippy/extras/probe.py:            probe.I2C_BD_send_cmd3.send([probe.oid, "1022"])
klippy/extras/probe.py:        if probe.I2C_BD_send_cmd3 is not None:
klippy/extras/probe.py:            probe.I2C_BD_send_cmd3.send([probe.oid, "1018"])
pi@mainsail:~/klipper $
```

Zchulski, 2023-02-23:
Ok I finally got the sensor to communicate correctly but I think there is some kind of problem.  The raw data coming from the sensor isn't matching what i'm seeing in real life I can touch the sensor to the bed and i'm still getting 511 on the raw data.  The numbers im mainly seeing is somewhere between 900 and 500 and it doesnt matter what the distance the sensor is from the bed I still get the same numbers which doesn't seem right.  And I cant move the z axis without it and i cant calibrate it without having the sensor work.  so i might be between a rock and a hard spot with this sensor.  are these values im seeing supposed to be correct

markniu, 2023-02-24:
how about the value of distance after sending `M102   S-2`  ? 
If the raw calibration data after` M102 M-5 `are continue increased within 0 - 1000 that means the sensor works, but all that raw data were from the last calibration before the sensor was shipped out.  
you can move down the nozzle close to the bed by moving the z motor with hand before sending `M102 M-5` while printer is power off.

Zchulski, 2023-02-24:
What I'm failing to understand here is what exactly is the difference between s2 and s5? It doesn't matter if the sensor Is 1mm from the bed or touching the bed I still get the same value from the sensor

Zchulski, 2023-02-24:
I either get 511 or 1024 doesn't matter how far the sensor is from the bed with m102 s-2

markniu, 2023-02-24:
Could you paste some data after M102 S-5 here? for example the first 10 data.

Zchulski, 2023-02-24:
[image]

Zchulski, 2023-02-24:
i'm also seeing another issue when you run M102 S-6 only one of my z stepper motors is actually driving the moving.  I have a dual Z setup so both of those steppers should be driving the Z axis not just one of them

Zchulski, 2023-02-24:
tried homing the machine and crashed my head so thanks for that

Zchulski, 2023-02-24:
[image]

Raw data from this picture is saying 1024 according to your graph that number can't be correct

markniu, 2023-02-24:
and I also find that the frequency of processer stm32H743 is 480 MHz that is much higher than stm32f103(72MHz) which I used, so you can adjust the delay value from 800000 to 4800000 and then get the data again.

Zchulski, 2023-02-24:
1024 is the only number I get now

Zchulski, 2023-02-24:
> and I also find that the frequency of processer stm32H743 is 480 MHz that is much higher than stm32f103(72MHz) which I used, so you can adjust the delay value from 800000 to 4800000 and then get the data again.

You also are gonna need to fix the dual stepper issue as well I can't calibrate this if both my steppers arnt working

markniu, 2023-02-24:
the sensor has erased all the calibrate data in memory after sending M102 S-6 and will write the new calibrate in after finishing calibration.  if the sensor is out the sensor range while calibrate the data will be 1024
so you need to calibrate again.

markniu, 2023-02-24:
> You also are gonna need to fix the dual stepper issue as well I can't calibrate this if both my steppers arnt working

Oh, Thanks for feedback this, I will fix this as soon as possible.

Zchulski, 2023-02-24:
As of right now with this sensor I'm getting nothing but a value of 1024 with m102 s2 and s5

-------- Original Message --------

ptegler, 2023-02-24:
> tried homing the machine and crashed my head so thanks for that

why would you try to zero when your sensor wasn't working to begin with? hhmmm well... why are you trying to run a forked klipper that would get over written the first time klipper updates?  markniu...are you configuring along with mainsail, moonraker etc?   I have had this sensor in hand for a month+ now....have not even tried to use it until Klipper accepts it into the main app

Zchulski, 2023-02-24:
> why would you try to zero when your sensor wasn't working to begin with? hhmmm well... why are you trying to run a forked klipper that would get over written the first time klipper updates? [...]

according to mark the sensor is working just not in a real world application.  Also klipper won't update with the forked version, it says its dirty and you can't even do a hard reset on klipper.  I'm running the forked version on a separate SD card so when I need to use the printer I can just swap out SD cards.  I don't see this getting implemented into the main fork until he gets alot more kinks worked out of it.  for instance the dual stepper issue i'm running into, and who knows how many other boards are not gonna be supported right out of the box like mine wasn't.  Props for getting that fixed in a reasonable amount of time. 

i'm not entirely sure what the issue with my sensor is but it was driving me bonkers so I gave up for the night it just isn't registering anything from the bed.  I know the distance from the nozzle to probe is reasonably good because I made a mount for it in solidworks.

Zchulski, 2023-02-25:
re-downloaded the fork of klipper and now the dual z axis is working but I am still unable to get the sensor to calibrate.  It doesn't matter the distance to the bed the nozzle and probe are from the bed I am still getting a return of 1024 on the M102 commands.  Either my sensor is busted or this isn't going to work for me.  im seriously about to give up on this en devour in klipper. @markniu if you have any other ideas shoot them over to me I will keep trying but I don't know if this is going to work with this sensor

Zchulski, 2023-02-25:
@markniu 

I found part of my problem,  being that my clk and sda pin were configured incorrectly once that was fixed some of the above issues were fixing but i'm still having issues calibrating.  my calibration numbers are anywhere between 1024 and 511 on a single calibration.  whatever is going on here is getting weirder by the day.  Can you maybe put something in your klipper fork making klipper error out if the pins are configured backwards may help out new players here.  what else can i do to fix the calibration issues

markniu, 2023-02-27:
> @markniu
>
> I found part of my problem, being that my clk and sda pin were configured incorrectly once that was fixed some of the above issues were fixing but i'm still having issues calibrating. my [...]

adding some error message out is good idea.  I have fixed the double z axis problem, you can update the code from the fork.

closed 2023-04-25 by markniu