markniu/Bed_Distance_sensor #7: Getting Klipper internal error in MKS Robin Nano v3

proturka, 2023-03-04:
I bought your excellent precise distance sensor and i tried in arduino -which uses serial i2c- Its working very well as you documented.

But when i tried to connect to use with klipper -to board MKS Robin Nano v3- after i send M102 S-1 command, klipper gives internal error. And i think this issue is related to connection with the board.

I did every single step in this document, including printer.cfg: https://github.com/markniu/Bed_Distance_sensor/wiki/Installing-for-Klipper

I also compiled klipper with your repo and updated board's firmware without any issue.

I searched for the connection necessities in your documents (github readme, issues and wiki) but no results

I tried 2 different pin config. I used board's 5V / GND pins. and for the data pins, i used PB10/PB11 and PD10/PD11 as sda/scl but both didnt gave any results.

Any clue about what i am missing?

This is klipper log:

Internal error on command:"M102"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 216, in cmd_M102
    self.bd_sensor.I2C_BD_send("1015")#1015   read distance data
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 56, in I2C_BD_send
    self.I2C_BD_send_cmd.send([self.oid, data])
  File "/home/pi/klipper/klippy/mcu.py", line 530, in send
    cmd = self._cmd.encode(data)
  File "/home/pi/klipper/klippy/msgproto.py", line 175, in encode
    t.encode(out, params[i])
  File "/home/pi/klipper/klippy/msgproto.py", line 78, in encode
    out.extend(bytearray(v))
TypeError: string argument without an encoding
Transition to shutdown state: Internal error on command:"M102"

markniu, 2023-03-06:
I have not met this error on my pi3, but you can try this way to fix:
run `nano /home/pi/klipper/klippy/msgproto.py `
and find the line 78 and change the `out.extend(bytearray(v))` to `out.extend(bytearray(v, 'utf8'))`
reference to this https://stackoverflow.com/questions/31161243/python-string-argument-without-an-encoding

proturka, 2023-03-06:
I tried it but no change. 

But after i downgrade python version to 2, it worked like a charm. 

but now sensor only shows 1024. rarely gave 835 / 1023, but mostly 1024 only.

markniu, 2023-03-06:
What's the data returned after send command M102 S-1 and M102 S-5

proturka, 2023-03-06:
M102 S-1 "" (non visible chars)
M102 S-5 shows many 1024s

markniu, 2023-03-06:
there should be some text string after M-1. it seems there is communicating problem, I think need to increase the delay value in the configuration file, I will test it with octopus board that MCU is also STM32F4

proturka, 2023-03-07:
when i increased delay to 1600000 it kinda worked. 

M102 S-1 output is: "1 pandapdm"

when i increased to 1700000 same non visible chars, 2000000 is same.

when i set value to 1500000 it outputs "V1.0 pandapi3d.com". i think it is the best value for delay. it worked very well

thank you for your response. 

product is great. 

now i will test on real print soon. 

one last question, is it possible to use it as z end stop? or it is aligns only while printing with applying babysteps?

markniu, 2023-03-07:
that's great. it is only used as end stop while homing z axis.

proturka, 2023-03-07:
It is still unstable. Sometime it needs fully restart to delay setting change be applied. I am trying to find stable setting for delay, but i got no result yet. 

And M102 S-5 is starts from ~900 . In your documents it should start with ~0-75 as i can see. 
 
Is there any setting to make it work right, or i should wait more stable version for klipper?. I am not choosing marlin because there is huge quality gap, klipper is much better. 

Or i should calibrate it with your tool with marlin, and then start to use this calibration setting with klipper?

proturka, 2023-03-07:
I tried various distances. I even tried equal distance (almost). But i will check again and inform you about it. 

I have a related question. Is m102 s-2 gives relative result with calibration? Or it is absolute value from sensor?

markniu, 2023-03-07:
M102 S-2 gives absolute value from sensor but need to *0.01. for example, if it returns 435 that means 4.35mm
I think I should change the source code that let it return 4.35mm instead of 435.

proturka, 2023-03-07:
Then i can say, m102 s-2 gives correct values always. Relative with klipper’s z coordinate. 

Why there is difference between those? If you need some logs or spesific tests for it, i can provide

markniu, 2023-03-07:
the data after command M102 S-5 are calibration raw data, the sensor will get a new raw data at every distance and then it can calculate the actual distance value based on the calibration raw data.
the octopuls board is out of stock in my local market so I just placed a new order of board MKS robin V3 hope I can get it 2 days later.

neshami, 2023-03-11:
> I have not met this error on my pi3, but you can try this way to fix:
run nano /home/pi/klipper/klippy/msgproto.py 
and find the line 78 and change the out.extend(bytearray(v)) to out.extend(bytearray(v, 'utf8'))
reference to this https://stackoverflow.com/questions/31161243/python-string-argument-without-an-encoding

At least, after this edit finaly works as expected.

markniu, 2023-03-12:
Just update this tutorial :  https://github.com/markniu/Bed_Distance_sensor/wiki/Installing-for-Klipper
and also please update the sourcecode that fixed one communication bug for stm32f407 
[image]
[image]

markniu, 2023-03-12:
I also found the PA5 in the board MKS_RobinNanoV3 cannot be use as sda_pin because it works as other fuction in the bootloader that send a lot of signals while booting that let the BDsensor worked as unusual.

proturka, 2023-03-16:
I tried with updated source code now, but still same. calibration values starts with 900+. m102 s-2 is working well, fast and precised. but calibration is not. 

Do you have any suggestion to try?

proturka, 2023-03-16:
M102 S-2 shows 2.3mm only. And this is the photo of distance between nozle:

[image]

markniu, 2023-03-16:
that's close enough, and do calibrated again,if still problem send the first 10 calibration data give by M102 S-5 here

proturka, 2023-03-16:
Here the file. same thing happened. M102 S-1 and S-2 works well.

[image]

markniu, 2023-03-16:
is it a glass bed? Put one metal like thin knife close to the sensor instead of your bed, and do a calibrate again.

proturka, 2023-03-16:
its acrylic glass but on the top, there is aliminum plate. sensor able to detect it. 

but i will try calibration with knife or similar metal

proturka, 2023-03-17:
I did the test, and this is the result.

[image]

It looks like no change. idk what is the problem. I kept knife in zero distance with sensor

markniu, 2023-03-17:
> I did the test, and this is the result.
> 
> [image]
> 
> It looks like no change. idk what is the problem. I kept knife in zero distance with sensor

not sure what's the problem, could you send me your order number? I will send a new one to you.

proturka, 2023-03-18:
order number is 10350. thank you for your support

closed 2023-04-25 by markniu