markniu/Bed_Distance_sensor #8: On M102 command printer shutsdown

neshami, 2023-03-11:
When I send M102 command kliiper shuts down with message "Internal error on command:"M102"".
Creality board with STM32F103 for Ender 3S1 printer.
[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/10948333/klippy.log)

proturka, 2023-03-11:
Install klipper with python 2. It fixes problem. Same happened to me before.

neshami, 2023-03-11:
Can you send me message wit how to install with phyton 2?
Thanks in advance.

proturka, 2023-03-11:
I installed klipper with kiauh tool, and while install it asking python version. I tried 2 instead 3 and it worked normal.

neshami, 2023-03-11:
Klipper reports: ERROR

Option 'position_endstop' in section 'stepper_z' must be specified

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

neshami, 2023-03-11:
Unknown command:"M102"

neshami, 2023-03-11:
With latest git klipper for pull when compile firmware in firmware. bin is`t included BDsensor.o.
Response is as in my previous message. 
Unknown command:"M102"

proturka, 2023-03-11:
did you select custom repository? kiauh is picking original klipper github as repo source. need to change it from kiauh settings first. then need to compile and -if necessary- flash  the board again.

neshami, 2023-03-11:
Internal error on command:"M102"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/pi/klipper/klippy/webhooks.py", line 425, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/pi/klipper/klippy/gcode.py", line 216, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 196, in cmd_M102
    self.bd_sensor.I2C_BD_send("1016")#1016 // // read sensor version
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 56, in I2C_BD_send
    self.I2C_BD_send_cmd.send([self.oid, data])
  File "/home/pi/klipper/klippy/mcu.py", line 530, in send
    cmd = self._cmd.encode(data)
  File "/home/pi/klipper/klippy/msgproto.py", line 175, in encode
    t.encode(out, params[i])
  File "/home/pi/klipper/klippy/msgproto.py", line 78, in encode
    out.extend(bytearray(v))
TypeError: string argument without an encoding
MCU 'mcu' shutdown: Command request
clocksync state: mcu_freq=72000000 last_clock=4646036940 clock_est=(603.610 2829914922 71999358.209) min_half_rtt=0.000733 min_rtt_time=589.449 time_avg=603.609(413.426) clock_avg=2829914922.417(29766415415.715) pred_variance=497473376.934
Dumping serial stats: bytes_write=1484 bytes_read=14489 bytes_retransmit=9 bytes_invalid=0 send_seq=188 receive_seq=188 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0
Dumping send queue 100 messages
Sent 0 566.360553 566.360313 6: seq: 18, get_uptime
Sent 1 566.415440 566.415200 6: seq: 19, get_clock
Sent 2 566.469471 566.469231 6: seq: 1a, get_clock
Sent 3 566.523405 566.523165 6: seq: 1b, get_clock
Sent 4 566.577503 566.577263 6: seq: 1c, get_clock
Sent 5 566.631916 566.631676 6: seq: 1d, get_clock
Sent 6 566.686092 566.685852 6: seq: 1e, get_clock
Sent 7 566.740324 566.740084 6: seq: 1f, get_clock
Sent 8 566.794369 566.794129 6: seq: 10, get_clock
Sent 9 566.800646 566.800406 6: seq: 11, get_config
Sent 10 566.800951 566.800711 6: seq: 12, get_clock
Sent 11 566.810955 566.810675 7: seq: 13, allocate_oids count=22
Sent 12 566.811351 566.810991 9: seq: 14, config_endstop oid=0 pin=PC14 pull_up=2
Sent 13 566.811631 566.811233 7: seq: 15, config_trsync oid=1
Sent 14 566.812151 566.811527 13: seq: 16, config_I2C_BD oid=3 sda_pin=PC14 scl_pin=PC13 delay=1800000
Sent 15 566.812591 566.811812 11: seq: 17, config_digital_out oid=13 pin=PA0 value=0 default_value=0 max_duration=0
Sent 16 566.812991 566.812065 10: seq: 18, set_digital_out_pwm_cycle oid=13 cycle_ticks=720000
Sent 17 566.813431 566.812341 11: seq: 19, config_digital_out oid=14 pin=PC0 value=0 default_value=0 max_duration=0
Sent 18 566.813831 566.812587 10: seq: 1a, set_digital_out_pwm_cycle oid=14 cycle_ticks=720000
Sent 19 566.814151 566.812830 8: seq: 1b, config_analog_in oid=15 pin=PC4
Sent 20 566.814751 566.813114 15: seq: 1c, config_digital_out oid=16 pin=PA7 value=0 default_value=0 max_duration=360000000
Sent 21 566.815191 566.813343 11: seq: 1d, set_digital_out_pwm_cycle oid=16 cycle_ticks=7200000
Sent 22 566.815511 566.813562 8: seq: 1e, config_buttons oid=17 button_count=1
Sent 23 566.816631 566.814181 28: seq: 1f, config_analog_in oid=18 pin=ADC_TEMPERATURE, config_stepper oid=5 step_pin=PC2 dir_pin=PB9 invert_step=0 step_pulse_ticks=144, config_digital_out oid=19 pin=PC3 value=1 default_value=1 max_duration=0, config_endstop oid=6 pin=PA5 pull_up=1, config_trsync oid=7
Sent 24 566.817111 566.814411 12: seq: 10, config_stepper oid=8 step_pin=PB8 dir_pin=PB7 invert_step=0 step_pulse_ticks=144
Sent 25 566.817471 566.814643 9: seq: 11, config_endstop oid=9 pin=PA6 pull_up=1
Sent 26 566.817751 566.814850 7: seq: 12, config_trsync oid=10
Sent 27 566.818231 566.815105 12: seq: 13, config_stepper oid=11 step_pin=PB6 dir_pin=PB5 invert_step=0 step_pulse_ticks=144
Sent 28 566.820271 566.817508 51: seq: 14, config_analog_in oid=20 pin=PC5, config_digital_out oid=21 pin=PA1 value=0 default_value=0 max_duration=360000000, set_digital_out_pwm_cycle oid=21 cycle_ticks=7200000, config_stepper oid=12 step_pin=PB4 dir_pin=PB3 invert_step=0 step_pulse_ticks=144, finalize_config crc=3442363115, queue_digital_out oid=13 clock=194366491 on_ticks=0, queue_digital_out oid=14 clock=194378373 on_ticks=0
Sent 29 566.822311 566.817508 51: seq: 15, query_analog_in oid=15 clock=226800000 sample_ticks=72000 sample_count=8 rest_ticks=21600000 min_value=18817 max_value=32303 range_check_count=4, queue_digital_out oid=16 clock=194396452 on_ticks=0, buttons_add oid=17 pos=0 pin=PC15 pull_up=0, buttons_query oid=17 clock=300240000 rest_ticks=144000 retransmit_count=50 invert=1
Sent 30 566.824511 566.821137 55: seq: 16, query_analog_in oid=18 clock=300960000 sample_ticks=72000 sample_count=8 rest_ticks=21600000 min_value=10994 max_value=15264 range_check_count=4, query_analog_in oid=20 clock=302400000 sample_ticks=72000 sample_count=8 rest_ticks=21600000 min_value=731 max_value=32303 range_check_count=4, queue_digital_out oid=21 clock=194565598 on_ticks=0, get_config
Sent 31 566.842588 566.842308 7: seq: 17, stepper_get_position oid=5
Sent 32 566.847286 566.847006 7: seq: 18, stepper_get_position oid=8
Sent 33 566.851383 566.851103 7: seq: 19, stepper_get_position oid=11
[... cut ...]
Receive: 94 628.514991 627.848377 15: seq: 1a, analog_in_state oid=20 next_clock=349032704 value=31358
Receive: 95 628.664994 627.848377 15: seq: 1a, analog_in_state oid=15 next_clock=359832704 value=31343
Receive: 96 628.795070 627.848377 15: seq: 1a, analog_in_state oid=18 next_clock=369192704 value=13994
Receive: 97 628.815014 627.848377 15: seq: 1a, analog_in_state oid=20 next_clock=370632704 value=31357
Receive: 98 628.836308 628.833468 11: seq: 1b, clock clock=351069644
Receive: 99 628.917736 628.914734 12: seq: 1c, shutdown clock=356921107 static_string_id=Command request

proturka, 2023-03-11:
https://github.com/markniu/Bed_Distance_sensor/issues/7#issue-1609687260

same error happened to me also here, you can fix it with downgrading python version.

closed 2023-03-11 by neshami