markniu/Bed_Distance_sensor #12: Unable to calibrate sensor on SKR 2B

kendrickxy, 2023-04-06:
Hi,

I have an Anycubic Predator with SKR2B on Marlin.
I followed the installation steps, but I'm not able to get it to calibrate.

At first, I tried swapping SDA and SCL, but without success.

I've made a few modifications in bdl.cpp from line 175:

    
```
if (config_state == -6) {
        //BD_I2C_SENSOR.BD_i2c_write(1019); // begin calibrate
        //delay(1000);
        SERIAL_ECHOLNPGM("ENABLE STEPPERS PREcalibrate");
        gcode.stepper_inactive_time = SEC_TO_MS(60 * 5);
        gcode.process_subcommands_now(F("M17 Z"));
        gcode.process_subcommands_now(F("G1 Z-0.05"));     
        safe_delay(1000);
        SERIAL_ECHOLNPGM("SET POSITION PREcalibrate");
        gcode.process_subcommands_now(F("G92 Z0"));
        gcode.process_subcommands_now(F("G1 Z0.05"));   
        gcode.process_subcommands_now(F("G92 Z0"));
        safe_delay(1000);
        SERIAL_ECHOLNPGM("WHILE START PREcalibrate")

        while (planner.get_axis_position_mm(Z_AXIS) > 0.15) {    // changed >0.00001 to >0.015 and >0.15
          
          safe_delay(10);   // changed (1) to (10)
        }
        SERIAL_ECHOLNPGM("WHILE END PREcalibrate");

        z_pose = 0.00001;
        safe_delay(100);
        BD_I2C_SENSOR.BD_i2c_write(CMD_START_CALIBRATE); // Begin calibrate
        SERIAL_ECHOLNPGM("Begin calibrate");
        safe_delay(200);
        config_state = -7;
      }
```

Below are the outputs for M102 S-1, S-5 and S-6.

```
// M102 S-1

ok
BDsensor version:???????????????????

// M102 S-5

ok
Calibrate data:0,1023, check:0
Calibrate data:1,1023, check:0
Calibrate data:2,1023, check:0
Calibrate data:3,1023, check:0
Calibrate data:4,1023, check:0
Calibrate data:5,1023, check:0
Calibrate data:6,1023, check:0
Calibrate data:7,1023, check:0
Calibrate data:8,1023, check:0
Calibrate data:9,1023, check:0
Calibrate data:10,1023, check:0
Calibrate data:11,1023, check:0
Calibrate data:12,1023, check:0
Calibrate data:13,1023, check:0
Calibrate data:14,1023, check:0
Calibrate data:15,1023, check:0
Calibrate data:16,1023, check:0
Calibrate data:17,1023, check:0
Calibrate data:18,1023, check:0
Calibrate data:19,1023, check:0
Calibrate data:20,1023, check:0
Calibrate data:21,1023, check:0
Calibrate data:22,1023, check:0
Calibrate data:23,1023, check:0
Calibrate data:24,1023, check:0
Calibrate data:25,1023, check:0
Calibrate data:26,1023, check:0
Calibrate data:27,1023, check:0
Calibrate data:28,1023, check:0
Calibrate data:29,1023, check:0
Calibrate data:30,1023, check:0
Calibrate data:31,1023, check:0
Calibrate data:32,1023, check:0
Calibrate data:33,1023, check:0
Calibrate data:34,1023, check:0
Calibrate data:35,1023, check:0
Calibrate data:36,1023, check:0
Calibrate data:37,1023, check:0
Calibrate data:38,1023, check:0
Calibrate data:39,1023, check:0

// M102 S-6

ok
ENABLE STEPPERS PREcalibrate
SET POSITION PREcalibrate
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:64374B:64374C:64374
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:64366B:64366C:64366
WHILE START PREcalibrate //stuck in an infinite loop
```

From what we can see in the log above, it's stuck in an infinite loop once reaching `while (planner.get_axis_position_mm(Z_AXIS)>0.00001)`

I've changed `>0.00001` to `>0.015`, but nothing changed.

Any tips on solving the issue?

Thanks

markniu, 2023-04-07:
seems that there had connect problem:
the `check:0 ` means the data check is wrong, the correct value should be `check:1`, 
the message after M102 S-1 should be `BDsensor version:V1.0 pandapi3d.com`

did the z motor move while doing calibration?

markniu, 2023-04-07:
Just find the SKR 2 board use the stm32f4x that has higher processing speed than stm32f1, so you can try to increase the delay value from 20 to 100
`#define  I2C_BD_DELAY  100`

kendrickxy, 2023-04-07:
We're getting somewhere!

```
// M102 S-1

ok
BDsensor version:V1.0 pandapi3d.com

// M102 S-5

ok
Calibrate data:0,313, check:1
Calibrate data:1,327, check:1
Calibrate data:2,355, check:1
Calibrate data:3,406, check:1
Calibrate data:4,436, check:1
Calibrate data:5,451, check:1
Calibrate data:6,493, check:1
Calibrate data:7,493, check:1
Calibrate data:8,493, check:1
Calibrate data:9,566, check:1
Calibrate data:10,609, check:1
Calibrate data:11,632, check:1
Calibrate data:12,654, check:1
Calibrate data:13,663, check:1
Calibrate data:14,689, check:1
Calibrate data:15,711, check:1
Calibrate data:16,720, check:1
Calibrate data:17,743, check:1
Calibrate data:18,753, check:1
Calibrate data:19,764, check:1
Calibrate data:20,785, check:1
Calibrate data:21,794, check:1
Calibrate data:22,803, check:1
Calibrate data:23,820, check:1
Calibrate data:24,829, check:1
Calibrate data:25,838, check:1
Calibrate data:26,846, check:1
Calibrate data:27,854, check:1
Calibrate data:28,860, check:1
Calibrate data:29,867, check:1
Calibrate data:30,874, check:1
Calibrate data:31,880, check:1
Calibrate data:32,881, check:1
Calibrate data:33,887, check:1
Calibrate data:34,893, check:1
Calibrate data:35,897, check:1
Calibrate data:36,902, check:1
Calibrate data:37,907, check:1
Calibrate data:38,911, check:1
Calibrate data:39,911, check:1

// M102 S-6

ok
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:64375B:64375C:64375
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:64372B:64372C:64372
WHILE BEGIN
```

But as you see it's still stuck in the loop.

I home all 3 axis, send the command `M102 S-6` and the print head lower to the center of the plate as usual.
Once it's there, it's not able to come out of the loop though.

I'll try to run more tests.

markniu, 2023-04-08:
Hi, I think I know what's the problem is the home axis is not allowed before calibration. for the BDsensor is also used as the end stop of Z axis.

kendrickxy, 2023-04-09:
Hi,

At first, I tried to not home before (just tried again now), but it just runs the print head into the bed. not sure why.

Here are the logs I collected (with homing):

```
// M102 S-1

ok
BDsensor version:V1.0 pandapi3d.com

// M102 S-5

ok
Calibrate data:0,313, check:1
Calibrate data:1,327, check:1
Calibrate data:2,355, check:1
Calibrate data:3,406, check:1
Calibrate data:4,436, check:1
Calibrate data:5,451, check:1
Calibrate data:6,493, check:1
Calibrate data:7,493, check:1
Calibrate data:8,493, check:1
Calibrate data:9,566, check:1
Calibrate data:10,609, check:1
Calibrate data:11,632, check:1
Calibrate data:12,654, check:1
Calibrate data:13,663, check:1
Calibrate data:14,689, check:1
Calibrate data:15,711, check:1
Calibrate data:16,720, check:1
Calibrate data:17,743, check:1
Calibrate data:18,753, check:1
Calibrate data:19,764, check:1
Calibrate data:20,785, check:1
Calibrate data:21,794, check:1
Calibrate data:22,803, check:1
Calibrate data:23,820, check:1
Calibrate data:24,829, check:1
Calibrate data:25,838, check:1
Calibrate data:26,846, check:1
Calibrate data:27,854, check:1
Calibrate data:28,860, check:1
Calibrate data:29,867, check:1
Calibrate data:30,874, check:1
Calibrate data:31,880, check:1
Calibrate data:32,881, check:1
Calibrate data:33,887, check:1
Calibrate data:34,893, check:1
Calibrate data:35,897, check:1
Calibrate data:36,902, check:1
Calibrate data:37,907, check:1
Calibrate data:38,911, check:1
Calibrate data:39,911, check:1

// M102 S-6

ok
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:64354B:64354C:64354
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:64351B:64351C:64351
WHILE BEGIN
760.437
741.838
723.237
704.638
686.025
667.425
648.825
630.225
611.625
593.025
574.425
555.825
537.225
518.625
500.013
481.413
462.812
444.212
425.613
407.013
388.413
376.975
376.975
376.975
376.975
376.975
376.975
```

The values we see at the end are the following code :
```
while (planner.get_axis_position_mm(Z_AXIS)>0.005) {  //  CHANGED FROM 0.00001 TO 0.005       
   safe_delay(750);
   SERIAL_ECHO_F(planner.get_axis_position_mm(Z_AXIS), 3);//
}
```

For some reason, the printer starts at 760. The home point is exactly 450mm above the bed, but 760 - 376 is less than 400.

Here's my actual delta height   `#define DELTA_HEIGHT 450.00`

neshami, 2023-04-09:
With BDSensor you can't read height in mm except for first 3.9mm.
Therefor: data 0, 313
                .....................
                data 39,911
are values sensor returns for sensor heigth from 0mm to 3.9mm.

kendrickxy, 2023-04-09:
Here's a little more investigation. Not sure what to do from this point.

Below, the modifications done in bdl.cpp:
```
else if (config_state <= -6) {   // Start Calibrate
      safe_delay(10);
      if (config_state == -6) {
        //BD_I2C_SENSOR.BD_i2c_write(1019); // begin calibrate
        //delay(1000);
        gcode.stepper_inactive_time = SEC_TO_MS(60 * 5);
        gcode.process_subcommands_now(F("M17 Z"));
        gcode.process_subcommands_now(F("G1 Z-0.05"));     
        safe_delay(1000);
        gcode.process_subcommands_now(F("G92 Z0"));
        gcode.process_subcommands_now(F("G1 Z0.05"));   
        gcode.process_subcommands_now(F("G92 Z0"));
        safe_delay(1000);
        SERIAL_ECHOLNPGM("WHILE BEGIN");//
        safe_delay(10);//
        while (planner.get_axis_position_mm(Z_AXIS)>377.0f) {  //  CHANGED FROM 0.00001 TO 377
            
            safe_delay(70);
            //SERIAL_ECHO_F(planner.get_axis_position_mm(Z_AXIS), 3);//
            //SERIAL_ECHOLNPGM("\n");
        }
        SERIAL_ECHOLNPGM("WHILE END!");//

        z_pose = 0.0+377;
        safe_delay(100);
        BD_I2C_SENSOR.BD_i2c_write(CMD_START_CALIBRATE); // Begin calibrate
        SERIAL_ECHOLNPGM("Begin calibrate");
        safe_delay(200);
        config_state = -7;
      }
      else if (planner.get_axis_position_mm(Z_AXIS) < 387.0f) {
        SERIAL_ECHOLNPGM("IF AXIS < 378.0f.0f");//
        if (z_pose >= MAX_BD_HEIGHT) {
          SERIAL_ECHOLNPGM("IF Z_POSE > MAX_BD_HEIGHT");//
          BD_I2C_SENSOR.BD_i2c_write(CMD_END_CALIBRATE); // End calibrate
          SERIAL_ECHOLNPGM("End calibrate data");
          z_pose = 384;
          config_state = 0;
          safe_delay(1000);
        }
        else {
          SERIAL_ECHOLNPGM("ELSE");//
          float tmp_k = 0+377;
          char tmp_1[30];
          sprintf_P(tmp_1, PSTR("G1 Z%d.%d"), int(z_pose), int(int(z_pose * 10) % 10));
          gcode.process_subcommands_now(tmp_1);

          SERIAL_ECHO(tmp_1);
          SERIAL_ECHOLNPGM(" ,Z:", current_position.z);
   
          SERIAL_ECHOLNPGM("ELSE LOOP");//
          while ((0.004+377.0f) < abs(z_pose - tmp_k)) {    //
            tmp_k = planner.get_axis_position_mm(Z_AXIS);
            safe_delay(10);
          }
          SERIAL_ECHOLNPGM("ELSE LOOP END!");//
          safe_delay(100);
          if(z_pose<=0.4+377)
            safe_delay(500);
          tmp = (z_pose + 0.00001f); //*10
          BD_I2C_SENSOR.BD_i2c_write(tmp);
          SERIAL_ECHOLNPGM("w:", tmp, ",Zpose:", z_pose);
          z_pose += 0.1001;
          //queue.enqueue_now_P(PSTR("G90"));
        }
      }
      SERIAL_ECHOLNPGM("done.");//
    }
```

Here is the output:

```
// M102 S-1
ok
BDsensor version:V1.0 pandapi3d.com

// M102 S-6
ok
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:64354B:64354C:64354
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:64351B:64351C:64351
WHILE BEGIN
WHILE END!
Begin calibrate
done.
IF AXIS > 378.0f.0f
ELSE
G1 Z377.0 ,Z:377.05
ELSE LOOP
ELSE LOOP END!
w:377,Zpose:377.00
done.
done.
done.
```

Here's what happens on the printer:

After homing, and entering M102 S-6:
- The print head lowers to the bed.
- A fraction of a second after reaching the bed, it moves back up to height 377.

What could I be missing here?

kendrickxy, 2023-04-09:
After a little bit of research, I've found this parameter in Marlin:

```
/**
 * Set Z_IDLE_HEIGHT if the Z-Axis moves on its own when steppers are disabled.
 *  - Use a low value (i.e., Z_MIN_POS) if the nozzle falls down to the bed.
 *  - Use a large value (i.e., Z_MAX_POS) if the bed falls down, away from the nozzle.
 */
#define Z_IDLE_HEIGHT Z_MIN_POS
```

Now I'm able to start the leveling with the head near the bed. But I'm still having some issues with the file bdl.cpp.
Here's the output of logs:

```
ok
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:30150B:30150C:30150
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:30150B:30150C:30150
start while
Z_AXIS at: 376.93
Z_AXIS at: 376.93
Z_AXIS at: 376.93
Z_AXIS at: 376.93
Z_AXIS at: 376.93
```
With the value reset to their default parameters, I'm still having the same issues :c

markniu, 2023-04-10:
Hi, could you share your config files and pins?  actually, I have not tested on the delta printer, the config for the endstop should be different as other printers.   and one other thing you should know is the homing process use the virtual Z_MIN_PIN 
[image]

kendrickxy, 2023-04-10:
Hi,

Most of the issue is solved now :)
Here's the output:

```
// M102 S-1

ok
BDsensor version:V1.0 pandapi3d.com

// M102 S-5

ok
Calibrate data:0,44, check:1 Calibrate data:1,45, check:1
Calibrate data:2,47, check:1 Calibrate data:3,57, check:1
Calibrate data:4,57, check:1 Calibrate data:5,76, check:1
Calibrate data:6,95, check:1 Calibrate data:7,115, check:1
Calibrate data:8,139, check:1 Calibrate data:9,157, check:1
Calibrate data:10,185, check:1 Calibrate data:11,208, check:1
Calibrate data:12,224, check:1 Calibrate data:13,249, check:1
Calibrate data:14,273, check:1 Calibrate data:15,296, check:1
Calibrate data:16,325, check:1 Calibrate data:17,342, check:1
Calibrate data:18,367, check:1 Calibrate data:19,390, check:1
Calibrate data:20,411, check:1 Calibrate data:21,433, check:1
Calibrate data:22,456, check:1 Calibrate data:23,475, check:1
Calibrate data:24,494, check:1 Calibrate data:25,519, check:1
Calibrate data:26,535, check:1 Calibrate data:27,553, check:1
Calibrate data:28,570, check:1 Calibrate data:29,588, check:1
Calibrate data:30,605, check:1 Calibrate data:31,620, check:1
Calibrate data:32,637, check:1 Calibrate data:33,654, check:1
Calibrate data:34,672, check:1 Calibrate data:35,685, check:1
Calibrate data:36,695, check:1 Calibrate data:37,709, check:1
Calibrate data:38,723, check:1 Calibrate data:39,735, check:1

// M102 S-6

ok
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:30150B:30150C:30150
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:30150B:30150C:30150
WHILE START
WHILE END!
Begin calibrate
G1 Z0.0 ,Z:0.00 w:0,Zpose:0.00
G1 Z0.1 ,Z:0.10 w:1,Zpose:0.10
G1 Z0.2 ,Z:0.20 w:2,Zpose:0.20
G1 Z0.3 ,Z:0.30 w:3,Zpose:0.30
G1 Z0.4 ,Z:0.40 w:4,Zpose:0.40
G1 Z0.5 ,Z:0.50 w:5,Zpose:0.50
G1 Z0.6 ,Z:0.60 w:6,Zpose:0.60
G1 Z0.7 ,Z:0.70 w:7,Zpose:0.70
G1 Z0.8 ,Z:0.80 w:8,Zpose:0.80
G1 Z0.9 ,Z:0.90 w:9,Zpose:0.90
G1 Z1.0 ,Z:1.00 w:10,Zpose:1.00
G1 Z1.1 ,Z:1.10 w:11,Zpose:1.10
G1 Z1.2 ,Z:1.20 w:12,Zpose:1.20
G1 Z1.3 ,Z:1.30 w:13,Zpose:1.30
G1 Z1.4 ,Z:1.40 w:14,Zpose:1.40
G1 Z1.5 ,Z:1.50 w:15,Zpose:1.50
G1 Z1.6 ,Z:1.60 w:16,Zpose:1.60
G1 Z1.7 ,Z:1.70 w:17,Zpose:1.70
G1 Z1.8 ,Z:1.80 w:18,Zpose:1.80
G1 Z1.9 ,Z:1.90 w:19,Zpose:1.90
G1 Z2.0 ,Z:2.00 w:20,Zpose:2.00
G1 Z2.1 ,Z:2.10 w:21,Zpose:2.10
G1 Z2.2 ,Z:2.20 w:22,Zpose:2.20
G1 Z2.3 ,Z:2.30 w:23,Zpose:2.30
G1 Z2.4 ,Z:2.40 w:24,Zpose:2.40
G1 Z2.5 ,Z:2.50 w:25,Zpose:2.50
G1 Z2.6 ,Z:2.60 w:26,Zpose:2.60
G1 Z2.7 ,Z:2.70 w:27,Zpose:2.70
G1 Z2.8 ,Z:2.80 w:28,Zpose:2.80
G1 Z2.9 ,Z:2.90 w:29,Zpose:2.90
G1 Z3.0 ,Z:3.00 w:30,Zpose:3.00
G1 Z3.1 ,Z:3.10 w:31,Zpose:3.10
G1 Z3.2 ,Z:3.20 w:32,Zpose:3.20
G1 Z3.3 ,Z:3.30 w:33,Zpose:3.30
G1 Z3.4 ,Z:3.40 w:34,Zpose:3.40
G1 Z3.5 ,Z:3.50 w:35,Zpose:3.50
G1 Z3.6 ,Z:3.60 w:36,Zpose:3.60
G1 Z3.7 ,Z:3.70 w:37,Zpose:3.70
G1 Z3.8 ,Z:3.80 w:38,Zpose:3.80
G1 Z3.9 ,Z:3.90 w:39,Zpose:3.90
End calibrate data
```

The problem came from bdl.cpp as suspected. I simply replaced all `planner.get_axis_position_mm(Z_AXIS)` with `current_position.z`. Not sure why the command was off.

Still, I'm having some issues leveling the bed using automatic leveling `#define AUTO_BED_LEVELING_UBL`.

Here's my Z_MIN_PIN configuration:

```
#define Z_MIN_PIN          			         I2C_BD_SDA_PIN
#define Z_MIN_PROBE_PIN                  I2C_BD_SDA_PIN  //for skr2B
```

The command M119 outputs TRIGGERED when the nose is touching the bed. Though, not as the light turns on. I'm assuming it's working as expected.

Attached you can find my config files and pins.
[Configuration_adv.h.txt](https://github.com/markniu/Bed_Distance_sensor/files/11192941/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/markniu/Bed_Distance_sensor/files/11192943/Configuration.h.txt)
[pins_BTT_SKR_V2_0_common.h.txt](https://github.com/markniu/Bed_Distance_sensor/files/11192946/pins_BTT_SKR_V2_0_common.h.txt)

markniu, 2023-04-24:
this issue of calibration is the same as https://github.com/markniu/Bed_Distance_sensor/issues/13#issuecomment-1517310271  , I have pulled it to marlin github, I think the next version of marlin will be updated.

closed 2023-04-25 by markniu