markniu/Bed_Distance_sensor #13: check:0

gacek4000, 2023-04-16:
Hi, I have SKR 1.4 and Marlin-2.1.2. After connecting the BD sensor I keep getting "Calibrate data:28,1023, check:0". The red LED lights up when something metal is under the sensor, then the "BD sensor Z" value on the display changes from 10.14 to 0.0
Any ideas ?

[image]

i set:

#define  I2C_BD_SDA_PIN    P0_01  // Please change to the actual number which the SDA wire is connected to your mainboard
#define  I2C_BD_SCL_PIN    P0_00  // Please change to the actual number which the SCL wire is connected to your mainboard
#define  I2C_BD_DELAY  50

//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
// Force the use of the probe for Z-axis homing
//#define USE_PROBE_FOR_Z_HOMING

#define BD_SENSOR
#define Z_SAFE_HOMING
and so on...

markniu, 2023-04-17:
you can replace those 2 files with the following link:
https://github.com/markniu/Marlin_for_pull/blob/bugfix-2.1.x/Marlin/src/feature/bedlevel/bdl/bdl.cpp
https://github.com/markniu/Marlin_for_pull/blob/bugfix-2.1.x/Marlin/src/gcode/probe/M102.cpp

then you can send M102 S-1 to read the version of BDsensor that can help you to check if the connection is success.

gacek4000, 2023-04-17:
before replacing the 2 files you described:
https://photos.app.goo.gl/fSZpwm4N7xALxyRs5

after change:
https://photos.app.goo.gl/6uFVhADqvVGMHKbYA
additionally, in this case, instead of "BD sensor Z: 10.14" on the screen, there is "Settings stored" from the beginning

i2c configuration didn't change anything in either case

markniu, 2023-04-18:
That should be all the old calibrate data in the BDsensor has been erased to default 0xFF,  Please sending M102 S-6 and waiting it finished.

gacek4000, 2023-04-18:
<facepalm> 
ill check after job, thanks man !

gacek4000, 2023-04-18:
No, it doesn't work. It behaves the same as at the beginning: it starts moving down at the default speed. When I start the calibration, I have a screwdriver under the sensor, the LED was on red.

markniu, 2023-04-18:
the z axis should moving up.
and please make sure the calibration is finished by waiting there is message "End calibrate data" return back. or else there will return back 1023 and check 0 after M102 S-5

gacek4000, 2023-04-18:
when i send M102 S-6 the head goes down quite fast, nothing shows up in the console. homing and other head control from pronterface works fine

gacek4000, 2023-04-18:
I just replaced 'current_position.z' like kendrickxy and after 'M102 S-6' the head went down again, but in the console I had:

Begin calibration
G1Z0.0 ,Z:0.00
w:0,Zpose:0.00
G1Z0.1 ,Z:0.10
w:1,Zpose:0.10
G1Z0.2 ,Z:0.20
w:2,Zpose:0.20
G1Z0.3 ,Z:0.30
w:3,Zpose:0.30
G1Z0.4 ,Z:0.40
w:4,Zpose:0.40
G1Z0.5 ,Z:0.50

when I set
#define Z_MIN_PIN I2C_BD_SDA_PIN
#define Z_MIN_PROBE_PIN I2C_BD_SDA_PIN

printer behavior as above

[Configuration_adv.txt](https://github.com/markniu/Bed_Distance_sensor/files/11263324/Configuration_adv.txt)
[Configuration.txt](https://github.com/markniu/Bed_Distance_sensor/files/11263325/Configuration.txt)
[pins_BTT_SKR_common.txt](https://github.com/markniu/Bed_Distance_sensor/files/11263326/pins_BTT_SKR_common.txt)
[pins_BTT_SKR_V1_4.txt](https://github.com/markniu/Bed_Distance_sensor/files/11263327/pins_BTT_SKR_V1_4.txt)
[bdl.txt](https://github.com/markniu/Bed_Distance_sensor/files/11263328/bdl.txt)

gacek4000, 2023-04-18:
* M119 pokazuje 'z_probe: open' bez względu co jest pod czujnikiem (dioda świeci/ nie świeci)
* Z_MIN_PROBE_ENDSTOP_INVERTING true/false odpowiednio zmienia 'z_probe: open/triggered' ale głowa w obu przypadkach idzie na dół przy M102 S-6

markniu, 2023-04-19:
please add this `||P == Z_MIN_PROBE_PIN` in the endstop.cpp  like 
[image]

and try M119 again

markniu, 2023-04-19:
> * M119 shows 'z_probe: open' no matter what is under the sensor (LED on/off)
> * Z_MIN_PROBE_ENDSTOP_INVERTING true/false changes 'z_probe: open/triggered' accordingly but the head goes down on M102 S-6 in both cases

The homing position of your delta printer is the in the max position, but the calibration position is in the zero, so the head will go down to the zero after the calibration command M102 S-6.  recommend to checking the position before calibration by send gcode M114

gacek4000, 2023-04-19:
i did what you said and:

M119
Reporting endstop status
x_max: open
y_max: open
z_max: open
z_probe: open
filament: open
ok

with metal under sensor. calibration process - still the same.

'#define Z_MIN_PROBE_PIN                  P0_10'    is this correct ? When I uncomment it, the program doesn't compile

markniu, 2023-04-20:
although I haven't delta printer but have one SKRV1.4 board, I will run it on that board late or tomorrow late. and will update that after testing.

markniu, 2023-04-21:
[bdl.zip](https://github.com/markniu/Bed_Distance_sensor/files/11292705/bdl.zip)
I have test it with skrV1.4 board, and it can do calibration and M119 success.  please replace the those files. including the file probe.h in Marlin\src\module folder. and don't homing printer before calibration.

gacek4000, 2023-04-23:
It works ! everything works, screen and calibration. Thanks a lot buddy :) What have you done? Have I mixed something up?

markniu, 2023-04-24:
That's great, thanks for test it on delta printer.  
the position value got by  `planner.get_axis_position_mm(Z_AXIS)` is not zero in zero position for delta printer, so I fixed it by  `planner.get_axis_position_mm(Z_AXIS)-pos_zero_offset`

closed 2023-04-24 by gacek4000

gacek4000, 2023-04-25:
Hi again, I guess I rushed to announce success :( calibration works, M119 also but:
when trying to calibrate the delta (G33) or when trying to print a test print, the head goes down and does not react to the presence of the table under the sensor and collides with the table.
i set "#define DELTA_HEIGHT" and #define "Z_MAX_POS" to be less than the height of the printer. Then, when going down, even though I put metal under the sensor (the LED is on), the printer does not respond and then goes back up and I get a "Homing Failed" error
Any ideas ?