markniu/Bed_Distance_sensor #14: delta

gacek4000, 2023-04-26:
Hi again, I guess I rushed to announce success :( calibration works, M119 also but:
when trying to calibrate the delta (G33) or when trying to print a test print, the head goes down and does not react to the presence of the table under the sensor and collides with the table.
i set "#define DELTA_HEIGHT" and #define "Z_MAX_POS" to be less than the height of the printer. Then, when I start G33, and I put metal under the sensor (the LED is on), the printer does not respond and then goes back up and I get a "Homing Failed" error
[probe.txt](https://github.com/markniu/Bed_Distance_sensor/files/11326044/probe.txt)
[Configuration.txt](https://github.com/markniu/Bed_Distance_sensor/files/11326045/Configuration.txt)
[Configuration_adv.txt](https://github.com/markniu/Bed_Distance_sensor/files/11326046/Configuration_adv.txt)
[bdl.txt](https://github.com/markniu/Bed_Distance_sensor/files/11326048/bdl.txt)
[bdl_cpp.txt](https://github.com/markniu/Bed_Distance_sensor/files/11326049/bdl_cpp.txt)
[endstops.txt](https://github.com/markniu/Bed_Distance_sensor/files/11326050/endstops.txt)

Any ideas ?

markniu, 2023-04-26:
there should be some more returned log message after G33, and you may see what happed from the log.
and check G28 again, because the homing process of G33 and source code is the same as G28.

gacek4000, 2023-04-26:
Send: G28
Recv: >>> G28  X0.00 Y0.00 Z450.00
Recv: Machine Type: Delta
Recv: Probe: BD_SENSOR
Recv: Probe Offset X0.00 Y0.00 Z0.00 (Aligned With & Same Z as Nozzle)
Recv: >>> home_delta  X0.00 Y0.00 Z450.00
Recv:   current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
Recv: >>> homeaxis(X)
Recv: Home Fast: 675.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, 675.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Move Away: -1.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, -1.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Re-bump: 2.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, 2.00, 12.50)
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: adjDistance:-0.25
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, -0.25, 12.50)
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: <<< homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: Home Fast: 675.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Y, 675.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Move Away: -1.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Y, -1.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Re-bump: 2.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Y, 2.00, 12.50)
Recv:  T:22.25 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: <<< homeaxis(Y)
Recv: >>> homeaxis(Z)
Recv: Home Fast: 675.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, 675.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Move Away: -1.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, -1.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Re-bump: 2.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, 2.00, 3.13)
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: adjDistance:-0.84
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, -0.84, 3.13)
Recv: echo:busy: processing
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: <<< homeaxis(Z)
Recv: >>> set_axis_is_at_home(X)
Recv:   current_position= X0.00 Y0.00 Z460.00 :
Recv: <<< set_axis_is_at_home(X)
Recv: >>> set_axis_is_at_home(Y)
Recv:   current_position= X0.00 Y0.00 Z460.00 :
Recv: <<< set_axis_is_at_home(Y)
Recv: >>> set_axis_is_at_home(Z)
Recv:   current_position= X0.00 Y0.00 Z450.00 :
Recv: <<< set_axis_is_at_home(Z)
Recv:   current_position= X0.00 Y0.00 Z450.00 : sync_plan_position
Recv: <<< home_delta  X0.00 Y0.00 Z450.00
Recv: X:0.00 Y:0.00 Z:450.00 E:0.00 Count A:264538B:264538C:264538
Recv: <<< G28  X0.00 Y0.00 Z450.00
Recv: ok

gacek4000, 2023-04-26:
Send: M111 S32
Recv: echo:DEBUG:DETAIL
Recv: ok
Recv:  T:22.13 /0.00 B:22.19 /0.00 @:0 B@:0
Recv:  T:22.13 /0.00 B:22.03 /0.00 @:0 B@:0
Send: G28
Recv: >>> G28  X0.00 Y0.00 Z450.00
Recv: Machine Type: Delta
Recv: Probe: BD_SENSOR
Recv: Probe Offset X0.00 Y0.00 Z0.00 (Aligned With & Same Z as Nozzle)
Recv: >>> home_delta  X0.00 Y0.00 Z450.00
Recv:   current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
Recv: >>> homeaxis(X)
Recv: Home Fast: 675.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, 675.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Move Away: -1.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, -1.00, [25.00])
Recv:  T:22.00 /0.00 B:21.72 /0.00 @:0 B@:0
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Re-bump: 2.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, 2.00, 12.50)
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: adjDistance:-0.25
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, -0.25, 12.50)
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: <<< homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: Home Fast: 675.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Y, 675.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Move Away: -1.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Y, -1.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Re-bump: 2.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Y, 2.00, 12.50)
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: <<< homeaxis(Y)
Recv: >>> homeaxis(Z)
Recv: Home Fast: 675.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, 675.00, [25.00])
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Move Away: -1.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, -1.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Re-bump: 2.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, 2.00, 3.13)
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: adjDistance:-0.84
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, -0.84, 3.13)
Recv:  T:21.88 /0.00 B:21.88 /0.00 @:0 B@:0
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: <<< homeaxis(Z)
Recv: >>> set_axis_is_at_home(X)
Recv:   current_position= X0.00 Y0.00 Z460.00 :
Recv: <<< set_axis_is_at_home(X)
Recv: >>> set_axis_is_at_home(Y)
Recv:   current_position= X0.00 Y0.00 Z460.00 :
Recv: <<< set_axis_is_at_home(Y)
Recv: >>> set_axis_is_at_home(Z)
Recv:   current_position= X0.00 Y0.00 Z450.00 :
Recv: <<< set_axis_is_at_home(Z)
Recv:   current_position= X0.00 Y0.00 Z450.00 : sync_plan_position
Recv: <<< home_delta  X0.00 Y0.00 Z450.00
Recv: X:0.00 Y:0.00 Z:450.00 E:0.00 Count A:264538B:264538C:264538
Recv: <<< G28  X0.00 Y0.00 Z450.00
Recv: ok
Send: M113 S2
Recv: ok

gacek4000, 2023-04-26:
I put metal under the sensor a couple of times while going down:

Send: G33 P4
Recv: G33 Auto Calibrate
Recv: Checking... AC at radius:155.00
Recv: .Height:450.00  Ex:-0.25  Ey:0.00  Ez:-0.84  Radius:227.40
Recv: .               Tx:+0.13  Ty:-0.18  Tz:+0.05
Recv: >>> home_delta  X0.00 Y0.00 Z450.00
Recv:   current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
Recv: >>> homeaxis(X)
Recv: Home Fast: 675.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, 675.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Move Away: -1.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, -1.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Re-bump: 2.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, 2.00, 12.50)
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: adjDistance:-0.25
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(X, -0.25, 12.50)
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: <<< homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: Home Fast: 675.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Y, 675.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Move Away: -1.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Y, -1.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Re-bump: 2.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Y, 2.00, 12.50)
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: <<< homeaxis(Y)
Recv: >>> homeaxis(Z)
Recv: Home Fast: 675.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, 675.00, [25.00])
Recv:  T:22.25 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Move Away: -1.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, -1.00, [25.00])
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: Re-bump: 2.00mm
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, 2.00, 3.13)
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: adjDistance:-0.84
Recv: >>> do_homing_move  X0.00 Y0.00 Z460.00
Recv: ...(Z, -0.84, 3.13)
Recv: echo:busy: processing
Recv: <<< do_homing_move  X0.00 Y0.00 Z460.00
Recv: <<< homeaxis(Z)
Recv: >>> set_axis_is_at_home(X)
Recv:   current_position= X0.00 Y0.00 Z460.00 :
Recv: <<< set_axis_is_at_home(X)
Recv: >>> set_axis_is_at_home(Y)
Recv:   current_position= X0.00 Y0.00 Z460.00 :
Recv: <<< set_axis_is_at_home(Y)
Recv: >>> set_axis_is_at_home(Z)
Recv:   current_position= X0.00 Y0.00 Z450.00 :
Recv: <<< set_axis_is_at_home(Z)
[... 188 lines cut ...]
Recv: Axis Z min:0.00 max:450.00
Recv: Axis Z min:0.00 max:370.65
Recv: Iteration : 01                                std dev:30.209
Recv: >>> home_delta  X0.00 Y0.00 Z0.00
Recv:   current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
Recv: echo:busy: processing
Recv:  T:21.88 /0.00 B:21.56 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.00 /0.00 B:22.03 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.00 /0.00 B:21.88 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.00 /0.00 B:21.88 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:21.88 /0.00 B:22.03 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.00 /0.00 B:21.88 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.13 /0.00 B:21.88 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.13 /0.00 B:22.03 /0.00 @:0 B@:0
Recv: echo:Homing Failed
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error"
Send: M112

gacek4000, 2023-04-26:
i have BD sensor is connected to i2c port of SKR, andz min probe on P0_10 and:

#define Z_MIN_PIN                        P0_10
#define Z_MIN_PROBE_PIN            P0_10
#define BD_SENSOR

effect as above. M119 shows no reaction on probe status. only react on BD status. when i set:

#define FIX_MOUNTED_PROBE
//#define BD_SENSOR

i get proper reaction in M119 and during calibration process. BD sensor should disable probe ?

markniu, 2023-04-28:
from the log we can see the printer `home_delta X0.00 Y0.00 Z450.00 `  first ,then probe some points and then `home_delta X0.00 Y0.00 Z0.00` , but the last home processing to Z0 is last for short time, so I think we cannot put a metal to trigger the probe in the middle way while in probing process.

there is no toolhead move by default while in probing with BDsensor, but for debug you can let the BDsensor just work as normal proximity sensor by delete this code in probe.cpp :
[image]

gacek4000, 2023-04-30:
This is not a solution that satisfies me.
I also noticed that even though M119 shows that Z_probe is triggered, I can still move my head up and down, for some reason the printer completely ignores the Z_probe status

gacek4000, 2023-05-25:
Hello it is me again. I think the previous problems were my fault; I've made some progress but it still doesn't work.
I tested the sensor on merlin version: 2.1.2  &  2.1.2.1  &  your _pull, allways I get the result as below.

- calibration goes correct, M102 S-5 from the sensor looks ok.

- M119 works properly.

- G30 & G33 & G29 doesn't work properly - the head always collides with the bed and the log (M111 S32) (attachment) shows that the sensor did not detect the bed.

- when trying to print, the head hits the bed and prints in collision with the bed. The distance readings on the screen changes from 0 to 0.20

I noticed 2 things:

1.  i.e. G30 -> head goes to bed. When, shortly before the collision the head with the bed, I put metal under the sensor and hold it for a short while, then the sensor detects it and I have a "correct" reading of the bed. When I touch the sensor for a short while and take the metal away, the sensor does not detect it and the head continues to slide down

2.  head is up, nothing is moving. I keep sending M119, I put metal under the sensor and only after a while, after a few M119s, there is information that the sensor has been activated

My conclusion: there is a delay in reading from the sensor. I tried to change I2C_BD_DELAY from 10 to 200, no difference.
Any ideas :) ? 

[LOGS.txt](https://github.com/markniu/Bed_Distance_sensor/files/11568547/LOGS.txt)

markniu, 2023-08-11:
hello, I just get one delta printer and I have checked the klipper firmware works with bd sensor.
I will check the marlin firmware next.

kendrickxy, 2023-10-17:
Hi @markniu let us know if you got a follow up on that.
Thanks!

markniu, 2023-10-18:
> Hi @markniu let us know if you got a follow up on that. Thanks!

Here is the beta version for marlin delta printer: https://github.com/markniu/MarlinPULL