markniu/Bed_Distance_sensor #17: M102 S-1 returns nothing.

nicr4wks, 2023-05-23:
Hi All,

I'm using the sensor on an SKR2 board connected to a custom HEVO printer with Marlin-2.1.2

I've been able to calibrate the sensor and can home Z no problem. I have a problem where the "M102 S-1" does not return any data. "M102 S-5" works fine and shows all calibration data so it seems like communication to the sensor is OK.

Can I be sure that "M102 S2" is working OK since other commands are giving problems?

Thanks.

markniu, 2023-05-23:
Hi, there is a new pull request into marlin about some optimize and new command like  "M102 S-1": https://github.com/MarlinFirmware/Marlin/pull/25847  , I think this will be accepted in the next marlin version.
and the that pull request fork link is https://github.com/markniu/Marlin_pull , you can use that marlin fork link too.
other command in the official marlin like M102 S2 is working.

nicr4wks, 2023-05-23:
Thanks, I'll wait for official Marlin release and test it out!

closed 2023-05-25 by nicr4wks