markniu/Bed_Distance_sensor #19: mcu 'mcu': Unknown command: I2C_BD_send

coexmark, 2023-05-24:
I followed the Klipper install instructions and get the following error:

```
MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.11.0-201-g37315bf3-dirty
MCU(s) which should be updated:
mcu: Current version v0.10.0-623-g5b1a6676
Up-to-date MCU(s):
<none>

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'mcu': Unknown command: I2C_BD_send
```

markniu, 2023-05-25:
seems that the firmware you flashed is not right one, did you flashed the new compiled firmware after install patch of BDsensor?

noohnak, 2023-06-27:
Im having same issue and i flashed over usb

markniu, 2023-06-27:
y

> Im having same issue and i flashed over usb

you can check the mcu version in the klippy.log file to see if the firmware is flashed right. or share some error message in that log file.

closed 2023-07-18 by markniu