markniu/Bed_Distance_sensor #22: Open hardware, or commercial

punjabik, 2023-06-25:
Is this project open hardware or commercial?

I cannot find schematics like in PandaCan and firmware of the board (if there is any).
I don't want to invest money in closed solution, so I'd like have answer to this question before buy and investing time.

Thank You.

markniu, 2023-06-26:
I would like to open source this, but I really want to find a way to support me to continue working in the 3D printing world, sorry.

punjabik, 2023-06-26:
> ??? Beacon, BLTouch, etc are not open source...why would you believe this is/should be?

Because I find this project on hackaday.io which is **repository of Open Hardware Projects** as they stated on their website.

So this is not open hardware project and should not use hackaday for gaining publicity and profit. I am glad, that I did not buy product and carefully check content of github first.

[markniu](https://github.com/markniu) there is plenty of ways how to support development.
Currently there is not even a simple button to send a donation.

closed 2023-06-26 by markniu