markniu/Bed_Distance_sensor #24: random klipper errors

noohnak, 2023-07-15:
I can not get klipper with mainsail to run without giving me error code after error code. seemingly random. first it was please specify pin for probe. then after restart it was no pin assigned for probe etc.

noohnak, 2023-07-15:
oh then after all errors are fixed or fix themselves mcu wont connect. if i delete all the config related to BDsensor it works fine connects and everything

ManuelPeiro, 2023-07-15:
What board are you connecting it to... a BlTouch connector?

noohnak, 2023-07-15:
a spider v2.2 and the bltouch pins

markniu, 2023-07-16:
check the section name [BDsensor] and the first two word is capitalize, other words are lower case

noohnak, 2023-07-17:
klipper: ERROR 
Unknown pin chip name 'probe'

noohnak, 2023-07-19:
its all correct still having unknown pin chip name 'probe'

noohnak, 2023-07-19:
is there maybe a file im missing

markniu, 2023-07-19:
you can send me or share the klippy.log file.

noohnak, 2023-07-19:
[klippy (1).log](https://github.com/markniu/Bed_Distance_sensor/files/12093820/klippy.1.log)

markniu, 2023-07-19:
is there a file BDsensor.py in the path:/home/pi/klipper/klippy/extras ?

markniu, 2023-07-19:
you can run this command in the ssh to check if there is a file BDsensor.py
`ls ~/klipper/klippy/extras/BDsensor.py -l`

noohnak, 2023-07-24:
there was no file in that path.

markniu, 2023-07-24:
> there was no file in that path.

that means the patch code of BDsensor for klippy was not installed or not successful, did you run the `python ~/Bed_Distance_sensor/klipper/install_BDsensor_patch.py ` after git clone the patch code ? or run again and to see if it is installed successful?

CognitiveFeline, 2023-10-29:
it's moonraker viewing Klipper as "dirty" you're hitting update after setup, and Klipper/Klippy via moonraker auto cleans maker out and removes everything added "after" so the manual addition of BDsensor is being cleansed. you have to re-install BDsensor and avoid hitting refresh on the moonraker updates, either remove Klipper from the update check or just reinstall after each update ignoring when it simply says it's "dirty". 

What I did was added BDsensor repo to the update-manager to track and make it easy to reinstall whenever Klipper has an update, not sure if it's possible it have it ignore the BDsensor add on installed into Klippy but this is as far as I've gotten for what stare d off a very puzzling repeating error and reinstall thinking i did something wrong... which I guess sort of did but hope this helps you cause otherwise BDsensor works great with Klipper and moonsail is hands down 10x better with it all!

- BDsensor
new: 1 commit available

- Klipper
v0.11.0-304-gf7567a0d-dirty  **_instead of_** v0.11.0-304-gf7567a0d

 _this is what will appear along wiht something like whats written below and offering to hard or soft fix it for you..._
> Repo is dirty. Detected the following modifed files: ['scripts/buildcommands.py', 'src/Makefile']
 
_so either just ignore this if the version number is the same that it wants to clean/update to or better solution my current ignore solution_

- KlipperScreen
v0.3.6-65-g4a14962d
- mainsail
v2.8.0
- mobileraker
v0.4.0-29-g5a4cae4a
- moonraker
v0.8.0-188-ga71c5c15
- moonraker-obico
v1.5.1-0-g3ed711f5
- System
OS-Packages

So as mentioned, I added the BDsensor to moonraker.cfg to update the install after/when I get Klipper updates, accidently clean it, or otherwise when needed to put it back quickly and easily, usually when the error you see, "Probe" yadda yadda yadda. Plus this lets me know if changes have been made to BDsensor as well.

I just copied how KlipperScreen wanted me to set up the updates for it.

```
[update_manager KlipperScreen]
type: git_repo
path: ~/KlipperScreen
origin: https://github.com/jordanruthe/KlipperScreen.git
env: ~/.KlipperScreen-env/bin/python
requirements: scripts/KlipperScreen-requirements.txt
install_script: scripts/KlipperScreen-install.sh
managed_services: KlipperScreen
```

Just added it like this to the bottom of the file.

```
[update_manager BDsensor]
type: git_repo'
path: ~/Bed_Distance_sensor'
origin: https://github.com/markniu/Bed_Distance_sensor.git'
primary_branch: new'
install_script: klipper_Beta/install_BDsensor.sh 
```
Hope this helps :)

markniu, 2023-10-31:
> it's moonraker viewing Klipper as "dirty" you're hitting update after setup, and Klipper/Klippy via moonraker auto cleans maker out and removes everything added "after" so the manual addition of [...]
 
This is a useful