markniu/Bed_Distance_sensor #25: klipper && BDsensor no connect

gerkow, 2023-07-20:
v0.11.0-248-gd725dfd3-dirty
EBB42 over U2C can0

Build file /home/biqu/klipper/klippy/../out/klipper.elf(1055796): Thu Jul 20 15:24:18 2023
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
File "/home/biqu/klipper/klippy/serialhdl.py", line 68, in _get_identify_ data
params = self.send with response(msg, 'identify_response')
File "/home/biqu/klipper/klippy/serialhdl.py", line 261, in send with response
return src.get_response([cmd], self.default_cmd_queue)
File "/home/biqu/klipper/klippy/serialhdl.py", line 318, in get _response
self.serial.raw_ send wait_ack(cmds[-1], minclock, reqclock
File "/home/biqu/klipper/klippy/serialhdl.py", line 253, in raw_send wait_ack
self. _error("Serial connection closed")
File "/home/biqu/klipper/klippy/serialhdl.py", line 61, in error
raise error(self.warn_prefix + (msg % params))
serialhdl.error: mcu 'mcu': Serial connection closed

markniu, 2023-07-20:
from that log we can only see that the klipper can not connect to the mcu of mainboard not the EBB42. but not sure what's the reason, you can return back to use your earliy config file that without BDsensor to seem if it still report the same error

gerkow, 2023-07-20:
Thanks for the Quick respond.
2,or 3? weeks ago it had been working. After klipper update it stopped.

I installed new image from scratch.
Flashed U2C new. Flashed canboot.
Made firmware from pure klipper sources.
flash_can works (and do it allways)
the EBB is working (extruderstepper, fan...)
After applying the BDsensor patch and upload firmware it dont connect.
I See RX on can0 but no TX

I digged in sources and have no Club what the case is.
Seems to freeze mcu somewhere.

markniu, 2023-07-21:
you can try to run the config that without the BDsensor,  in that case the klippy or the firmware in mcu will not run the patch code. and that will help to find out what's problem.

gerkow, 2023-07-21:
The problem was 
[BDsensor]
delay: 20000    (too much for STM32G0B1)
Set to 20 corrected it

Solved

markniu, 2023-07-21:
OK，I will add a note of that range limit in the installation

closed 2023-12-16 by markniu