markniu/Bed_Distance_sensor #26: MCU'S  need update

noohnak, 2023-07-26:


markniu, 2023-07-26:
did you flash the firmware into SB2040 with the new firmware after patch the code of BDsensor?

noohnak, 2023-07-26:
no i only changes the spider 2.2 controll board. i will try updating the can toolhewlead mcu also.

noohnak, 2023-07-27:
am i supposed to do the toolhead also? So o i do it the same way as my control board?

markniu, 2023-07-27:
> am i supposed to do the toolhead also? So o i do it the same way as my control board?

`make menuconfig` and choose the setting for SB2040 and `make `the new firmware for the SB2040, then flash into SB2040

noohnak, 2023-07-27:
ok i flashed the sb2040 and still get

MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.11.0-248-gd725dfd3-dirty
MCU(s) which should be updated:
mcu: Current version v0.11.0-226-g0245073c
SB2040: Current version v0.11.0-248-gd725dfd3
Up-to-date MCU(s):
<none>

[klippy (7).log](https://github.com/markniu/Bed_Distance_sensor/files/12182427/klippy.7.log)

markniu, 2023-07-27:
seems you connect the BD sensor on the spider board? if in that case, could you try to make meuconfig and make for the spider firmware and flash into spider board again?

noohnak, 2023-08-06:
so the only way i can get it to work is if i build and flash my board through kiauh. but then whn i go to save config after setup and restart firmware it gives error mcu's need update again.

markniu, 2023-08-07:
could you read the distance value or the version information from the BDsensor after setup?

noohnak, 2023-08-11:
Yes after loading firmware with kiauh I could control printer normally. and read value and version info. but then try to save and reload. klipper errors again

markniu, 2023-08-11:
> 

Hello, I have not repeated this problem, could you attach the klippy.log file?

closed 2023-12-16 by markniu