markniu/Bed_Distance_sensor #27: Auto leveling

ESSAMADECARAMBA, 2023-07-31:


markniu, 2023-07-31:
yes, with any free 2 gpios on your board.  which board does the bdsensor connect to?

Townie-au, 2023-11-19:
> yes, with any free 2 gpios on your board. which board does the bdsensor connect to?

Is this only with marlin you're talking about?

markniu, 2023-12-16:
GPIO means general-purpose input/output is a digital signal pin that can be used as input and output for digital communication. so need to check if the pin can be used as input and output on the mainboard.

Since some of the pins in the connectors on the mainboard maybe not connected to the gpios of MCU directly (e.g. there maybe a filter capacitor on them or isolated by mosfet, diode or optocoupler, but it's OK if they are isolated by resistors or pullup/pulldown by resistors), they cannot work with BDsensor. and the firmware will report connection error. For example

The connectors for FANs and Heaters are isolated by mosfet,
The connectors for temperature thermistors and Endstops/Probe in some boards like MKS are normally connected to GND by filtered capacitors,

the wires CKL and SDA of BDsensor can be connected to any GPIO pins of your board. You can also connect the BDsensor cable into the Bltouch port directly (**except the MKS board**), for example:

BLtouch    |    BDsensor
 5V       -->     5V
 GND      -->     GND
 S        -->     CLK/SCL    (Input)
 GND      -->     GND
 Zmin     -->     SDA    (Input/Output)

closed 2023-12-16 by markniu