markniu/Bed_Distance_sensor #28: Replicape on BeagleBone, wont fit on pru

gaby64, 2023-08-26:
Trying to build for the BeagleBone PRU, I get:
```region `imem' overflowed by 3236 bytes```

I thought i might be able to get away with just building it for host since I would be using host pins, but that did not work, I get
```
MCU Protocol error

...

mcu 'mcu': Unknown command: I2C_BD_send
```

Any way to reduce the size needed on the BeagleBone PRU?

markniu, 2023-09-01:
seems the error : `region `imem' overflowed by 3236 bytes` comes from the make command.  which board do you used for the BDsensor? and would you take screenshot about your `make menuconfig` ?

gaby64, 2023-09-01:
I got around this by having the BD_sensor connected to host, it wasn't working at first but then I found I had to specify host and gpiochip, I'm still having issues finding the right to use.

The hardware is the Replicape on BeagleBone Black.

markniu, 2023-09-01:
we have not test that on host before, so don't connect to host. the BDsensor should connect to MCU on the board. 
I will try to compile with BeagleBone PRU like you did before.

gaby64, 2023-09-01:
[image]
[image]

I think without bit-banging i2c won't be possible, so it will not fit on the pru memory, maybe with some debugging features disabled/excluded.

gaby64, 2023-09-01:
Also the Replicape inputs are connected to a logic level converter CD4050BM, as the BeagleBone Black is not 5V tolerant.

markniu, 2023-09-01:
> Also the Replicape inputs are connected to a logic level converter CD4050BM, as the BeagleBone Black is not 5V tolerant.
> 
> Can the BD_Sensor work with 3.3V?

the power pins(5V) for the BD sensor must be 5V. the pins of CLK and SDA can be work with 3.3v to 5V.

gaby64, 2023-09-01:
So it should fine to connect to BBB gpio pins directly to BD_Sensor and power it with 5V.
I will test this tomorrow, although it will be through host.

gaby64, 2023-09-02:
Success with getting a connection returning the proper string, distance measurement works, but I am running into an issue.
After successfully calibrating, I tried running PROBE_ACCURACY , I got:
```
PROBE_ACCURACY at X:-0.669 Y:0.373 Z:522.645 (samples=10 retract=2.000 speed=0.8 lift_speed=0.8)
Communication timeout during homing probe
.set_position Z is 10.150 mm
```

markniu, 2023-09-03:
the probe_accuracy is used for proxmity probe sensor that will move the z axis up and down then check the accuacy of the probe sensor, so the command of probe_accuracy is actually no needed for BDsensor,  you can check the accuacy of BDsensor by sending the M102 S-2 at the same points for several times.

and what type of printer do you have? delta ?

gaby64, 2023-09-03:
Delta. I also get the error when issuing a PROBE command.

markniu, 2023-09-03:
did it report communication timeout during PROBE command?

gaby64, 2023-09-03:
Yes

gaby64, 2023-09-03:
It errors before it even has the chance to move down to the probing position

markniu, 2023-09-04:
> It errors before it even has the chance to start moving down to the probing position.
> 
> Its not moving at all, I even increased the TRSYNC_TIMEOUT, no stepper movement during PROBE command.

how about using a normal proximity probe?

gaby64, 2023-09-04:
im not sure what you mean

how could I debug why I am getting "Communication timeout during homing probe"?

I tried to add more logging to the code, I use PROBE_ACCURACY because its more clear where the command begins in the log:
```
PROBE_ACCURACY at X:-0.669 Y:0.373 Z:522.645 (samples=10 retract=2.000 speed=40.0 lift_speed=40.0)

pos -0.669,0.373,522.645 -5.000
BD lower_probe0
BD home_start
Stats 133808.9: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000002 mcu_task_stddev=0.000003 bytes_write=3018 bytes_read=3526 bytes_retransmit=0 bytes_invalid=0 send_seq=183 receive_seq=183 retransmit_seq=0 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=199999959 host: mcu_awake=0.004 mcu_task_avg=0.000043 mcu_task_stddev=0.000045 bytes_write=1212 bytes_read=7973 bytes_retransmit=0 bytes_invalid=0 send_seq=138 receive_seq=138 retransmit_seq=0 srtt=0.002 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49998833 adj=49997787 heater_bed: target=0 temp=20.0 pwm=0.000  sysload=1.38 cputime=5.039 memavail=355064 print_time=133754.239 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.3 pwm=0.000
BD home_wait0
BD raise_probe
BD raise_probe
Communication timeout during homing probe, trigger_time: -1.0, move_end_print_time: 133754.51820993
get_position Z is 521.461 mm
.set_position Z is 10.150 mm
```

gaby64, 2023-09-04:
Starting closer to the bed (5mm), I get a different result

```
PROBE_ACCURACY at X:-0.669 Y:0.373 Z:5.000 (samples=10 retract=2.000 speed=40.0 lift_speed=40.0)

pos -0.669,0.373,5.000 -5.000
BD lower_probe0
BD home_start
Stats 134494.1: gcodein=0  mcu: mcu_awake=0.000 mcu_task_avg=0.000001 mcu_task_stddev=0.000002 bytes_write=30506 bytes_read=26884 bytes_retransmit=0 bytes_invalid=0 send_seq=1607 receive_seq=1607 retransmit_seq=0 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=200000039 host: mcu_awake=0.003 mcu_task_avg=0.000037 mcu_task_stddev=0.000036 bytes_write=7222 bytes_read=91263 bytes_retransmit=0 bytes_invalid=0 send_seq=980 receive_seq=980 retransmit_seq=0 srtt=0.002 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49998209 adj=49998589 heater_bed: target=0 temp=20.0 pwm=0.000  sysload=1.15 cputime=29.275 memavail=354976 print_time=134439.416 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.1 pwm=0.000
BD home_wait0
BD raise_probe
BD raise_probe
probe at -0.669,0.373 is z=-3.450045 bd0.000  pos_new:-3.450045
pos -0.669,0.373,-1.450 -5.000
BD lower_probe0
BD home_start
BD home_wait0
BD raise_probe
BD raise_probe
Probe triggered prior to movement
get_position Z is -1.450 mm
.set_position Z is 0.920 mm
```

gaby64, 2023-09-04:
Tweaked the printer.cfg and got success doing a PROBE_ACCURACY starting at 5mm
```
[printer]
minimum_z_position: -0.5

[BDsensor]
scl_pin: host:gpiochip3/gpio19
sda_pin: host:gpiochip0/gpio5
delay: 40
z_offset:0 # this `z_offset` must be set to 0.
x_offset: 0
y_offset: 18
speed: 4
samples:2
no_stop_probe: True # enable this for fast probe, the toolhead will not stop at the probe point.
position_endstop: 0.8 # the Z axis will stop at this position (mm) while homing z, recommend value is 0.4~1.0

[bed_mesh]
speed: 150
horizontal_move_z: 1
mesh_radius: 80
mesh_origin: 0, 0
round_probe_count: 5

[delta_calibrate]
radius:100
horizontal_move_z: 20
speed:50
```

```PROBE_ACCURACY at X:-0.669 Y:0.373 Z:5.000 (samples=10 retract=2.000 speed=4.0 lift_speed=4.0)

pos -0.669,0.373,5.000 -0.500
BD lower_probe0
BD home_start
Stats 135053.6: gcodein=0  mcu: mcu_awake=0.000 mcu_task_avg=0.000001 mcu_task_stddev=0.000002 bytes_write=8553 bytes_read=7013 bytes_retransmit=0 bytes_invalid=0 send_seq=368 receive_seq=368 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=199999947 host: mcu_awake=0.003 mcu_task_avg=0.000039 mcu_task_stddev=0.000037 bytes_write=1597 bytes_read=10853 bytes_retransmit=0 bytes_invalid=0 send_seq=194 receive_seq=194 retransmit_seq=0 srtt=0.002 rttvar=0.002 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49998334 adj=49999118 heater_bed: target=0 temp=20.3 pwm=0.000  sysload=0.82 cputime=24.107 memavail=353920 print_time=134998.990 buffer_time=0.000 print_stall=0 extruder: target=0 temp=20.9 pwm=0.000
Stats 135054.6: gcodein=0  mcu: mcu_awake=0.000 mcu_task_avg=0.000001 mcu_task_stddev=0.000002 bytes_write=9673 bytes_read=8102 bytes_retransmit=0 bytes_invalid=0 send_seq=438 receive_seq=438 retransmit_seq=0 srtt=0.002 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=199999932 host: mcu_awake=0.123 mcu_task_avg=0.000872 mcu_task_stddev=0.000945 bytes_write=2229 bytes_read=12003 bytes_retransmit=0 bytes_invalid=0 send_seq=246 receive_seq=246 retransmit_seq=0 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49998511 adj=49999158 heater_bed: target=0 temp=20.1 pwm=0.000  sysload=0.82 cputime=24.265 memavail=353912 print_time=134999.940 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.8 pwm=0.000
BD home_wait0
BD raise_probe
BD raise_probe
probe at -0.669,0.373 is z=-0.238555 bd0.700  pos_new:0.461445
pos -0.669,0.373,1.761 -0.500
BD lower_probe0
BD home_start
Stats 135055.6: gcodein=0  mcu: mcu_awake=0.000 mcu_task_avg=0.000001 mcu_task_stddev=0.000002 bytes_write=10300 bytes_read=8661 bytes_retransmit=0 bytes_invalid=0 send_seq=477 receive_seq=477 retransmit_seq=0 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=199999933 host: mcu_awake=0.123 mcu_task_avg=0.000872 mcu_task_stddev=0.000945 bytes_write=2602 bytes_read=12597 bytes_retransmit=0 bytes_invalid=0 send_seq=275 receive_seq=275 retransmit_seq=0 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49998712 adj=49999495 heater_bed: target=0 temp=20.1 pwm=0.000  sysload=0.82 cputime=24.393 memavail=353928 print_time=135000.954 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.6 pwm=0.000
BD home_wait0
BD raise_probe
BD raise_probe
probe at -0.669,0.373 is z=-0.234971 bd0.640  pos_new:0.405029
pos -0.669,0.373,1.765 -0.500
BD lower_probe0
BD home_start
Stats 135056.6: gcodein=0  mcu: mcu_awake=0.000 mcu_task_avg=0.000001 mcu_task_stddev=0.000002 bytes_write=11019 bytes_read=9244 bytes_retransmit=0 bytes_invalid=0 send_seq=518 receive_seq=518 retransmit_seq=0 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=199999944 host: mcu_awake=0.123 mcu_task_avg=0.000872 mcu_task_stddev=0.000945 bytes_write=2992 bytes_read=13244 bytes_retransmit=0 bytes_invalid=0 send_seq=306 receive_seq=306 retransmit_seq=0 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49998885 adj=49999876 heater_bed: target=0 temp=20.1 pwm=0.000  sysload=0.82 cputime=24.531 memavail=353928 print_time=135002.153 buffer_time=0.000 print_stall=0 extruder: target=0 temp=20.9 pwm=0.000
BD home_wait0
BD raise_probe
BD raise_probe
probe at -0.669,0.373 is z=-0.238850 bd0.600  pos_new:0.361150
pos -0.669,0.373,1.761 -0.500
BD lower_probe0
BD home_start
Stats 135057.6: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000003 mcu_task_stddev=0.000003 bytes_write=11795 bytes_read=9870 bytes_retransmit=0 bytes_invalid=0 send_seq=562 receive_seq=562 retransmit_seq=0 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=199999967 host: mcu_awake=0.123 mcu_task_avg=0.000872 mcu_task_stddev=0.000945 bytes_write=3382 bytes_read=13905 bytes_retransmit=0 bytes_invalid=0 send_seq=337 receive_seq=337 retransmit_seq=0 srtt=0.003 rttvar=0.002 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49998796 adj=50000205 heater_bed: target=0 temp=20.1 pwm=0.000  sysload=0.82 cputime=24.675 memavail=353928 print_time=135003.355 buffer_time=0.000 print_stall=0 extruder: target=0 temp=20.5 pwm=0.000
BD home_wait0
BD raise_probe
BD raise_probe
Stats 135058.7: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000003 mcu_task_stddev=0.000003 bytes_write=12395 bytes_read=10443 bytes_retransmit=0 bytes_invalid=0 send_seq=601 receive_seq=601 retransmit_seq=0 srtt=0.002 rttvar=0.002 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=199999970 host: mcu_awake=0.123 mcu_task_avg=0.000872 mcu_task_stddev=0.000945 bytes_write=3718 bytes_read=14527 bytes_retransmi [... cut ...] adj=50001326 heater_bed: target=0 temp=20.1 pwm=0.000  sysload=0.85 cputime=25.653 memavail=353952 print_time=135011.180 buffer_time=0.000 print_stall=0 extruder: target=0 temp=20.6 pwm=0.000
BD home_wait0
BD raise_probe
BD raise_probe
probe at -0.669,0.373 is z=-0.232434 bd0.650  pos_new:0.417566
get_position Z is 1.768 mm
.set_position Z is 0.650 mm
probe accuracy results: maximum -0.228703, minimum -0.242434, range 0.013732, average -0.234418, median -0.234374, standard deviation 0.004292
```

markniu, 2023-09-05:
I am also testing and debuging the delta printer with my stm32 board + raspberrypi, it just works, but not easy to calibrate.

gaby64, 2023-09-07:
Btw, I have optical endstops at the top of my delta printer, I will never home with the probe.
It is only for bed mesh.

gaby64, 2023-09-07:
[image]

I've had success running M102 S-6 calibration, DELTA_CALIBRATE and BED_MESH_CALIBRATE and then printing. But this is all still with TRSYNC_TIMEOUT at 0.040.

markniu, 2023-09-07:
> [image]
> 
> I've had success running M102 S-6 calibration, DELTA_CALIBRATE and BED_MESH_CALIBRATE and then printing. But this is all still with TRSYNC_TIMEOUT at 0.040.

looks nice. 
about the timeout, maybe you can try to add this code in the BD_sensor.c and compile & flash 
this will reduce the frequency while probing.
[image]

gaby64, 2023-09-10:
I was able to reduce TRSYNC_TIMEOUT to 0.030 with your changes.

gaby64, 2023-09-11:
> @gaby64 are you still using host for the i2c connection of the probe? If so, the timeout may be for you the round-trip between the pru and host communications.

Yes I am, can't run it on the pru¼3¼33398

gaby64, 2023-09-13:
@markniu 
What if i2c bus was supported? Could it allow reducing blocking in code?

goeland86, 2023-09-13:
> > @gaby64 are you still using host for the i2c connection of the probe? If so, the timeout may be for you the round-trip between the pru and host communications.
> 
> Yes I am, can't run it on the pru. How would you debug the timeout? Is it expected to have higher then 25ms roundtrip?

PRU communication goes through a rather slow memory buffer exchange from what I remember, so it's quite slow as a result. 25ms seems longer than it should be, but if you factor in that it's a single core, it wouldn't be all that surprising you'd get to that latency overall. Unfortunately the Beaglebone + Replicape combination is too limited in terms of computing capacity for what Klipper is doing. Redeem was a lot lighter, but also less feature-rich. The explosion of features in the newer Klipper code has outgrown what the Replicape can reasonably do with advanced hardware like the i2c BD sensor.