markniu/Bed_Distance_sensor #29: branche-name-change and mainsail-update-manager

koira, 2023-09-11:
The change of the branches-name from "master"to "new" apparently created some confusion within mainsail's update-manager which reports like shown below since a few hours:
[image]

streetgt, 2023-10-19:
I can confirm thatn, @markniu please create again the master branch, otherwise we can only pull the updates via ssh.

streetgt, 2023-10-19:
> The change of the branches-name from "master"to "new" apparently created some confusion within mainsail's update-manager which reports like shown below since a few hours: [image]
> 
> and this is my entry in moonraker.conf
> 
> [update_manager BDsensor] type = git_repo channel = dev path = ~/Bed_Distance_sensor origin = https://github.com/markniu/Bed_Distance_sensor.git install_script = ./klipper/install_BDsensor.sh is_system_service = False managed_services = klipper info_tags = desc=Bed Distance Sensor

If you want to update via UI, you have to add:

```
[update_manager BDsensor]
primary_branch: new 
```

Townie-au, 2023-11-11:
My listing in update manager says unknown and I have what you have above in the config.