markniu/Bed_Distance_sensor #30: how can i work z real time adjust on the printing。please help

whycch, 2023-09-13:


DarKingF0x, 2023-09-13:
Unfortunately for Klipper version, real time adjustment is not available.

whycch, 2023-09-14:
how fast i can read the distace data in a cycle. less than 1ms is it work? i still want to adaptive it work in klipper.

markniu, 2023-09-14:
recommend the cycle time is 10ms.

whycch, 2023-09-14:
are you have a technology communication group on QQ or wechat?

markniu, 2023-09-14:
> are you have a technology communication group on QQ or wechat or others application
thanks, I just create a QQ group: 616205664

zimmra, 2023-11-01:
> Unfortunately for Klipper version, real time adjustment is not available.

Is this a limitation of Klipper, or does the code here just not currently support it?

markniu, 2023-11-02:
there is some limitation of klipper, but maybe it's possible so I am resuming to develop the code in klipper for the real time adjustment. please do not use the klipper_beta for it is in development.

Eloura74, 2023-11-06:
can't wait to see it work, I received mine today

LAZER555, 2023-11-24:
> there is some limitation of klipper, but maybe it's possible so I am resuming to develop the code in klipper for the real time adjustment. please do not use the klipper_beta for it is in development.

Hi, there was an update to BDsensor, and it said the realtime lveling was in it. How can we use it? Or is it not working yet?

markniu, 2023-11-24:
1. update the files in ~/Bed_Distance_sensor/klipper_Beta;
2. go into the folder ~/Bed_Distance_sensor/klipper_Beta  and run` ./install_BDsensor.sh`.
3. compile and flash the klipper.bin firmware again;

Edit the start gcode in your slicer:
remove the auto bed leveling command `BED_MESH_CALIBRATE` and add `BDSENSOR_SET REAL_TIME_HEIGHT=`xx under the G28,
also z_tilt or gantry level is recommended. 
here xx is the first layer height that will do real time leveling, for example, set the first layer height to 0.32mm
```
G28
QUAD_GANTRY_LEVEL
G28
BDSENSOR_SET REAL_TIME_HEIGHT=0.32
```

note: this real time leveling works only with the BDsensor and Z motors are connected in the same MCU.

whycch, 2023-11-24:
woo,i Can't wait to test tonight

Townie-au, 2023-11-24:
> 1. update the files in ~/Bed_Distance_sensor/klipper_Beta;
> 2. go into the folder `cd ~/Bed_Distance_sensor/klipper_Beta`  and run` ./install_BDsensor.sh`.
> 3. compile and flash the klipper.bin firmware again;
> 
> Edit the start gcode in your slicer: remove the auto bed leveling command `BED_MESH_CALIBRATE` and add `BDSENSOR_SET REAL_TIME_HEIGHT=`xx under the G28, also z_tilt or gantry level is recommended. here xx is the first layer height that will do real time leveling, for example, set the first layer height to 0.32mm
> 
> ```
> G28
> QUAD_GANTRY_LEVEL
> G28
> BDSENSOR_SET REAL_TIME_HEIGHT=0.32
> ```
> 
> note: this real time leveling works only with the BDsensor and Z motors are connected in the same MCU.

So I have the BD sensor connected to the EBB42 and the Z steppers connected to SKR MINI E3 board. Different mcus so won't work?

koira, 2023-12-06:
> 
> note: this real time leveling works only with the BDsensor and Z motors are connected in the same MCU.

is this limitation for the time being only? will you be able to adapt to the many voron/canbus-iusers which mostly will have their  bdsensor connected to the external MCU while the z-steppers are controlled by an Octopus or thelike? any chance this will ever work?