markniu/Bed_Distance_sensor #33: M102 S-1 returns nothing

lennarz, 2023-11-05:
Hi All,

I'm using the sensor on an MKS SKIPR board with Klipper

I have a problem where the "M102 S-1" does not return any data.
the problem also occurs after a complete reinstallation. 
what could be the reason? 

Thanks.

markniu, 2023-11-05:
which pins do you used? the endstop pin on that SKIPR board can not be used for BD sensor,because there are capacitor on that pins which will filter the signal.

lennarz, 2023-11-05:
I have connected the sensor like this 
(Z-Endstop)-> SDA ->S GND ->G 
(3D Touch)-> CLK ->S 5V ->V GND ->G

lennarz, 2023-11-05:
and only // Unknown command:"M102" return

markniu, 2023-11-05:
> I have connected the sensor like this (Z-Endstop)-> SDA ->S GND ->G (3D Touch)-> CLK ->S 5V ->V GND ->G

please choose one of the pins in the EXP1 or EXP2 as the SDA.

lennarz, 2023-11-05:
I have a PIN from Exp1 in use now and still, in the console, when I enter the command 'M102 S-1', it says: // Unknown command: 'M102'.

lennarz, 2023-11-05:
I have a PIN from Exp1 in use now and still, in the console, when I enter the command M102 S-1, it says: // Unknown command: M102

lennarz, 2023-11-05:
I have a PIN from Exp1 in use now and still, in the console, when I enter the command M102 S-1, it says: // Unknown command: M102 #

markniu, 2023-11-05:
> I have a PIN from Exp1 in use now and still, in the console, when I enter the command M102 S-1, it says: // Unknown command: M102

Check if there is a file BDsensor.py in the ~/klipper/klippy/extras/, if not please install again by running cd  ~/Bed_Distance_sensor/klipper/ and ./install_BDsensor.sh

and also check again if there is [BDsensor] in your priner.cfg

lennarz, 2023-11-05:
the new instalation happen and here is the situation now, and the following errors are being displayed:

    M119 -> !! Bed Distance Sensor data error: 10.24
    M102 S-2 -> BDsensor: Connection Error"

closed 2023-11-08 by lennarz