markniu/Bed_Distance_sensor #34: BTT Octopus 1.1 M102 S-1 returns blank

nonsintetic, 2023-11-08:
I'm getting a blank response on M102 S-1. 

I'm using a BTT Octopus 1.1 board. The BDSensor is connected to the BL_Touch header. Here is my config:

```
[BDsensor] 
scl_pin:PB6
sda_pin:PB7
delay: 20 # 20us per pulse, this value should be >=20 but must be below 50
z_offset:0    # this `z_offset` must be set to 0. 
z_adjust:0.0  # z axis adjustment, replace the z_offset function. within -0.3 to 0.3mm
x_offset: 0
y_offset: 22
#no_stop_probe: # enable this for fast probe, the toolhead will not stop at the probe point.
position_endstop: 0.8 # the Z axis will stop at this position (mm) while homing z, recommend value is 0.4~1.0
speed:0.8 # this speed only works for the z tilt and PROBE_ACCURACY command.
```

I've verified the wires, they go to the correct pins on the board, I have 5V on the sensor's power pins when the printer is running.

markniu, 2023-11-08:
How about try other pins? like the free gpio pins in the EXP1 or EXP2 connector

nonsintetic, 2023-11-08:
I've tried the PD6 and PD5 pins on the raspberry pi header (tx and rx) and same issue. I will remove my screen and try some pins on the EXP header as well.

DarKingF0x, 2023-11-08:
In your setting have you try putting ^PB7 instead of PB7?

markniu, 2023-11-08:
you can also try to increase the delay value from 20 to 40

nonsintetic, 2023-11-08:
It worked with the 20ms delay. Here is the config for anyone that has similar issues on BTT Octopus 1.1:

```
[BDsensor] 
scl_pin:PD6
sda_pin:PD5
delay: 40 # 20us per pulse, this value should be >=20 but must be below 50
z_offset:0    # this `z_offset` must be set to 0. 
z_adjust:0.0  # z axis adjustment, replace the z_offset function. within -0.3 to 0.3mm
x_offset: 0
y_offset: 22
#no_stop_probe: # enable this for fast probe, the toolhead will not stop at the probe point.
position_endstop: 0.8 # the Z axis will stop at this position (mm) while homing z, recommend value is 0.4~1.0
speed:0.8 # this speed only works for the z tilt and PROBE_ACCURACY command.
```

Thank you @markniu

closed 2023-11-08 by nonsintetic