markniu/Bed_Distance_sensor #35: MKS SKIPR: BDsensor: Connection Error

lennarz, 2023-11-08:
The following errors are displayed  

M119 -> !! Bed Distance Sensor data error: 10.24
M102 S-2 -> BDsensor: Connection Error"

markniu, 2023-11-10:
it's probably that you connected the SDA wire to the endstop pin on SKPR board that has a filter capacitor on that pin.

lennarz, 2023-11-10:
I have connected the sensor like this
(EXP2)-> SDA ->PE8 GND ->G
(3D Touch)-> CLK ->S 5V ->V GND ->G

markniu, 2023-11-10:
that should be OK, and was the returned message after the command M102 S-1 correct?  if it's correct then check if the BDsensor been calibrated.

lennarz, 2023-11-10:
the message that comes back is empty

markniu, 2023-11-10:
How about change the (3D Touch) -> CLK  to other pin, like PE9 in the EXP2.  PE9-->CLK

markniu, 2023-11-10:
if still not working try to increase the delay value in BDsensor section from 20 do 30 or 40

lennarz, 2023-11-10:
unfortunately it still does not work

markniu, 2023-11-10:
was there a light on while a metal close to the coil? and was it very hot on the aluminum case? and please try other pins again in the EXP1 or EXP2.

lennarz, 2023-11-10:
yes the light is on and the aluminium case is cold. i try another pin on EXP1   (EXP1)-> SDA ->PD9 CLK ->PE11  GND ->G
(3D Touch)->  5V ->V GND ->G

lennarz, 2023-11-10:
but nothing happen

markniu, 2023-11-10:
have no idea, please send me an email about your order information, I will send a new one to you.

lennarz, 2023-11-10:
i have sent you an email via the store with the order no

Chris-Hirschberg, 2023-11-12:
i have exactly the same issue on my Btt octopus 1.1 with stm32 446. tried other pins, changed the delay, but no changes at all. led ist always on. when i try m102 s-1 the response is empty, at m102 s-2 it says bdsensor: connection error. with m119 he says sensor error 10.24. i don´t know whats happening.

mataksik, 2023-11-17:
similar problem on Manta m4p+cb1, 
when bdsensor conected on bltouch port

led ist always on
m102 s-1    response is empty
m102 s-2 response:  bdsensor: connection error
M119  response: Bed Distance Sensor data error:10.24

changed to 
CLK -> PA15 (EXP1)
GND -> GND  (EXP1)
SDA -> PA9 (EXP1)

finaly working

Chris-Hirschberg, 2023-11-17:
Will Check that and Report If it works, thank you. Regards.

Chris-Hirschberg, 2023-11-17:
no change. The sensor does the same thing (or not) as before.
everything stays the same...

vicsanhez, 2023-11-18:
HI, i had exactly the same problem. I put the SDA pin to GND and 1.22 in EXT2 connector.
Now it is running, will perform some testing and give feedback.
SCL pin is to servo pin as for bl touch.

Chris-Hirschberg, 2023-11-18:
OK. where exactly on exp2?
I don't understand 1.22. and sda to ground?!

I've attached a screenshot, maybe you can explain it to me in detail. Thanks.

mataksik, 2023-11-18:
I am realy confused

M102 S-1 = V1.1 pandapi3d.com
M102 S-2 = BDsensor:Out of measure Range or too close to the bed
M119 = x:open y:open z:open
M102 S-6 = Calibrating from 0.0mm to 3.9mm, don't power off the printer, Please Waiting... Calibrate Finished! You can send M102 S-5 to check the calibration data
M102 S-5 = 1015 BDSensor mounted too close or too high! 0.4mm to 2.4mm from BED at zero position is recommended, 
**Klipper state: Shutdown** Internal error on command:"M102" 

[stepper_z]
endstop_pin: probe:z_virtual_endstop # 

[BDsensor] 
scl_pin: PA15 
sda_pin: PA9      
delay: 200
z_offset:0 
z_adjust:0 
x_offset: -34
y_offset: 0
samples:3
position_endstop: 0.8 

Board Manta m4p+cb1
mcu(stm32g0b1xx)
Version: v0.12.0-8-g187cc2f1
Load: 0.01, Awake: 0.00, Freq: 64 MHz,
Version: v0.12.0-8-g187cc2f1-dirty

markniu, 2023-11-18:
you can move the nozzle a little higher, let the coil to the bed about 1.5mm, and send M102 S-6 again.

mataksik, 2023-11-18:
M102 S-6 goes up crashing into nozle, not detecting plate or metal until mesage "Calibrate Finished!"
M102 S-5 = Internal error on command:"M102" , Klipper state: Shutdown. BDSensor mounted too close or too high! 0.4mm to 2.4mm from BED at zero position is recommended. 1015

light always on, aluminum case cold.
defecive sensor?

Chris-Hirschberg, 2023-11-18:
My Sensor Shows also Always triggered.

M102 s-1 Shows nothing. M102 s-2 Shows once Connection Error and when i
unplug sda Always triggered and 0.00mm 🤷🏼‍♂️

Wrote Yue Mark, he'll send me a new one.

 I think i got a defective one. A friend of Mine has exactly the Same
printer, we built it sideways together. An His Sensor works neat.

markniu, 2023-11-18:
the sensor can not let the printer move, here we need to find out why the bed goes up into the nozle after M102 S-6? 
we have tested every sensor before it packaged

markniu, 2023-11-18:
[image]
from the code in klipper we can see it will let the z axis up 0.1mm at each time while calibration, and also you can menu move the z axis to see if the direction of z axis is right.

markniu, 2023-11-18:
How to check if the sensor is well? 
1, check if the sensor works well inside.  
     the led is flash quickly or off while there is no metal under the coil and the led on while there is a metal in its range.  if all that true that means the sensor inside works well. 
2, check if the connection to motherboard of printer works. 
   Return nothing after M102 S-1 that means the MCU on the printer cann't talk to sensor.

mataksik, 2023-11-18:
1. led is always on, no flashing 
2. M102 S-1 returns V1.1 pandapi3d.com
3. M102 S-2 returns  BDsensor:Out of measure Range or too close to the bed

4. M119 returns  x:open y:open z:open
5. M102 S-6 returns  Calibrating from 0.0mm to 3.9mm, don't power off the printer, Please Waiting... Calibrate Finished! You can send M102 S-5 to check the calibration data
6. M102 S-5 returns 1015 BDSensor mounted too close or too high! 0.4mm to 2.4mm from BED at zero position is recommended, Klipper state: Shutdown Internal error on command:"M102"

lennarz, 2023-11-18:
Hi 
I don't understand 1.22. and sda to ground?!

I've attached a screenshot, maybe you can explain it to me in detail. 

Thanks.

mataksik, 2023-11-18:
> check if the sensor works well inside.
> the led is flash quickly or off while there is no metal under the coil and the led on while there is a metal in its range. if all that true that means the sensor inside works well.

for my surprise after holding and little squeezing aluminium housing of sensor led light turned off and sensor started to react to melal for a moment. after some firmware changes again BDsensor led was always on. Squeezing helped again. traying to calibrate

lennarz, 2023-11-18:
it may be a broken cable

mataksik, 2023-11-18:
> it may be a broken cable

no indications of broken wires
for a small moment while BDsensor was connected, calibrated and homed once, tried second time it turned off, crashed nozzle into bed.  randomly turns on and off. Defective item.

markniu, 2023-11-19:
Have you try to move the nozzle a little higher and try calibrate again?if the coil too close or too high to the metal bed will let the data of calibration to 1015 that means out of range, and several users report this same to me and all are fixed by mount a little higher.

mataksik, 2023-11-19:
Dear Markniu
How can I calibrate if sensor not responding?
if it randomly starts working for few minutes in 20sec video led light is on. 
https://github.com/markniu/Bed_Distance_sensor/assets/93599081/25c59deb-eb93-4fdb-9da1-74c87612f7fc

markniu, 2023-11-19:
it seems that the distance of the coil to the bed is more than 4mm, if in that case it is normal, the coil is in the analog circuit not digital, like the antenna. although the signal is very weak, it can still detected something or just the noise. 
I think this led is causing confusion for users and I will change the design and let the led connect to mcu in the next version.

markniu, 2023-11-19:
you can put the coil close to the metal bed and then see if the led is still not stable?

mataksik, 2023-11-19:
sensor reacting to heat. for first it reacted to heat of fingers. i tried to heat with heat gun and it started to react. now it's stable for about 10 minutes

markniu, 2023-11-19:
> sensor reacting to heat. for first it reacted to heat of fingers. i tried to heat with heat gun and it started to react. now it was stable for about 10 minutes, cooled and stopped responding to metal
> 
>  heating.mp4

It may be that the black wire inside is loose or broken. You can contact the store's customer service.

closed 2023-11-23 by markniu

vicsanhez, 2023-11-24:
Hi sorry i did not receive any message about the thread.
#define  I2C_BD_SDA_PIN    P1_22 
[image]

markniu, 2023-11-25:
do you have multimeter?then you can check if there have pins in the connector is shorted to the aluminum case?