markniu/Bed_Distance_sensor #36: Homing Z - "Endstop z still triggered after retract"

KiwiChrist, 2023-11-10:
exEnder5+ converted to corexy motion system. Homing z has been working fine up to BDsensor install.
A BDsensor has been installed on my Ender3v2 without any issues at all.

I have current and up to date Klipper installed on RPI4.
Main MCU is a SKR2.
BDSensor is connected the my Huvud v0.61 toolhead board.

**BDSensor wiring connection:**
_Sensor --->  Huvud Board_
SDA ---->  SWDIO/PA14
Grnd ----> Grnd
CLK ---->  CLK/PA13
5v ----> 5v
Grnd ----> Grnd

**Printer.cfg**
Printer.cfg is set up exactly as described in the example configuration. 
I have set the BDsensor 1.65mm above the nozzle height using a PCB board under the sensor and with the nozzle just touchng the bed. X position is 0, Y position is 20mm.

Homing retraction distance is set to 12mm.

[bed_mesh], [screws_tilt_adjust] & [z_tilt] are configured and all "horizontal_move_z:" are set to 1.

**Checking the Position:**
M102 S1
Recv: V1.0 pandapi3d.com
M102 S2
0.2mm

**Calibration:**
z_offset is set to 0.
M102 S-6

M102 S-5
Raw data is from 116 rising to 1015.

_So far so good._

**Check with M119**
M119
Recv: x:open y:open z:open

_Bed is then raised until nozzle touches_

M119
Recv: x:open y:open z:TRIGGERED
Result is okay.

_I then home all axis and I then get:_
Endstop z still triggered after retract
.set_position Z is 0.700 mm

_I check z_enstop is triggering asgain using the "Endstops" in Mainsail and it is very inconsistant or very slow to trigger._
_Home Z retraction is then set to 20mm and it "Endstop z still triggered after retract" does not occur.

Have I got an very sensative BDsensor or am I missing something in my settings?_

Sample of my settings:
[Printer Configuration Snipit.txt](https://github.com/markniu/Bed_Distance_sensor/files/13314931/Printer.Configuration.Snipit.txt)

markniu, 2023-11-10:
you can try to slow down the second_homing_speed from 3 to 0.8.

KiwiChrist, 2023-11-10:
Thanks for the quick reply Mark,

_Previous setup:
I had the [BDsensor] located in my 'huvud_mcu.cfg' file and I had [safe_z_home] and [bed_mesh] located in 'homing.cfg' file.
I recomplied [BDsensor], [safe_z_home] and [bed_mesh] into printer.cfg._

I changed speed it to 0.8. re calibrated and still no joy and TBH, the nozzle was now crashing into the bed. Seemed to have gotten worse.

With that happening I went back to the beginning and reinstalled the BDsensor software/firmware, checked averything again, M102 S-1, M102 S-2,  re calibrated M102 S-6, M102 S-5 with raw data result from 215 to 1015, M119 z open, Lowered Z nozzle just shy of touching the bed and M119 z TRIGGERED. All boxes ticked (again).  Homed all and it worked.

Between slowing down the speed to 0.8, reinstalling software and recompling the location of [BDsensor], [safe_z_home] and [bed_mesh] it seems to be working now.

I do have a spare BDsensor just in case, so if this occurs again I might swap it out to compare.

I have tested homing z a dozen times for consistancy and all seems to be fine. 
[Home Z testing.txt](https://github.com/markniu/Bed_Distance_sensor/files/13323168/Home.Z.testing.txt)

Onto bed mesh now then z_Tilt or Screws_Adjust.

IMO - this little unit is still a game changing little piece of kit for bed leveling. 

**Great work and Thanks!!**

KiwiChrist, 2023-11-12:
### **UPDATE**
Refer to my other post that lead me to this... https://github.com/markniu/Bed_Distance_sensor/issues/38#issue-1989221697

After having a couple hours break form testing the accuracy of the BDsensor and doing the sanity check with the other printer, I decided to check the installation of the problematic BDsensor.  

Upon my initial installation of the problematic BDsensor I did notice a wire being visible in the epoxy, but never considered it to be a problem. Since removing the BDsensor and comparing it to the spare I purchased I can see this could be the culprit of not being accurate. Refer to the photo...

[image]

The (L) BDsensor in the photo is the one I had installed. Comparing it to (R) in the photo I do not think it is meant to look like that.  I think quality control may have missed this.

I have replaced (L) with (R) resulting in everyting working as it should.

Any thoughts?
Please advise.
Jason.

KiwiChrist, 2023-11-12:
### **All Problems Solved!**
Compare the BDsensors in this photo. The one with the wire showing in the epoxy was the problematic one.

**A day of testing and tweaking settings**
After taking a couple hours break from the testing, I concluded that the only thing it could be is a bad BDsensor as I did recall seeing that the two BDsenors looked different on the epoxy side opening the boxes, but did not think much of it. So I uninstalled the installed one to check it and sure enough a wire is visble in the epoxy. IMO not how it should be, possibly missed by QC.

**Sanity Check**
I did a PROBE_ACCURRACY  test from a known good BDsensor on my other printer.
Refer to my other post showing test results -  https://github.com/markniu/Bed_Distance_sensor/issues/38#issue-1989221697

**Remedy**
Now comparing the PROBE_ACCURRACY results from the known good to the faulty one and having a spare BDsesnor I purchased with the faulty one I installed it. Once the spare was installed I calibrated it following the instructions again and instantly the homing z, probe_accuracy and z_tilt all worked. The PROBE_ACCURRACY results were now how I expected them to be with no major fluctuations when probing.

Again refer to my other post showing test results -  https://github.com/markniu/Bed_Distance_sensor/issues/38#issue-1989221697

Hope this helps someone else having similar results.

[image]

**A big thank you to Mark for the quick reply to my email.**
Regards
Jason.

closed 2023-11-15 by KiwiChrist