markniu/Bed_Distance_sensor #37: Quad gantry level behavior

Galloz, 2023-11-10:
Printer: Voron 2.4 350mm

When doing QGL the head travels horizontally at 1mm distance (_horizontal_move_z:1_) from the bed.

This distance is too little especially in large format printers: when the z motors are not holding there could be a corner that looses 2 or 3mm. 
At the next QGL the nozzle will be dragged into the bed damaging it (near 3 corners I have quite deep scars in my PEI sheet) and potentially also bend the hotend/printhead assembly.

Setting higher _horizontal_move_z_ will cause QGL to fail.

At each corner of QGL the ideal behaviour would be for the head to move down, read the value, move up, and only then travel.
The same behavior as when the _[probe]_ feature is used.

markniu, 2023-11-11:
Hello, you can do the z-tilt before QGL.the BD sensor will work as normal probe while doing z tilt

Galloz, 2023-11-12:
Thanks fo the hint, I've implemented the z_tilt in the QGL macro, still have to fien tune the speeds.

```
[gcode_macro QUAD_GANTRY_LEVEL]
rename_existing: QUAD_GANTRY_LEVEL_1
gcode:
  Z_TILT_ADJUST
  G28 Z
  {% if printer["gcode_macro status_leveling"] != null %}
    status_leveling
  {% endif %}
    QUAD_GANTRY_LEVEL_1 { rawparams }
  {% if printer["gcode_macro status_ready"] != null %}
    status_ready
  {% endif %}
```

Might be helpful to highlight the z_tilt importance before QGL in the instructions

closed 2023-11-23 by Galloz

CanadianDreamer, 2024-02-14:
Could you share your 2.4 printer.cfg file. This could give me a starting point for my BDsensor project. Thanks