markniu/Bed_Distance_sensor #39: KAMP Adaptive_Meshing - Probe Position

KiwiChrist, 2023-11-13:
Previous to installing the mighty little BDsensor I use Adaptive_Meshing, where the probe will only probe/mesh the area where the part is positioned on the print bed. KAMP as I understand accommodated for the position of the probe relative to the nozzle.  
KAMP does not seem to do this anymore with BDsensor.

Obviously the easy fix is to not include KAMP/Adaptive_Meshing and just scan the whole bed but I would like adaptive meshing to work.

The probe is set relative to the nozzle X-38 Y-8. But I cannot see where the BED_MESH_CALIBRATE compensates for the probe XY offset anywhere.

Is this a BDsensor issue or something else?

markniu, 2023-11-13:
you can delete the no_stop_probe in the BDsensor

KiwiChrist, 2023-11-13:
No change when disabling no_stop_probe.
I have attached a sample of the printer.cfg to check if I have missed anything.

I assume [BDsensor] x_offset and y_offset are for the sensor position relative to the nozzle, as there no information for this on installation page. I can always set back to x0, y0 and see what happens.  ?

[Printercfg sample.txt](https://github.com/markniu/Bed_Distance_sensor/files/13340208/Printercfg.sample.txt)

KiwiChrist, 2023-11-14:
I completely reset stepper_x's and stepper_y's position_endstop:, position_min: and position_max: and then setting the probe position to home in the center of the bed. I then checked [BDsensor] x_offset and y_offset making sure the probe offsets are correct.

Aftyer doing all that KAMP Adaptive_meshing does it's own thing with resizing the bed msh area and probe offsets etc, therefore I do not thing KAMP works with [BDsensor].  

So I'm disabling KAMPs Adaptive_Meshing and just doing a full bed scan

markniu, 2023-12-20:
The BDsensor uses the bed mesh points which generated by klipper, so it should be work with KAMP.

KiwiChrist, 2023-12-23:
Via Klipper discord and with a whole day of testing yesterday I have come to the comclusion that the probe offsets xy in the [BDsensor] module are not being registered correctly and KAMP will never work properly.

Mark, can you please confirm that the probe offset is working and registereing for KAMP to correctly run.

The image below are the results of a whole day testing and adjusting settings yesterday. 
Comparing the Test 1 to Test 2 the probe column does not alter it's configuration with the x39.5 and x-39.5 settings. 
The result remains as (84.7,84.7) respectively down the column.

[image]

I beleive there is someting not implemented correctly in the BDsensor module.

Look forward to your reply.

markniu, 2023-12-25:
> rs to the left of the bed center
This is normal, the points of the (84.7,84.7) is depends on the printing module. you can check it by switching it to other probe sensor.

markniu, 2023-12-25:
> Via Klipper discord and with a whole day of testing yesterday I have come to the comclusion that the probe offsets xy in the [BDsensor] module are not being registered correctly and KAMP will never work properly.
> 
> The probing always occurs to the left of the bed center.
> 
> Mark, can you please confirm that the probe offset is working and registereing for KAMP to correctly run.
> 
> The image below are the results of a whole day testing and adjusting settings yesterday. Comparing the Test 1 to Test 2 the probe column does not alter it's configuration with the x39.5 and x-39.5 settings. The result remains as (84.7,84.7) respectively down the column with no difference.
> 
> Clearly the tool adjusted values do change from Test1 to Test 2, but the Probe column remains the same. Something is not being calculated correctly somewhere.
> 
> <img alt="image" width="389" src="https://private-user-images.githubusercontent.com/20054687/292647135-c7f7335d-ec51-4ff7-8bfc-ab75f8bed98a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM0OTAzOTMsIm5iZiI6MTcwMzQ5MDA5MywicGF0aCI6Ii8yMDA1NDY4Ny8yOTI2NDcxMzUtYzdmNzMzNWQtZWM1MS00ZmY3LThiZmMtYWI3NWY4YmVkOThhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI1VDA3NDEzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ1YTlmZTJkNTlhMzNjMGRhMWE3MGM5NTljZmM4M2EzMWExMzY4M2UwMzczYjA3MGIxMjM3Y2ZmOTYxOTMwMTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.NrSRIxgjLLH4XBkqGszbgGvO3GxiFJAsI4fybEWHyEg">
> I beleive there is someting not implemented correctly in the BDsensor module.
> 
> Look forward to your reply.

This is normal, the points of the (84.7,84.7) is depends on the printing module. you can check it by switching it to other probe sensor.

markniu, 2023-12-25:
The bed mesh points are the same as the original klipper, the BDsensor  has not changed any of that.

KiwiChrist, 2023-12-28:
Shouldn't the probe be scanning in the position of the part and not to one side?

The part has been positioned centrally in the slicer (relative to 0,0), therefore I would expect the probe to scan the center of the bed where the part acturally is.  The image below is a representation of what is happening.

Euclid probe accounts for the probe x & y offset relative to the nozzle and probes where the part is positioned on the bed.

[image]

What am I missing?

markniu, 2023-12-30:
Thanks! I check again and find that there is an option in the code that should be enabled for using XY offset.
please update the BDsensor.py

KiwiChrist, 2023-12-30:
PERFECT! 
I can confirm that the BDsensor probe offset XY is now being implemented correctly and Adaptive Mesh is now working as expected now.

This is now IMO the perfect little non contact probing device.
Great work!

closed 2023-12-30 by KiwiChrist