markniu/Bed_Distance_sensor #40: Design suggestion/request: rp2040 has no hardpoints to mount it, sensor cable fixed? sensor itself only has 1 mounting hole

tastyratz, 2023-11-13:
Looking at this unit I see a few design considerations missing:

1. the sensor itself only attaches with a single screw. This doesn't stop the whole thing from rotating or moving around and puts a lot of strain on just 1 shcs. 2 screws would prevent that.
2.  The rp2040 case has no "ears" for mounting in any way. The only way to fix it to a print head is to sleeve it or glue it or do something weird. Why not give it some attachment points? 
3. There is no STL sleeve for the rp2040 to attach it in any way, only the inductive sensor
4. The rp2040 is going to struggle in any hot print environment and adds weight. Why put it in the chamber at all? Why not allow the inductive sensor to connect via a pin so it can be mounted in the electronics bay? Does that change the characteristics of the readout to extend that wire? Would it need to be shielded or have issues traveling down the breakout board pin to be reattached in the bay?

Just a few thoughts I had here

markniu, 2023-11-18:
Thank you for your great advice. 
this BDsensor comes with the idea of real time leveling that needs to install the sensor closer to the nozzle, so the size should be small and simple mount for most printers. and we will consider the 2 screws for the next version.
here the MCU is not rp2040 because that chip size is too big to insert into the case.  
we can extend the wire between coil and mcu but needs to find a stable small connector and wire like you said need shield and without issues for traveling nozzle.

KiwiChrist, 2023-11-19:
IMO two screws is not required. Simply model a rib at least 1/3 around the sensor and it will be keyed in place. The sensor will not rotate and if the sensor has an impact it can break free.
[image]

tastyratz, 2023-11-23:
Even a 2 pin JSTXH connector will probably be fine as long as a shielded wire is specified and you could run the extra pin to ground for the shielding, it depends on just how sensitive it is in practice

The MCU might not be 2040 but it's still an attachment without any holes or mounting hard points to put anywhere. It doesn't even have a flat it's the worst possible shape. In any type of high temperature or enclosed environment, it would be a huge problem. Maybe that's a feature more than a bug to prevent using this in scenarios where it's unlikely to be accurate.