markniu/Bed_Distance_sensor #41: Homing Shuts Down Klipper

ceb0610, 2023-11-13:
When I send the G28 command, the X-axis homes fine and the Y-axis homes fine. When my printer tries to home the Z-axis, I get the message "Homing failed due to printer shutdown." This is after a fresh install of klipper and the BDsensor script. Any help would be appreciated.

[klippy (1).log](https://github.com/markniu/Bed_Distance_sensor/files/13342846/klippy.1.log)

markniu, 2023-11-14:
could you try to set the z_adjust value in the BDsensor section to zero? and also delete the probe section in printer.cfg.

ceb0610, 2023-11-14:
That worked. I still need to adjust Z a bit. Is (-) down and (+) up?

KiwiChrist, 2023-11-14:
When you calibrate the BDsensor make sure the nozzel in touching the bed... this sets the BDsensor z to 0.  [BDsensor] z_offset is remains at 0.