markniu/Bed_Distance_sensor #42: Repo is dirty

LAZER555, 2023-11-14:
Hi, i just installed bd sensor on klipper. And there's an error in an uplate section. What if i update klipper, will the bd sensor stop working?
[image]

closed 2023-11-23 by LAZER555

Trash-Face, 2024-02-07:
What did you do to fix this? I'm having this issue now and i think its causing a random disconnect of the mcu.

Trash-Face, 2024-02-07:
If anyone else is having this issue, you need to remake the klipper.bin after you have installed the BDsensor files. Re-Flash the mcu and reboot everything. 

If you want to use the newest version of klipper, you will need to update the host klipper install and then reinstall the BDsensor files before you remake the .bi