markniu/Bed_Distance_sensor #43: Nozzle keeps crashin into bed

LAZER555, 2023-11-19:
Nozzle keeps crashin into bed. I calibrated it many times. M119 is triggering. But when i home all axis it always crashes. I do it with heated bed. Should i do it with cold bed? 
Sensor worked a few days, then it started crashing. 

[BDsensor] 
sda_pin:PA1  
scl_pin:PB0 
delay: 20 # 20us per pulse, this value should be >=20 but must be below 50
z_offset:0 # this `z_offset` must be set to 0. 
z_adjust: -0.0 # z axis adjustment, replace the z_offset function. within -0.3 to 0.3mm
x_offset: -28
y_offset: 0
no_stop_probe: # enable this for fast probe, the toolhead will not stop at the probe point.
position_endstop: 1.0 # the Z axis will stop at this position (mm) while homing z, recommend value is 0.4~1.0
speed:0.8 # this speed only works for the z tilt and PROBE_ACCURACY command.

KiwiChrist, 2023-11-19:
I'd check the probe. I had a similar problem, inconsistant probe accuracy. Calibrated and triggered fine, but would play up when in actual use whether the bed was hot or cold.

https://github.com/markniu/Bed_Distance_sensor/issues/38#issue-1989221697
https://github.com/markniu/Bed_Distance_sensor/issues/36#issuecomment-1807065889

markniu, 2023-11-20:
could you try to slow down the speed of homing z  in the stepper_z section? like this
```
[stepper_z]
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0.5
homing_speed: 8
second_homing_speed: 0.8
```

LAZER555, 2023-11-20:
I put 4 in the homing speed, 8 is too fast. For now it works :) Thank you

LAZER555, 2023-11-20:
Ok it crashed again, when i launched the print before homing. The bed was hot, the homing speed was 4. And it crashed before starting the print

LAZER555, 2023-11-21:
I calibrated it with the bed at 80 and it's working for now :P

koira, 2023-12-06:
@markniu just out of curiosity: shouldn't it be totally unrelated to the bed being hot or cold after all? I mean the bed expands when it heats up, which should be recognized buy the sensor, or not?

markniu, 2023-12-07:
> @markniu just out of curiosity: shouldn't it be totally unrelated to the bed being hot or cold after all? I mean the bed expands when it heats up, which should be recognized buy the sensor, or not?

there is temperature drift for inductive sensor if it works higher than 80c but calibrated in 25C. 
Here is a test with BDsensor: https://www.youtube.com/watch?v=bQysyx_jPp0&t=93s
I calibrate the sensor while the bed is about 20C, and then find that the temperature drift after heating the bed around 85C, but nearly no drift below that temperature.

Squirrelf, 2023-12-18:
Happened to me as well, homed, it works (kinda, seem stop stop right were the nozzle is), home again and that time full on crash doesn't stop digging into the bed until emergency shutdown. 
totally unreliable. Cold bed, just homing one after the other.