markniu/Bed_Distance_sensor #44: I think faulty BDSensor

almizi, 2023-11-21:
So testing all the calibration several (what feels like several dozen times), I have had 0 luck with it getting any sort of accurate inaccuracies, consistent failures, or any blatant things sticking out.

The light on the device lights up, it tells the printer that it is reading there is something within range.

I only have the individual plugs not one big plug. I don't have the 5-wire configuration like I've seen in pictures.

I don't know if I just got one faulty device or if there is just some sort of bug that hasn't been fixed yet or something, but it hasn't worked even before I updated the firmware.

I am using an Ender 3 Pro V2
BTT SKR Mini E3 V3
Marlin bugfix-2.1.x

markniu, 2023-11-23:
does the version command M102 S-1 work? and which pins did you used for the BDsensor?

markniu, 2023-11-23:
it seems that there is a problem between the printer board to the sensor. need to check the config and the hardware connection.
this is a very simple PCBA inside that is made by the biggest PCB company in China, and they produced so many PCBA for many companies. and also we flash the firmware and then test this BDsensor before package, so the fault hardware is very low.

almizi, 2023-11-25:
> it seems that there is a problem between the printer board to the sensor. need to check the config and the hardware connection. this is a very simple PCBA inside that is made by the biggest PCB company in China, and they produced so many PCBA for many companies. and also we flash the firmware and then test this BDsensor before package, the probability of fault hardware is very low.

I have the BD sensor on pins:
SDA - PD5
CLK - PD4
Black - BL Touch Ground
RED -  BL Touch Power
[https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/hardware/BTT%20SKR%20MINI%20E3%20V3.0/Hardware](url)

> does the version command M102 S-1 work? and which pins did you used for the BDsensor?

Yes, this does work!

When I run M102 S-6 I got this:
```Send: M102 S-6
[...]
Recv: ok
Recv: c_z0:0.00-0.00
Recv: BD Sensor Calibrating...
Recv: G1Z0.0, Z:0.00
Recv: w:0, Z:0.00
[...]
Recv: G1Z0.1, Z:0.10
Recv: w:1, Z:0.10
Recv: G1Z0.2, Z:0.20
Recv: w:2, Z:0.20
[...]
Recv: G1Z0.3, Z:0.30
Recv: w:3, Z:0.30
[...]
Recv: G1Z0.4, Z:0.40
Recv: w:4, Z:0.40
[...]
Recv: G1Z0.5, Z:0.50
Recv: w:5, Z:0.50
Recv: G1Z0.6, Z:0.60
Recv: w:6, Z:0.60
[...]
Recv: G1Z0.7, Z:0.70
Recv: w:7, Z:0.70
Recv: G1Z0.8, Z:0.80
Recv: w:8, Z:0.80
[...]
Recv: G1Z0.9, Z:0.90
Recv: w:9, Z:0.90
[...]
Recv: G1Z1.0, Z:1.00
Recv: w:10, Z:1.00
[...]
Recv: G1Z1.1, Z:1.10
Recv: w:11, Z:1.10
[...]
Recv: G1Z1.2, Z:1.20
Recv: w:12, Z:1.20
[...]
Recv: G1Z1.3, Z:1.30
Recv: w:13, Z:1.30
Recv: G1Z1.4, Z:1.40
Recv: w:14, Z:1.40
[...]
Recv: G1Z1.5, Z:1.50
Recv: w:15, Z:1.50
[...]
Recv: G1Z1.6, Z:1.60
Recv: w:16, Z:1.60
[...]
Recv: G1Z1.7, Z:1.70
Recv: w:17, Z:1.70
Recv: G1Z1.8, Z:1.80
Recv: w:18, Z:1.80
[...]
Recv: G1Z1.9, Z:1.90
Recv: w:19, Z:1.90
Recv: G1Z2.0, Z:2.00
Recv: w:20, Z:2.00
[...]
Recv: G1Z2.1, Z:2.10
Recv: w:21, Z:2.10
[...]
Recv: G1Z2.2, Z:2.20
Recv: w:22, Z:2.20
[...]
Recv: G1Z2.3, Z:2.30
Recv: w:23, Z:2.30
Recv: G1Z2.4, Z:2.40
Recv: w:24, Z:2.40
[...]
Recv: G1Z2.5, Z:2.50
Recv: w:25, Z:2.50
[...]
Recv: G1Z2.6, Z:2.60
Recv: w:26, Z:2.60
[...]
Recv: G1Z2.7, Z:2.70
Recv: w:27, Z:2.70
[...]
Recv: G1Z2.8, Z:2.80
Recv: w:28, Z:2.80
[...]
Recv: G1Z2.9, Z:2.90
Recv: w:29, Z:2.90
Recv: G1Z3.0, Z:3.00
Recv: w:30, Z:3.00
[...]
Recv: G1Z3.1, Z:3.10
Recv: w:31, Z:3.10
Recv: G1Z3.2, Z:3.20
Recv: w:32, Z:3.20
[...]
Recv: G1Z3.3, Z:3.30
Recv: w:33, Z:3.30
[...]
Recv: G1Z3.4, Z:3.40
Recv: w:34, Z:3.40
[...]
Recv: G1Z3.5, Z:3.50
Recv: w:35, Z:3.50
Recv: G1Z3.6, Z:3.60
Recv: w:36, Z:3.60
[...]
Recv: G1Z3.7, Z:3.70
Recv: w:37, Z:3.70
Recv: G1Z3.8, Z:3.80
Recv: w:38, Z:3.80
[...]
Recv: G1Z3.9, Z:3.90
Recv: w:39, Z:3.90
[...]
Recv: BD Sensor calibrated.
[...]
Send: M102 S-5
Recv: ok
Recv: Calibrate data:0,210
Recv: Read Error.
Recv: Calibrate data:1,211
Recv: Calibrate data:2,361
Recv: Calibrate data:3,391
Recv: Calibrate data:4,420
Recv: Read Error.
Recv: Calibrate data:5,447
Recv: Read Error.
Recv: Calibrate data:6,473
Recv: Read Error.
Recv: Calibrate data:7,626
Recv: Calibrate data:8,652
Recv: Read Error.
Recv: Calibrate data:9,677
Recv: Calibrate data:10,700
Recv: Read Error.
Recv: Calibrate data:11,723
Recv: Read Error.
Recv: Calibrate data:12,872
Recv: Calibrate data:13,894
Recv: Read Error.
Recv: Calibrate data:14,915
Recv: Read Error.
Recv: Calibrate data:15,935
Recv: Calibrate data:16,954
Recv: Calibrate data:17,972
Recv: Read Error.
Recv: Calibrate data:18,980
Recv: Read Error.
Recv: Calibrate data:19,980
Recv: Read Error.
Recv: Calibrate data:20,1015
Recv: Calibrate data:21,1015
Recv: Calibrate data:22,1015
Recv: Calibrate data:23,1015
Recv: Calibrate data:24,1015
Recv: Calibrate data:25,1015
Recv: Calibrate data:26,1015
Recv: Calibrate data:27,1015
Recv: Calibrate data:28,1015
Recv: Calibrate data:29,1015
Recv: Calibrate data:30,1015
Recv: Calibrate data:31,1015
Recv: Calibrate data:32,1015
Recv: Calibrate data:33,1015
Recv: Calibrate data:34,1015
Recv: Calibrate data:35,1015
Recv: Calibrate data:36,1015
Recv: Calibrate data:37,1015
Recv: Calibrate data:38,1015
Recv: Calibrate data:39,1015```

LAZER555, 2023-11-25:
[image]
I got back to inductive probe. I'm tired of this bd sensor not working normally. To many inaccuracies. I tried this realtime feature, still the same results.

closed 2023-11-28 by markniu