markniu/Bed_Distance_sensor #46: Unable to connect with ebb36

terri1769, 2023-11-22:
I have an EBB36 CAN card communicating properly but so far the BDsensor won't connect.  I've triple checked the wiring against what little I can find online about it.

Console:
10:30 AM
BDsensor:Connection Error
10:30 AM
M102 S-2
10:30 AM

10:30 AM
M102 S-1

Associated logs and config files attached
[BDsensor.zip](https://github.com/markniu/Bed_Distance_sensor/files/13441458/BDsensor.zip)

markniu, 2023-11-23:
Hello, what version is your EBB36?
[image]

terri1769, 2023-11-23:
1.1.  I also confirmed the light works when I bring metal near.

markniu, 2023-11-23:
Are you sure the version is EBB36 CAN V1.1 (STM32G0B1) ?  from the klippy.log file I see you use the PA4 and PA5 for the BDsensor but there is for the PT100 and also no connector for the that pins from the sch file:
[image]

terri1769, 2023-11-23:
Thank you!  I have no idea where I got PA4 and PA5.  I changed those to PB8 and PB9 and it works.  I apologize for taking your time with this but thank you for your support!

terri1769, 2023-11-25:
Well... It worked for about 24 hours and is now back to not communicating.  The BDsensor config hasn't changed so I'm trying to track down any other changes I made.  Is there any kind of debug that might help?

markniu, 2023-11-25:
yes, you can send me your the klippy.log file. niujl123@sina.com .also do you have multimeter?then you can check if there have pins in the connector is shorted to the aluminum case?

terri1769, 2023-11-25:
Thanks.  I metered the connector pins and none are shorted to the case.  I also checked end to end on the cable to make sure each conductor was intact and none shorted.  I'll send the file momentarily.

closed 2023-12-07 by markniu

terri1769, 2023-12-17:
Hi Mark.  I finally got around to installing the new sensor.  Still don't
get a result when I do M102 S-1.  Here's a new klippy log.  Any suggestions?

markniu, 2023-12-19:
> Hi Mark. I finally got around to installing the new sensor. Still don't get a result when I do M102 S-1. Here's a new klippy log. Any suggestions?
> […](#)

you can try to connect the bdsensor to the I2C connector.  
[image]

terri1769, 2023-12-19:
OK, tried it.  Moved it to the I2C connector, which I know was working as I
had something else hooked to it that was working.  Changed SDA and CLK to
PB3 and PB4.  Same issue.

I have another on the way that I was hoping to use for another printer.
Guess I'll try it when it gets here.  Is there anything else I can test?  I
do have oscilloscope and protocol analyzer if they'd help.

Terri

markniu, 2023-12-20:
> OK, tried it. Moved it to the I2C connector, which I know was working as I had something else hooked to it that was working. Changed SDA and CLK to PB3 and PB4. Same issue. I have another on the way [...]

The oscilloscope will be useful, there will be signal on the clk and sda pin after sending M102 command. for example, M102 S-1 or M102 S-5, you can check if there is signal on the clk and sda pin with oscilloscope..

terri1769, 2023-12-20:
I'll scope it as soon as I can.  Does it use actual I2C protocol or
something else?

terri1769, 2023-12-20:
OK, hooked up the oscilloscope to the CAN board without the probe hooked
up.  Got a pulse on SDA and a train on CLK as I kinda expected.  Then I
hooked the probe back up and the thing suddenly works.  I didn't change any
of the wiring so I have no idea what fixed it.  It seems to be fixed now
though.

Thank you so much for your time and assistance.  Support like this is why I
like to support the small companies.  I now have 2 Panda Zhu boards and 2
BDsensors (plus the one that never would work).  I sure would love it if
Klipper would start supporting the ESP32 platform but I know that's out of
your hands.  Anyway, thanks again from a happy customer.

Terri

ptegler, 2023-12-20:
t's not just esp32....it's the non-standard I2C protocol bd is using that makes
it incompatible directly in klipper's main channel.  My sensor remains in the box,
bought over a year ago...saw al lthe org conversations that made it non-K compat.
It may one day go on one of my reprap printers, but I gave up eons back on this ever
properly fully working in Klipper.    :-(
ptegler

terri1769, 2023-12-20:
The ESP32 was for the PandaZhu boards  :)