markniu/Bed_Distance_sensor #47: Error in compiling firmware klipper_Beta

Galloz, 2023-11-23:
Wanted to try the beta for real time leveling with klipper but the firmware doesn't compile.
Followed the instructions changing only the folder from where running the script:
```
cd  ~/Bed_Distance_sensor/klipper_Beta/
./install_BDsensor.sh
```

But compiling the firmware throws an error:
```
  Building out/compile_time_request.o
Too many message ids
make: *** [Makefile:83: out/compile_time_request.o] Error 255
```

Doing the same for the main version of BDsensor and the firmware compiles without issues.

Board: BTT Octopus v1.0, STM32F446

markniu, 2023-11-24:
Hello,  this is problem in klipper for that there is a few ids left now. the solution is disable some unused features, 
 nano ~/klipper/src/Makefile , for example if you do not use the LCD12864 you can disable it like this. 
[image]

Galloz, 2023-11-24:
Thanks for the answer, it worked!

I think I'm going back to the main version, w
ith the beta the nozzle keeps crushing in the bed, sometimes at homing but mainly when doing z_tilt. 
In case it might help here is klipper's log.
[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/13461001/klippy.log)

Galloz, 2023-11-28:
Did that and at first both homing and QGL were fine.
Sent to print a simple test cube but the nozzle crashed again in the bed when doing z_tilt before QGL.

At 0,8mm/s it really takes a long time to start a print, any chance that the lift speed after sensing the bed might be at an higher speed?
[klippy 20231128.log](https://github.com/markniu/Bed_Distance_sensor/files/13491459/klippy.20231128.log)

closed 2023-11-30 by Galloz