markniu/Bed_Distance_sensor #48: Auto Leveling (G29) with Marlin

adsaxton, 2023-11-25:
I designed my own mount for the BDSensor, locating it 35.5mm behind and about 1.2mm left of the nozzle. My printers been adjusted to home X to -35. The BDS calibrated and homes fine, but I haven't figured out how to tram the bed yet. When I try to auto-level with G29, it scans the 1st and 2nd lines and then freezes Marlin. I have to cycle the power to get the printer back. I'm at a loss since no error is reported.

markniu, 2023-11-28:
did you try to comment out the BD_SENSOR_PROBE_NO_STOP ?  or please send me your configure files to me niujl123@sina.com

adsaxton, 2023-11-28:
No. I did not (yet). I'll try that today. Here are my config files: 
[Configs.zip](https://github.com/markniu/Bed_Distance_sensor/files/13490089/Configs.zip)

Ahuffins, 2023-12-15:
> I found the problem! It appears there is a bug in the current version of Marlin-bugfix-2.1.x (02010300) where it uses Z_STOP_PIN, rather than Z_MIN_PROBE_PIN in some parts of the leveling code, including "Real Time Leveling". Changing Z_STOP_PIN to the same as Z_MIN_PROBE_PIN resolved the issues.

I am experiencing the same problem and was wondering where exactly do I make this change

adsaxton, 2023-12-15:
> I am experiencing the same problem and was wondering where exactly do I make this change

It's in the PIN file. For my BTT SKR Mini E3:

`#define Z_STOP_PIN                          PC14  // PC2   // Z-STOP`

ibrahimhakkicandan, 2024-01-02:
im experiencing the same issue. it goes all the way to the edge and freezes. my printer ender 3 s1 pro and in the pin files there is nothing related even 'Z'.