markniu/Bed_Distance_sensor #49: BD sensor with Voron Tap

MJ0423, 2023-12-02:
I have a Voron 2.4 with klipper and I want to use BD sensor with Voron Tap. Actually use them together well.
But I think it would be better "QGL(or Z tilt)" and "Bed mesh"were separated. 
Voron Tap use for Z endstop and QGL
BD Sensor use for Bed Mesh
Is it posslble?
or Any chance of an update?

markniu, 2023-12-05:
I am adding an endstop_pin option in the BDsensor section that will allow the normal switch sensor to do homing ,Z tilt or QGL, the BD sensor is only used for bed mesh.

Reubertt, 2023-12-30:
Is this function not available yet or were you talking about Z endstop PIN? Because I would like to implement it on my printer but I don't want to mess up my configuration.

markniu, 2024-01-03:
I have added this function into the code, but not test much. just add `endstop_pin` in the section` [BDsensor]`. e.g.

`endstop_pin:PA7`

and note that the endstop reading command M119 doesn't work in this case.