markniu/Bed_Distance_sensor #50: Unknown pin chip name 'probe'

zarpyj, 2023-12-06:
Hello Markniu, hope you are fine.
I come to you, I have some issue with the installation of my BDSensor.
I have one VZBot 330 printer, controlled by one BTT Octopus Pro V1.0 (F429); I have also installed one CAN Bus on my printhead, one BTT EBB36 V1.2.
I have followed all the steps you descibed in your github and also the video of Kris Riley (https://www.youtube.com/watch?v=VDOYYskbxV8). I also then modified the Printer.cfg accordingly and now each time I start or restart my printer, I get the following message : 
Unknown pin chip name 'probe'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
I am a bit lost to be honest, I have try to recompile the firmawre multiple time now and no change.
[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/13588550/klippy.log)
[moonraker.log](https://github.com/markniu/Bed_Distance_sensor/files/13588551/moonraker.log)
[printer.log](https://github.com/markniu/Bed_Distance_sensor/files/13588574/printer.log)
Printer.log is the printer.cfg file.

Any help will be really appreciated.
Thank you in advance

Jean-Jacques

markniu, 2023-12-08:
right, the firmware of the MCU which the BDsensor connected need to compile and flash again after you run the BDsensor install command:`install_BDsensor.sh`
the error of Unknown pin chip name 'probe' means the software of klipper has not load the BDsensor.py successful, that's the reason of checking if the file BDsensor.py is in the klipper folder:`~/klipper/klippy/extras/`

zarpyj, 2023-12-09:
Hello Markniu, hope you are fine.
Finally, my BDSensor is working as it should, I have recompiled my CAN system accordingly and it is now fully operational.
Thank you for your support.
Jean-Jacques

zarpyj, 2023-12-10:
Hello Markniu, hope you are fine.
I reopened this ticket because I think I found a problem when searching for updates in Fluidd.
For example, yesterday I was able to make the first settings of the printer, adjust the Z_Offset, put the printer at the origins, etc. etc.
This morning, when I wanted to continue these fine adjustments to my printer, I first checked the updates and following the search I got the following message for Klipper: Repo is dirty. Detected the following modified files: ['scripts/buildcommands.py', 'src/Makefile'] , and the Unknown pin chip name 'probe' error message was displayed again.
I looked in your documentation, under FAQ, as you mention if the BDSensor.py file in the ~/klipper/klippy/extras/ directory.
He was no longer there; I then relaunched the installation script, then checked with WinSCP if the file was present, it was.
I then restarted my printer and the Unknown pin chip name 'probe' error message was displayed again.
I believe that every time we search for updates in Fluidd, or even in Mainsail, I haven't tested, the system checks the directories for files that should be there or not and deletes the files that for him should not be there; therefore the BDsensor.py file is systematically deleted, which causes these errors.
Do you know of a way to fix this problem, while regularly checking the installed packages for updates?
Thank you in advance and good day.
Jean-Jacques

markniu, 2023-12-12:
> Hello Markniu, hope you are fine. I reopened this ticket because I think I found a problem when searching for updates in Fluidd. For example, yesterday I was able to make the first settings of the [...]

right, the update of fluidd or klipper will remove the BDsensor.py file in the klipper folder. and then we need to run again `~/Bed_Distance_sensor/klipper/install_BDsensor.sh` , I think there is a way to do that, but I am not sure and have not test yet, for example add `install_script: ~/Bed_Distance_sensor/klipper/install_BDsensor.sh` in the fluidd senction` [update_manager fluidd] ` in the moonraker.conf file.

zarpyj, 2023-12-12:
If I check for updates in klipper, I get this message
Repo has untracked source files: ['klippy/extras/gcode_shell_command.py'] 

[image]

markniu, 2023-12-13:
another way is to update the BDsensor in the webpage after you update the klipper in the webpage. 
[image]

zarpyj, 2023-12-13:
Sorry to tell you this, your work around is not working at all, I think you have to check how you declare your files and script to ensure that each time you check for updates in the settings, your files are not deleted.
the leveling sensor works really well but the only annoying thing is this problem of files being deleted every time you search for updates.
I hope you will find a solution soon, I really would like to use your sensor.
Thank you in advance
Jean-Jacques

zarpyj, 2023-12-15:
Hello Markniu, hope you are fine,
I found in the Frix-x repo (https://github.com/Frix-x/klippain-shaketune) is installation script file to install its different scripts and functions. I made a copy of it and tried to adapt it for installing the BDSensor scripts and functions. I am attaching it to this message. Perhaps this will solve the problems encountered after searching for updates in Fluidd or Mainsail.
Sincerely
Jean-Jacques
[BDSensor_install.txt](https://github.com/markniu/Bed_Distance_sensor/files/13683206/BDSensor_install.txt)
Rename BDSensor_install.txt to BDSensor_install.sh

markniu, 2023-12-15:
Thanks! I also try to test that, but then I find that the latest moonraker has depricated this `install_script`
[image]

https://moonraker.readthedocs.io/en/latest/configuration/

zarpyj, 2023-12-15:
OK, thank you for your answer.
If you agree, I will ask Frix-x if he agree to put one eye on this "new" install script and try to debug the install.
Best regards
Jean-Jacques

markniu, 2023-12-15:
> OK, thank you for your answer. If you agree, I will ask Frix-x if he agree to put one eye on this "new" install script and try to debug the install. Best regards Jean-Jacques

That's great if anyone can help to fix this.

zarpyj, 2023-12-15:
will try

zarpyj, 2023-12-20:
Hello Mark, hope you are fine.
Will you be reachable via discord or skype or any other means, would it be simpler to be able to talk to each other directly rather than through this conversation channel?
It is about the way we install your nice bed levelling sensor.
Jean-Jacques

markniu, 2023-12-20:
I am usually online on facebook:
https://www.facebook.com/yue.li.1610092/
https://www.facebook.com/groups/380795976169477

NemisizZ, 2023-12-20:
HI
So im sitting here trying to get the BDSensor installed, but im running into the same issue or at least a similar issue as this.

I followed the guide, both the one here but also the video guide Chris made and both ends out in the same result where it says:
```
MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-37-g547bfbf8-dirty
MCU(s) which should be updated:
mcu: Current version v0.12.0-37-g547bfbf8
Up-to-date MCU(s):
<none>

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'mcu': Unknown command: I2C_BD_send
```
When i recover the Klipper repo, and it doesn't matter if its Hard or Soft its gonna say:
```
Unknown pin chip name 'probe'

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
```

And as i expected, when i do the recover it removes the installed files, basically uninstalling it, and if i manually place the files after its gonna return to the repo being Dirty! I also tried the repo update with no luck either, any idea what im doing wrong here?

zarpyj, 2023-12-22:
Hello Mark, hope you are fine.
I think I have found a workaround about this issue we have lot of us with this error Unknown pin chip name 'probe'.
Instead of checking in Fluidd or Mainsail for updates, seems to me that is better to check for this updates in Kiauh and apply the updates directly from Kiauh.
I let you check this approch and keep us in touch; and if it is working, please do not forget to complete the installation doc of the BDSensor.
Best regards
Jean-Jacques