markniu/Bed_Distance_sensor #53: Unable to make it working on MKS Robin nano 1.2

Carontes, 2023-12-11:
On my printer I have a 3D Touch V3.2 Pro by Geeetech and it works. Now I wanted to switch to the BD Sensor, so I just switched the 3D touch with the bd sensor, since they have the same wire connections, but it doesn't work. Then I thought it could have been something in my wire connection so I used the wires that come with the sensor and connected them directly to the motherboard but still no success. I used this code in Marlin 
`#define  I2C_BD_SDA_PIN  PA11
#define  I2C_BD_SCL_PIN    PA8`
according to [this](https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/blob/master/hardware/MKS%20Robin%20Nano%20V1.2_003/MKS%20Robin%20Nano%20V1.2_003%20PIN.pdf) page. Is there something I'm doing wrong or missing? Thank you

Carontes, 2023-12-12:
I tried it but when it's time to home the Z axis the printer fails and on the screen there's this message `BDsensor connect Err!: PRINTER HALTED Please Reset`
It's the same message as the one it happered with the "old" setting when I used the PA8

markniu, 2023-12-13:
oh, sorry, do not use PB2 to replace the PA8 but PA11.
#define  I2C_BD_SDA_PIN   PB2
#define  I2C_BD_SCL_PIN    PA8

markniu, 2023-12-13:
oh, sorry, not PA8 but PA11.
it should be like this:
#define  I2C_BD_SDA_PIN   PB2
#define  I2C_BD_SCL_PIN    PA8

markniu, 2023-12-13:
could you share your confiure files or send to me?

Carontes, 2023-12-13:
Where can I send it?

markniu, 2023-12-14:
niujl123@sina.com, you can upload to google drive and send me the link.

Carontes, 2023-12-15:
Did you receive it?

Carontes, 2023-12-17:
I tried disabling BD_SENSOR_PROBE_NO_STOP and it works, but of course I loose the fast way of leveling the bed (but it's always faster than doing it with a bl touch). So there must be something weird in that line, I don't know if it can helps but I also use a dual Z stepper system so I can use the auto-alignment feature in Marlin

markniu, 2024-01-31:
Here is one temp solution. 
https://github.com/MarlinFirmware/Marlin/issues/26738#issuecomment-1916999940
 replace the
`if (inInc > 0 ? (pos >= cmp) : (pos <= cmp)) break;`
with
`if (inInc > 0 ? ((pos+1) >= cmp) : (pos <= (cmp+1))) break;`

in the file `Marlin/src/gcode/bedlevel/abl/G29.cpp`

> I tried disabling BD_SENSOR_PROBE_NO_STOP and it works, but of course I loose the fast way of leveling the bed (but it's always faster than doing it with a bl touch).

closed 2024-03-20 by markniu