markniu/Bed_Distance_sensor #54: no_prob_stop not working

poisson21, 2023-12-13:
I have a functioning setup : BDsensor connected to a HermitCrab Can version with a combo BTT py /BTT octopus v1.1.
It work well and i can Z home and map the bed correctly with stop at each control point.

If i try to activate no_stop_prob :
it will do a first mapping without pause
Then restart one from the end of the first one backward
and then at the end stop with an error 'BDProbePointsHelper' object has no attribute 'mcu_probe' .
And so the mapping is not considered at all.

markniu, 2023-12-13:
could you attatch the klippy.log file?

poisson21, 2023-12-13:
here it is.
[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/13661721/klippy.log)

markniu, 2023-12-13:
did you use the beta version? I just found this bug is in the beta version and I just fixed

poisson21, 2023-12-13:
yes i use the beta version, i will update.

closed 2023-12-15 by poisson21