markniu/Bed_Distance_sensor #58: Pin '0' is not a valid pin name on mcu

wingman399, 2023-12-15:
So I run two boards for all my needs. It's a left over creality board that gets only my extruder, so that's my secondary mcu. Everything else is on an SKR 3. I have everything installed and in my config but it wont boot up. I checked that the BD.py  is in my files and it is but I'm lost on the pin issue I'm having. Can anyone point me in the right direction to get this up and going?
[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/13679922/klippy.log)
[moonraker.log](https://github.com/markniu/Bed_Distance_sensor/files/13679878/moonraker.log)

https://pastebin.com/HwLZkuv6 This is my printer.cfg.

wingman399, 2023-12-15:
And I did have a working system with my inductive probe prior to this. I'm trying the beta version as well.

Reubertt, 2023-12-15:
In the beta version, the pins must be connected to the main MCU as stated in the beta version readme.

> note: this real time leveling works only with the BDsensor and Z motors are connected in the same MCU.

If you are using pin aliases, check  #57

wingman399, 2023-12-15:
Right. I've got them on my main board but I can't even get initialized at this point and I have no idea where is wrong

markniu, 2023-12-16:
you can fixed by restart the firmware or update the BDsensor.py which is caused by adding the new feature extra endstop.

wingman399, 2023-12-16:
when I do the firware restart it keeps giving me the pin error. I'll try updating the py file and see if that helps

wingman399, 2023-12-17:
It didn't help. I tried taking out the aliases, all of them, and it didn't help. I've double checked the install guide again and everything matches up. Could sensorless homing make it not work right?? I do use it on my x axis.

markniu, 2023-12-17:
you can check again the file is updated as following shown  `nano ~/klipper/klippy/extras/BDsensor.py `
[image]

also needs to restart klipper process from the menu not in the main page, the best way is repower the printer.

closed 2023-12-26 by markniu