markniu/Bed_Distance_sensor #60: Mellow Fly SHT36 V2 STM F072

Psych0h3ad, 2023-12-20:
[image]
[image]

Hi, I'm connecting Fly SHT36 V2 but seems not recognized.
S-1 returns blank and S-2 return as an error.

Reubertt, 2023-12-20:
Probably the PA1 pin you are trying to use is not available for i2c connection, try using PB1.

Psych0h3ad, 2023-12-20:
> Probably the PA1 pin you are trying to use is not available for i2c connection, try using PB1.

Hi, I already tried PB1 as well :)

markniu, 2023-12-21:
then you can check if the led is on or off after letting a metal under the coil.  
and if you have multimeter, you can also check if there is a short circuit between the 5 pins on the sensor.