markniu/Bed_Distance_sensor #61: M102 S-1 Internal error on command:"M102"

wilde96, 2023-12-20:
Hi every time I use the M102 S-1 or M119 Command klipper shuts down I'm using a CR10S Pro V2 swapping from a BLtouch

 
[klippy (2).log](https://github.com/markniu/Bed_Distance_sensor/files/13729273/klippy.2.log)

markniu, 2023-12-21:
seems there is a problem of data communication.  
`ValueError: invalid literal for int() with base 10: b'\x80'`
do you have the schematic of that mainboard?  
you can try to increase the delay value in the section BDsensor from 20 to 40.

wilde96, 2023-12-21:
Hi thanks for the response, I have tried increasing the delay to 40 with no luck, also I don’t have a schematic for the board

wilde96, 2023-12-21:
I have found this 
[image]

KiwiChrist, 2023-12-22:
Yesterday the probe was working fine, but this morning I turn printer on and I have a similar issue with Klipper shutting down using M102 S-5 after calibrating. 

M102 S-5 initial responds with 1015 and then reports the sensor is too close to the bed however the probe is over 10mm from the bed.

[image]

I have adjusted the delay to 40 as suggested with same result.

markniu, 2023-12-23:
you can try to use the pins in the EXP1 or EXP2

wilde96, 2023-12-23:
Hi so what pins would I use?

markniu, 2023-12-24:
> Hi so what pins would I use?
any unused gpio pins , for example:
[image]

joshwbrick, 2024-01-23:
I am running into this same issue.

```

joshwbrick, 2024-01-23:
I am running into this same issue. I am replacing the BLTouch sensor. I am using the wiring that is already in place for the BLTouch sensor for BDSensor. I used the same pin mappings as my BLTouch config had, yet I get this error:

```
Internal error on command:"M102"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/josh/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/josh/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/josh/klipper/klippy/gcode.py", line 229, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/josh/klipper/klippy/gcode.py", line 211, in _process_commands
    handler(gcmd)
  File "/home/josh/klipper/klippy/extras/BDsensor.py", line 887, in cmd_M102
    self.process_M102(gcmd)
  File "/home/josh/klipper/klippy/extras/BDsensor.py", line 1109, in process_M102
    self.BD_version(gcmd)
  File "/home/josh/klipper/klippy/extras/BDsensor.py", line 917, in BD_version
    intd=int(pr['response'])
ValueError: invalid literal for int() with base 10: b'\x80\x00\xa8!'
```

My Config

```
[BDsensor] 
sda_pin: ^PD2
scl_pin: PB5
delay: 20 # 20us per pulse, this value should be >=18 but must be below 30
...
```

markniu, 2024-01-23:
Which motherboard is the BDsensor connected to?

> I am running into this same issue. I am replacing the BLTouch sensor. I am using the wiring that is already in place for the BLTouch sensor for BDSensor. I used the same pin mappings as my BLTouch [...]

joshwbrick, 2024-01-23:
It is a creality cr-10s pro v2. Will see if I can find more specifics on the MB.

joshwbrick, 2024-01-31:
My mainboard is v2.4

joshwbrick, 2024-02-01:
I've tried connecting SDA of BDSensor to PD2 (Pin 19) (aka Z_MAX), to PH3 which is Pin 6 from your highlight of EXP1 in the screenshot above. GND from BDSensor is connected to the EXP1 ground pin. 5V, and the other GND are connected up next to D11 (PB5).

I always get this same error. What can I try next?

markniu, 2024-02-02:
Could you try this
```
[BDsensor] 
sda_pin: PH3
scl_pin: PB5
delay: 25
```
and is it the same error? `ValueError: invalid literal for int() with base 10: b'\x80\x00\xa8!'`

> I've tried connecting SDA of BDSensor to PD2 (Pin 19) (aka Z_MAX), to PH3 which is Pin 6 from your highlight of EXP1 in the screenshot above. GND from BDSensor is connected to the EXP1 ground pin. 5V, and the other GND are connected up next to D11 (PB5) to which CLK is connected.
> 
> I always get this same error. What can I try next?

joshwbrick, 2024-02-02:
Yes same error. Here is my complete log file (config is output at the top)

https://gist.github.com/joshwbrick/67de437e3a61b8038721e992e47eee44

javiverdu, 2024-02-07:
Hi, i have the same problem with m102 s-1
[image]
my mainboard is CR-10S Pro V2 and i use the pins from the bltouch.

I also noticed while flashing klipper this warning:
src/BD_sensor.c: In function ‘command_I2C_BD_send’:
src/BD_sensor.c:441:19: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     int addr=atoi((char *)args[2]);

could this be the issue?

javiverdu, 2024-02-20:
can you use the gpio pins from the raspberry pi to connect the sensor?

javiverdu, 2024-02-20:
Confirmed: the code of the BdSensor does not work with Atmega boards, i flashed a creality v1.1.3 and found the same warning while flashing. 

i also tried flashing the SKR mini E3 v3 and only got:
lto-wrapper: warning: using serial compilation of 3 LTRANS jobs

the m102 s-1 command does work here
[image]

both commands have been tried without the BdSensor conected, so it is not a pinout problem.

markniu, 2024-02-21:
it works with the gpio pins from the raspberry pi https://github.com/markniu/Bed_Distance_sensor/issues/92
> can you use the gpio pins from the raspberry pi to connect the sensor?

javiverdu, 2024-02-21:
Any ideas to try to fix it by fiddling with the code?
src/BD_sensor.c: In function ‘command_I2C_BD_send’:
src/BD_sensor.c:441:19: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
int addr=atoi((char *)args[2]);

joshwbrick, 2024-02-21:
I hooked everything up to my rpi. Now when I run `M102 S-1` there is a pause and then the firmware crashes. In the klippy log there is not much:

```
Received 633.376414: b'{"id": 547780774784, "method": "gcode/script", "params": {"script": "M102 S-1"}}'
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[0.0, 0.0, 0.0, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.016666666666666666 extrude_factor=1.0 speed=25.0
Reactor garbage collection: (633.380998826, 0.0, 0.0)
MCU 'mcu' shutdown: Command request
clocksync state: mcu_freq=16000000 last_clock=1019712001 clock_est=(611.838 615989265 16000204.645) min_half_rtt=0.000245 min_rtt_time=588.824 time_avg=611.838(414.059) clock_avg=615989265.285(6625024093.597) pred_variance=22445522.903
Dumping serial stats: bytes_write=1339 bytes_read=10875 bytes_retransmit=9 bytes_invalid=6 send_seq=174 receive_seq=174 retransmit_seq=2 srtt=0.008 rttvar=0.004 rto=0.025 ready_bytes=0 upcoming_bytes=0
Dumping send queue 100 messages
....
```

javiverdu, 2024-02-22:
when you hooked everything to the rpi, did you flash klipper with the sensor files ? if so, did you try to flash the cr10board without the bdsensor files?

kitchung, 2024-02-23:
> Confirmed: the code of the BdSensor does not work with Atmega boards, i flashed a creality v1.1.3 and found the same warning while flashing.
>
> i also tried flashing the SKR mini E3 v3 and only got: [...]

I also get the same error when compiling for atmega2560, an 8-bit microprocessor.

markniu, 2024-02-27:
how about disconnect the BDsensor and send M102 S-1? to see if it still crashes.

kitchung, 2024-02-27:
> how about disconnect the BDsensor and send M102 S-1? to see if it still crashes.

MCU crashes even when the BDsensor is disconnected.

javiverdu, 2024-02-27:
Does it work with marlin? if not i would suggest changing the README.md in the hardware section:

> 

_"Supported on all mainboards except atmega (work in progress)"_

markniu, 2024-02-27:
I have tested and it works on the atmega2560 board at early with marlin firmware.
here this problem should be the data protocol error from the mcu to the pi.
> Does it work with marlin? if not i would suggest changing the README.md in the hardware section:
> 
> _"Supported on all mainboards except atmega (work in progress)"_

luco85, 2024-02-28:
Hi, i tried to install to my printer with ramps 1.4, but i got the error. There is any chance to use with this board or need to use the gpio pin of rpi?

kitchung, 2024-03-02:
> I have tested and it works on the atmega2560 board at early with marlin firmware. here this problem should be the data protocol error from the mcu to the pi.

Will the ability to use BD_sensor as a direct pin drop-in replacement for BLTouch for 8-bit MCUs such as atmega2560?

markniu, 2024-03-23:
I have tested on the atmega 8bit board, just found and fixed this protocol error, I will test more in other board and then update into the github.  here is the source code that needed to update into the folder `Bed_Distance_sensor`.  
[Uploading klipper.zip…]()

p4block, 2024-03-31:
At first glance, the error seems to be gone on my testbench printerless board with your new zip. Will test later on a real printer.

markniu, 2024-04-01:
> At first glance, the error seems to be gone on my testbench printerless board with your new zip. Will test later on a real printer.
Thanks

leftouterjoins, 2024-07-24:
FWIW, I've been able to get past this error using the latest version of the repo and I am using a BDsensor M (I messed up and fried my original BDsensor)