markniu/Bed_Distance_sensor #62: Klipper shuts down with M102 S-5 command

KiwiChrist, 2023-12-22:
Yesterday the probe was working fine, but this morning I turn printer on and I wanted to calibrate the probe.
It did calibrate with no problem, but as soon as I checked for the calibration results with M102 S-5 Klipper shuts down.

M102 S-5 initial responds with 1015 and then reports the sensor is too close to the bed however the probe is over 10mm from the bed.

[image]

I have adjusted the delay to 40 as suggested with same result.

Checking M119
at 10mm "open"
at 0.5mm "open"

Replaced the BDsensor with a new spare one and it worked out of the box. M119 results "open" and "triggered"

markniu, 2023-12-23:
This error means that it detected the first point in calibration data is too large(1015) by command M102 S-5,
it happens when the distance of coil to bed is below the 0.4mm or higher than 4mm.
you can check it by keeping the coil to bed about 1mm and calibrate it again

KiwiChrist, 2023-12-23:
> This error means that it detected the first point in calibration data is too large(1015) by command M102 S-5, it happens in calibration when the distance of coil to bed is below the 0.4mm or higher than 4mm you can check it by keeping the coil to bed about 1mm and calibrate it again

Thanks for the reply.
I will check the setup again, but I know I set the distance of the coil to the bed at 1.0mm as I use a feeler gauge with the nozzle touching the bed. As I said I replaced the BDsensor and it just worked without adjusting the height bracket.  Would the adjustment bracket screw above the coil make any difference, maybe use a nylon screw instead?