markniu/Bed_Distance_sensor #64: Hitting the bed and "Endstop z still triggered after retrect" error on klipper

yardi83, 2023-12-26:
Hi!

So i managed to get BD Sensor installed, wired and installed on Klipper following your instructions step by step.

Everything seems to be fine, Calibration seems fine with the first value beeing 230 and the second beeing 257.

```
> 21:18:29 
$ M102 S-5
21:18:29 
230
21:18:29 
257
21:18:29 
289
21:18:29 
321
21:18:29 
356
21:18:29 
391
21:18:29 
426
21:18:29 
460
21:18:29 
499
21:18:29 
517
21:18:29 
562
21:18:29 
610
21:18:29 
642
21:18:29 
681
21:18:29 
719
21:18:29 
743
21:18:29 
763
21:18:29 
814
21:18:29 
881
21:18:29 
920
21:18:29 
958
21:18:29 
980
21:18:29 
980
21:18:29 
980
21:18:29 
1015
21:18:29 
1015
21:18:29 
1015
21:18:30 
1015
21:18:30 
1015
21:18:30 
1015
21:18:30 
1015
21:18:30 
1015
21:18:30 
1015
21:18:30 
1015
21:18:30 
1015
21:18:30 
1015
21:18:30 
1015
21:18:31 
1015
21:18:31 
1015
21:18:31 
1015
21:18:59 
$ M84
21:19:36 
$ M119
21:19:36 
x:open y:open z:TRIGGERED
```

As you can see Z Endstop is triggering correctly and reports open when the bed is a few cm away from the Nozzle.

The Issue i`m having is that the nozzle is hitting the bed ever so slightly during homing, the bed moves dowm a few mm and klipper is giving me a "Endstop z still triggered after retract" error.
So homing correctly is still not possible.

Looked through printer.cfg but couldn`t figure out where the issue could be.

> [BDsensor] 
sda_pin: PE13 # Don't use aliases for the board pins 
scl_pin: PC5 # -clk pin example of connecting to main board Creality V4.2.7
#scl_pin:MKS_THR:gpio20 # example of connecting to CAN module like MKS THR42 
#sda_pin:MKS_THR:gpio11 # example of connecting to CAN module like MKS THR42 
delay: 20 # 20us per pulse, this value should be >=20 but must be below 50
z_offset: 0 # this `z_offset` must be set to 0. 
#z_adjust: 0.0 # z axis adjustment, replace the z_offset function. within -0.3 to 0.3mm
x_offset: 38
y_offset: -1
#no_stop_probe: # enable this for fast probe, the toolhead will not stop at the probe point.
position_endstop: 0.8 # the Z axis will stop at this position (mm) while homing z, recommend value is 0.4~1.0
speed: 0.8 # this speed only works for the z tilt and PROBE_ACCURACY command.

[safe_z_home]
home_xy_position = 162,162
speed = 150
z_hop = 10
z_hop_speed = 8

[force_move]
enable_force_move: true

[bed_mesh]
speed: 150
horizontal_move_z: 2
mesh_min: 38,5
mesh_max: 320,320
probe_count: 6,6
mesh_pps: 2,2
fade_start: 1
fade_end: 10
algorithm: bicubic
bicubic_tension:0.2

[z_tilt]
speed: 150
horizontal_move_z: 2
retries: 10
retry_tolerance: 0.0040
z_positions:
  396,186
  -71,186
points:
  285,186
  0,186

Any ideas on this?

I Have to mention that i`m on a corexy machine and using a steel pei sheet with a magnet sticker on a glass bed

markniu, 2023-12-27:
What's the homing speed of Z axis,  how about slow down more the homing speed of Z axis?

markniu, 2023-12-27:
also set the `horizontal_move_z` in the `bed_mesh `section form 2 to 1

rolohaun, 2023-12-27:
I am having this same issue also, my number are:

[stepper_z]
step_pin: PD7
dir_pin: !PD6
enable_pin: !PF10
rotation_distance: 40
gear_ratio: 60:20
microsteps: 16
endstop_pin: probe:z_virtual_endstop
homing_speed: 8
second_homing_speed: 0.8
position_max: 270

[BDsensor] 
sda_pin: PB2 # Don't use aliases for the board pins 
scl_pin: PB1 # # example of connecting to main board Creality V4.2.7
delay: 20 # 20us per pulse, this value should be >=20 but must be below 50
z_offset:0 # this `z_offset` must be set to 0. 
z_adjust:0.0 # z axis adjustment, replace the z_offset function. within -0.3 to 0.3mm
x_offset: 0
y_offset: 0
no_stop_probe: # enable this for fast probe, the toolhead will not stop at the probe point.
position_endstop: 0.8 # the Z axis will stop at this position (mm) while homing z, recommend value is 0.4~1.0
speed:0.8 # this speed only works for the z tilt and PROBE_ACCURACY command.

[safe_z_home]
home_xy_position:125,105
#speed:100
z_hop:10

[bed_mesh]
horizontal_move_z:1 # 0.7~1.0mm is recommended
algorithm: bicubic  # this should be bicubic if the bed mesh points count is > 6*6
mesh_min: 35, 6
mesh_max: 240, 198
probe_count: 5, 3

markniu, 2023-12-27:
Someone fixed it by re-install and re-flash the mcu, you can try that.

yardi83, 2023-12-27:
So i changed some things that you mentioned @markniu e.g. home speed to 5 (with 6 i got same error on homing again)
Also horizontal_move_z from 2 to 1 which was a mistake when i posted my config.

Also try to lower the z_hop in safe:z_home section @rolohaun Don`t know if this is directly related to this issue but it semms to help in imy case. And as a rook mk1 owner, building a rook 2020 right know i`m glad that you are playing around with BD Sensor aswell :)

Homing, z_tilt and bed mesh in general are working now but i get very inconsistent first layer when printing with pla (60°C bed).
First time the nozzle was way to close to the bed leaving some marks on my PEI sheet. Second and third prints where way to close to the bed aswell but don`t damaged my bed at least

How can i adjust that? My understand is that BD Sensor should care for first layer itself?

Also i can`t use TMC Autotune any longer since i installed the patched klipper version for bd sensor. I know, totally another topic but can you confirm at least that this isn`t working because of the "dirty" klipper install?

https://github.com/andrewmcgr/klipper_tmc_autotune

markniu, 2023-12-27:
> Homing, z_tilt , bed mesh in general and KAMP are working now but i get very inconsistent first layer when printing with pla (60°C bed). First time the nozzle was way to close to the bed leaving some marks on my PEI sheet. Second and third prints where way to close to the bed aswell but don`t damaged my bed at least
> 
> How can i adjust that? My understand is that BD Sensor should care for first layer itself?

You can adjust the offset by changing the value `z_adjust `in BDsensor section but the recommend way is adjust the nozzle at the zero position and calibration again.

yardi83, 2023-12-27:
When (and why?) exactly do i have to calibrate again?

markniu, 2023-12-27:
> When (and why?) exactly do i have to calibrate again?

For example if the motor is unlocked before calibration the zero position maybe changed while doing calibration, because of the movement inside the step motor from unlock to lock.
and you can also use "G28" and "G1 Z0" to check if the zero position is as expected and after calibration.

markniu, 2023-12-27:
I don't think of that. the install of BDsensor is only copy the source code file BDsensor.py and BD_sensor.c into the klipper folder, and only if there is a senction `BDsensor in the printer.cfg file that will run that source code, and also only in doing probing process. 

> Also i can`t use TMC Autotune any longer since i installed the patched klipper version for bd sensor. I know, totally another topic but can you confirm at least that this isn`t working because of the "dirty" klipper install?
> 
> https://github.com/andrewmcgr/klipper_tmc_autotune

yardi83, 2023-12-27:
> > When (and why?) exactly do i have to calibrate again?
> 
> For example if the motor is unlocked before calibration the zero position maybe changed while doing calibration, because of the movement inside the step motor from unlock to lock. and you can also use "G28" and "G1 Z0" to check if the zero position is as expected after calibration.

Okay, i got that. So after G28 i did G1 Z0 and the nozzle is hitting the bed slightly. I did M102 S-6 and these are the results

> 11:38:14 
$ M102 S-6
11:38:14 
// Calibrating, don't power off the printer
11:38:15 
// Please Wait...
11:38:44 
// Calibrate Finished!
11:38:44 
// You can send command BDSENSOR_READ_CALIBRATION to check the calibration data
11:38:54 
$ M102 S-5
11:38:54 
224
11:38:54 
225
11:38:54 
228
11:38:54 
236
11:38:54 
262
11:38:54 
293
11:38:54 
328
11:38:54 
362
11:38:54 
397
11:38:54 
431
11:38:54 
468
11:38:54 
505
11:38:55 
541
11:38:55 
577
11:38:55 
612
11:38:55 
632
11:38:55 
669
11:38:55 
700
11:38:55 
721
11:38:55 
786
11:38:55 
799
11:38:55 
880
11:38:55 
889
11:38:55 
949
11:38:55 
973
11:38:55 
973
11:38:55 
973
11:38:55 
1015
11:38:55 
1015
11:38:55 
1015
11:38:55 
1015
11:38:55 
1015
11:38:55 
1015
11:38:55 
1015
11:38:56 
1015
11:38:56 
1015
11:38:56 
1015
11:38:56 
1015
11:38:56 
1015
11:38:56 
1015

​
Second data is only one point more than the first one, so not good right? :(
I´m hesitant to start a print with that result because my build plate is already slightly damaged.

What potentially could be the issue here? Steel Plate too thin? Position of the probe overall?

markniu, 2023-12-27:
seems that the nozzle is still touch the bed after it move up 0.3mm. the sensor will use the first point as the zero position so that will cause problem.you can let nozzle a little higher about 0.1mm and calibrate again until it get good look data and the zero position is as expected.

yardi83, 2023-12-27:
okay with Z at 0,10mm i still got bad results from calibration but with 0,20mm i got that

> 12:17:22 
452
12:17:22 
467
12:17:22 
525
12:17:22 
564
12:17:22 
602
12:17:22 
639
12:17:22 
652
12:17:22 
715
12:17:22 
753
12:17:22 
789
12:17:22 
797
12:17:22 
849
12:17:23 
887
12:17:23 
925
12:17:23 
966
12:17:23 
966
12:17:23 
966
12:17:23 
1015
12:17:23 
1015
12:17:23 
1015
12:17:23 
1015
12:17:23 
1015
12:17:24 
1015
12:17:24 
1015
12:17:24 
1015
12:17:24 
1015
12:17:24 
1015
12:17:24 
1015
12:17:24 
1015
12:17:24 
1015
12:17:24 
1015
12:17:24 
1015
12:17:25 
1015

When i do G1 Z0 now what should it look like? I still don´`t get the part where the zero position is after calibration. Nozzle touching the bed?

yardi83, 2023-12-27:
Started a print and the nozzle is digging into the bed when the print is starting

I`m done for now...

markniu, 2023-12-27:
if it still too close to the bed, then use the z_adjust that is used for replaceing the z_offset

yardi83, 2023-12-27:
Can i still do babystepping over the web ui or klipperscreen?

markniu, 2023-12-27:
> Can i still do babystepping over the web ui or klipperscreen?
It can only be settled by manual now, but you can use the web ui to do babystepping and find what's the value that should be set

rolohaun, 2023-12-27:
> Someone fixed it by re-install and re-flash the mcu, you can try that. also try to slow down the home speed from 8 to 4 in section stepper_z

changing from speed 8 to 4 fixed it for me

yardi83, 2023-12-27:
With adjusting Z Offset values back and forth in printer.cfg i got one good print with a nice first layer but the next one wirhout changing anything and same GCode was way to close to the Bed. So not consistent at all for me. 
Don't know If more testing is worth it since i still don't know what my issue is.

markniu, 2023-12-28:
> With adjusting Z Offset values back and forth in printer.cfg i got one good print with a nice first layer but the next one wirhout changing anything and same GCode was way to close to the Bed. So [...]
Could you check the bed mesh data and map displayed on the mainsail or fluidd is consistent?

markniu, 2023-12-28:
would you tell me the hardware version? or the version after the M102 S-1

LAZER555, 2023-12-28:
I have the same problem, my version is V1.1 pandapi3d.com

yardi83, 2023-12-28:
> Could you check the bed mesh data and map displayed on the mainsail or fluidd is consistent?

Unfortunately not right now. I`m in the process of switching my toolhead on that printer what will take me 1 or 2 days. But i will replay to that for sure. I still wanna get that sensor running relaible, even with tzhe new toolhead.
I did not take any screenshots from my bed mesh but it wasn`t looking unusual as far as i renember

> would you tell me the hardware version? or the version after the M102 S-1

V1.1 pandapi3d.com for me aswell. Bought it at melow store over an aliexpress a couple of weeks before

Squirrelf, 2023-12-28:
Same issue, also have a v1.1
Sometime would home fine, cold bed, but nozzle pressing on the bed lightly, even with recalibration with nozzle 0.2mm above bed, same result. Light turns on well above the bed at least 1-2mm above.
Other random time ( as in testing homing right after the previous one ) with no change whatsoever would do a hard crash and keep digging into the bed until I hit emergency stop.

i know there is a v1.3 out, so maybe that would solves those issues?

neshami, 2023-12-28:
Same issue  I had with v1.0 one year ago and reverted to cr touch.

Sendt fra Outlook for Android<https://aka.ms/AAb9ysg>

rolohaun, 2023-12-28:
ok update for me, I have version 1.1 and I send my z_homing to 2 and I seem to have reliable first layers, Quad gantry level sometimes does not work but it seems ok. this sensor does seem to work, I wish the distance from the bed was greater it must be so close to the nozzle height. very easy to hit a printed part

markniu, 2023-12-29:
I am sorry for that bad experience and thank you for feedback this problem and support. the hardest thing for us is that we still do not know the actual reason since that we have not repeated this problem, and at the same time get some good feedback from some users, maybe it is the problem of the quality from the factory. now we have the new hardware version v1.3 since last month. 
anyone who met this problem can send email to pandapi or mellow webstore and request refund or replacement.

markniu, 2023-12-29:
> I have the same problem, my version is V1.1 pandapi3d.com
> 
> What i found out of my observations:
> 
> 1. You should calibrate only when hot (bed and maybe nozzle too).
> 2. You must home hot too, otherwise it crashes into bed.
> 3. KAMP is now working, which is great.
> 4. I made calibration with 0.1 from the bed and the numbers were good, but it's still scary to home the printer.
  you can try to increase the `position_endstop: 0.8`.  for example from 0.8 to 1.2

markniu, 2023-12-29:
> I have the same problem, my version is V1.1 pandapi3d.com
> 
> What i found out of my observations:
> 
> 1. You should calibrate only when hot (bed and maybe nozzle too).
> 2. You must home hot too, otherwise it crashes into bed.
> 3. KAMP is now working, which is great.
> 4. I made calibration with 0.1 from the bed and the numbers were good, but it's still scary to home the printer.
> 5. When i do endstop check in fluidd in the tune menu, z triggers before probe (there's a slight delay).
> 
> And one suggetion @markniu can you please make the window on z_adjust to more than 0.3 -0.3 to maybe 0.5 -0.5 ? Because it's not enough.

Thanks for this good information.   I will consider increasing a little more about the z_adjust.

LAZER555, 2023-12-29:
> > 4. I made calibration with 0.1 from the bed and the numbers were good, but it's still scary to home the printer.
> 
> you can try to increase the `position_endstop: 0.8`. for example from 0.8 to 1.2

There's a cap on 1.0 in there. I tried 1.2 and it gave me an error that i'm tryint to print lower than the bed. Maybe you can increase it too?

markniu, 2024-01-02:
> Okay, it crashes into bed 1 time out of 5. I asked mellow store for a replacement, and they sent it. One question, why can't it stop when homing, when the diode first lights up?

could you send me message from the facebook or email niujl123@sina.com ?  then I get more information and find the problem.

yardi83, 2024-01-02:
I hope u will give us an update here once you find a solution or find out the problem @markniu 
Still testing my new Setup with Stealhburner and a modified Mount for BD Sensor but right now after first calibration it homed fine 3 or 4 times.

After first calibration i did another one with your suggested height values (Z @0,1 and 0,2mm) and the second value was again only 1 or 2 numbers higher than the first one. But i didn`t checked if the zero possition is as expected.

Maybe it would be good to share which Type of printbeds the Users who have issues like this are using?
Because as i said i`m on a glas bed with just a magnet sticker and a PEI steel sheet. Still not 100% sure if this alone could be the reason for my issues. If so, a replacement from mellow wouldn`t do it for me.

markniu, 2024-01-03:
I have optimized the code of homing process that changed to the reading bdsensor data in timer task instead of unstabled main task.

this benefit higher home z speed and can void most of the crash to the bed,because of the stable sample timer and slow down the reading process.

hope that can fix this bed crash problem for you

LAZER555, 2024-01-03:
Can confirm, now it homes even on a cold bed. Thank you. Now the only thing left to fix in the dirty repo:P

yardi83, 2024-01-10:
Unfortunately the update didn't help much in my case. 
With some workarounds like putting a 0,10mm feeler gauge between the nozzle and bed the and heating the bed before calibration the calibration is doing fine and it homes correct most of the time. 
But when it comes to printing my first layers are still very unreliable and basicly vary every print. I know my bed is a little bit of a bowl and i can see that when printing some first layer test prints. Seems like the sensor isnt compensating the uneven areas on my bed.
I contacted mellow store on aliexpress and after some back and forth they replied with this 
[image]

Can you please help me to get a replacement so i can at least check if i got an issue with the sensor itself or if it is just not compatible with my printer somehow.

Thank you in advance

markniu, 2024-01-10:
the mellow also send your message to me early today, and I have recommended they send a new one to you. 

> Can you please help me to get a replacement so i can at least check if i got an issue with the sensor itself or if it is just not compatible with my printer somehow.
> 
> Thank you in advance

LAZER555, 2024-01-10:
Well after working well for a week, it crashed again, when i wasn't looking:P My replacement will arrive shortly so i'll test it:P

markniu, 2024-01-10:
You can check the heightmap of the mesh bed level is stable and the max data range of the data is right. 
We also find that the mechanical problem can also cause this problem. for comparison you can switch back to normal probe to check again.

>Seems like the sensor isnt compensating the uneven areas on my bed.

markniu, 2024-01-10:
what's the value of the z home speed and the second home speed?

> Well after working well for a week, it crashed again, when i wasn't looking:P My replacement will arrive shortly so i'll test it:P Nevermid that, it was my fault haha

LAZER555, 2024-01-10:
> what's the value of the z home speed and the second home speed?
> 
> > Well after working well for a week, it crashed again, when i wasn't looking:P My replacement will arrive shortly so i'll test it:P Nevermid that, it was my fault haha

i have homing_speed: 3
second_homing_speed: 1.5
homing_retract_speed: 3.5

But nevermind that last comment. Everything works, it was my mistake i put an object under x gantry haha