markniu/Bed_Distance_sensor #65: Internal error on command:"G28"

rolohaun, 2023-12-27:
I am using the BD_sensor on a Manta M8P plugged into the BLtouch port. I have verified the sensor works, but when I try to do a G28 I can an error.

If I remove the BDSensor section the printer homes fine

I am not using any pin aliases or anything my config is below

[BDsensor] 
sda_pin: PB2 # Don't use aliases for the board pins 
scl_pin: PB1 # example of connecting to main board Creality V4.2.7
delay: 20 # 20us per pulse, this value should be >=20 but must be below 50
z_offset:0 # this `z_offset` must be set to 0. 
z_adjust:0.0 # z axis adjustment, replace the z_offset function. within -0.3 to 0.3mm
x_offset: 0
y_offset: 0
no_stop_probe: # enable this for fast probe, the toolhead will not stop at the probe point.
position_endstop: 0.8 # the Z axis will stop at this position (mm) while homing z, recommend value is 0.4~1.0
speed:0.8 # this speed only works for the z tilt and PROBE_ACCURACY command.

markniu, 2023-12-27:
what's error? you can attach the klippy.log file should be better to find that the error.

rolohaun, 2023-12-27:
I got it, I found out a bed mesh is needed for the probe to work, just wasn't clear in the install guide

markniu, 2023-12-27:
> I got it, I found out a bed mesh is needed for the probe to work, just wasn't clear in the install guide

Thanks, I will this into the guide.

WinstonLeWolf, 2024-01-04:
> I got it, I found out a bed mesh is needed for the probe to work, just wasn't clear in the install guide

Please add this to the instructions.  I was racking my brain trying to fix the error and almost gave up.  Thank you for all the hard work.

Liquidus-tech, 2024-01-25:
Hi, let me start by saying that I'm new to Klipper. I have a self-built Corexy printer. Running G28 from the terminal I get an error. I'm sure I'm doing something wrong but I've been banging my head for too many days now. Can you help me? I attach log files
[Starting Klippy....txt](https://github.com/markniu/Bed_Distance_sensor/files/14051271/Starting.Klippy.txt)

markniu, 2024-01-25:
just add the section bed_mesh in your printer.cfg, here is a example
```
[bed_mesh]
speed: 120
horizontal_move_z: 1
mesh_min: 10,10
mesh_max: 320,295
probe_count: 10, 10
horizontal_move_z: 1
algorithm: bicubic

```

Liquidus-tech, 2024-01-26:
Great!  I thank you!  The printer finally moves.  Unfortunately I have other errors, I'll recompile it, I'll do it again and hope for the best

Chienhuycu, 2024-01-26:
Hello @markniu , Same problem with the recent Klipper Update (Release: v0.12.0-98-g5e433fff). Here is my Klippy.log for a custom VCore 3.1
[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/14068057/klippy.log)

Chienhuycu, 2024-01-27:
I found this on my log:   
File "/home/pi/klipper/klippy/extras/BDsensor.py", line 1190, in home_start
    trsync.start(print_time, self.trigger_completion, expire_timeout)
TypeError: MCU_trsync.start() missing 1 required positional argument: 'expire_timeout'

markniu, 2024-01-27:
Hi, you need to update the BDsensor.py file too, since there is an important update of klipper that change something about homing with probe.
> I found this on my log: File "/home/pi/klipper/klippy/extras/BDsensor.py", line 1190, in home_start trsync.start(print_time, self.trigger_completion, expire_timeout) TypeError: MCU_trsync.start() missing 1 required positional argument: 'expire_timeout'

Chienhuycu, 2024-01-27:
Hi @markniu, Is there a guide ?

markniu, 2024-01-27:
you can try this in the ssh
```
cd ~/Bed_Distance_sensor
git pull
```

Chienhuycu, 2024-01-27:
Thank you @markniu , I have just update with MoonRaker Update_Manager. Looking good

closed 2024-03-20 by markniu