markniu/Bed_Distance_sensor #66: Bed Mesh Ignoring Defined Probe Location?

vash2695, 2023-12-28:
I'm running an Ender 3 S1 Pro and just replaced my bltouch with a bdsensor and now the sensor goes off of the bed when attempting to build a mesh. The offsets shouldn't have changed this much and it doesn't make sense. I physically measured the X and Y offset to be sure it is set correctly and everything lines up. Identical offsets with a bltouch had no issues but this starts about 50mm off the front corner of the bed. Any idea why this is happening?

```
[BDsensor] 
sda_pin: ^PC14
scl_pin: PC13
delay: 20 # 20us per pulse, this value should be >=20 but must be below 50
z_offset:0 # this `z_offset` must be set to 0. 
z_adjust:0.0 # z axis adjustment, replace the z_offset function. within -0.3 to 0.3mm
x_offset: -30
y_offset: -40
no_stop_probe: # enable this for fast probe, the toolhead will not stop at the probe point.
position_endstop: 0.8 # the Z axis will stop at this position (mm) while homing z, recommend value is 0.4~1.0
speed:0.8 # this speed only works for the z tilt and PROBE_ACCURACY command.

[safe_z_home]
home_xy_position: 147,154
speed: 100
z_hop: 10
z_hop_speed: 5

[bed_mesh]
speed: 120
mesh_min: 7, 7
mesh_max: 218, 192
horizontal_move_z:1 # 0.7~1.0mm is recommended
algorithm: bicubic
probe_count: 10,10
fade_start: 1
fade_end: 10
fade_target: 0
mesh_pps: 2,2
bicubic_tension: 0.2
```

markniu, 2023-12-29:
you can also check the offset by command PROBE_CALIBRATE and to see if the nozzle will move to the probe sensor position after homing.

vash2695, 2023-12-29:
Tried using that command but got this:
[image]

markniu, 2023-12-29:
ignore that message on the webpage, just to see if the nozzle will move to the probe sensor position after z homing.

LAZER555, 2023-12-29:
Yeah i have the same thing 
[BDsensor] 
x_offset: -27
But it ignores it, so i had to change the mesh coordinates accordingly. Also when i print and use KAMP it ignores it too and probes left of the print place :P

vash2695, 2023-12-29:
> ignore that message on the webpage, just to see if the nozzle will move to the probe sensor position after z homing.

Sorry, I forgot to mention that the action triggered by that command, in my case, moved the extruder to the home position, then it very slowly moved up a short distance and then very slowly to the left until I stopped it.

> Yeah i have the same thing [BDsensor] x_offset: -27 But it ignores it, so i had to change the mesh coordinates accordingly. Also when i print and use KAMP it ignores it too and probes left of the print place :P

Weird, but that gives me a likely workaround, thank you! Now I'm wondering why it seems to be ignoring the setting for us and when else it could be ignoring it. Funny enough, it looks like this may be happening to some extent in the demo video from Chris's Basement: https://youtu.be/VDOYYskbxV8?si=cSnUEsmtJbzh7Bgu&t=1488

That, to me, looks like it's referencing the nozzle location instead of the sensor location, and because he's using an extruder with its sensor much closer on the Y axis it doesn't go off the bed and become an obvious issue.

vash2695, 2023-12-29:
Can confirm, changing my min and max works around the issue, but I'm not sure if this causes issues with where configuration _thinks_ the mesh is vs where the measurements are physically located on the bed.

LAZER555, 2023-12-29:
Yeah we're cutting and moving the mesh. Hopufully @markniu can fix the offset.

KiwiChrist, 2023-12-29:
I have the same problem relating the the probe offset using KAMP... refer to my post here.. https://github.com/markniu/Bed_Distance_sensor/issues/39

markniu, 2023-12-30:
Thanks! I check and test again and find that there is an option in the code that should be enabled for using XY offset.
please update the BDsensor.py , I am very sorry for that mistake. 
[image]

vash2695, 2023-12-30:
Can confirm things are working as expected post patch. Thank you!