markniu/Bed_Distance_sensor #67: Cannot home on Marlin 2.1.2.1/BTT Octopus 1.1

gabr42, 2023-12-30:
I am trying to install BDSensor but I'm only partially successful :(

What I did:
- This machine was already configured for BLTouch so I disconnected BLTouch and mounted BDSensor.
- Then I made the changes to the Marlin config as documented.
- I have also changed 

`#define Z_STOP_PIN                 Z_DIAG_PIN `

to 

`#define Z_STOP_PIN                 Z_MIN_PROBE_PIN`

as without that machine wouldn't home at all.

Configuration (incl. pins file) is attached.

Sensor seems to be calibrating; after M102 S-6 & M102 S-5 I get:

```
Send: M102 S-5
Recv: ok P15 B3
Recv: Calibrate data:0,469, check:1
Recv: Calibrate data:1,467, check:1
Recv: Calibrate data:2,464, check:1
Recv: Calibrate data:3,459, check:1
Recv: Calibrate data:4,467, check:1
Recv: Calibrate data:5,496, check:1
Recv: Calibrate data:6,525, check:1
Recv: Calibrate data:7,554, check:1
Recv: Calibrate data:8,582, check:1
Recv: Calibrate data:9,610, check:1
Recv: Calibrate data:10,636, check:1
Recv: Calibrate data:11,664, check:1
Recv: Calibrate data:12,689, check:1
Recv: Calibrate data:13,713, check:1
Recv: Calibrate data:14,738, check:1
Recv: Calibrate data:15,760, check:1
Recv: Calibrate data:16,781, check:1
Recv: Calibrate data:17,803, check:1
Recv: Calibrate data:18,822, check:1
Recv: Calibrate data:19,841, check:1
Recv: Calibrate data:20,860, check:1
Recv: Calibrate data:21,877, check:1
Recv: Calibrate data:22,893, check:1
Recv: Calibrate data:23,910, check:1
Recv: Calibrate data:24,925, check:1
Recv: Calibrate data:25,939, check:1
Recv: Calibrate data:26,952, check:1
Recv: Calibrate data:27,954, check:1
Recv: Calibrate data:28,954, check:1
Recv: Calibrate data:29,1015, check:1
Recv: Calibrate data:30,1015, check:1
Recv: Calibrate data:31,1015, check:1
Recv: Calibrate data:32,1015, check:1
Recv: Calibrate data:33,1015, check:1
Recv: Calibrate data:34,1015, check:1
Recv: Calibrate data:35,1015, check:1
Recv: Calibrate data:36,1015, check:1
Recv: Calibrate data:37,1015, check:1
Recv: Calibrate data:38,1015, check:1
Recv: Calibrate data:39,1015, check:1
```

(A bit weird as measured data firstly drops and then starts to raise, but maybe this is normal?)

M102 S-1 and  M102 S-2 however don't work (they don't return anything).

```
Send: M102 S-1
Recv: ok P15 B3
Send: M102 S-2
Recv: ok P15 B3
```

**My problem**:  When I home with G28, the process is sometimes successful, but sometimes the nozzle just crashes onto the bed without stopping (and Z stepper makes brrrrrr noise until I reset the machine).

I have tried reducing Z_PROBLE_FEEDRATE_FAST by 2, but that didn't make any change.

`#define Z_PROBE_FEEDRATE_FAST (2*60)`

Any suggestions about the source of this problem?

[Config.zip](https://github.com/markniu/Bed_Distance_sensor/files/13798599/Config.zip)

markniu, 2024-01-02:
the data of the calibration is not normal,  and seems that the nozzle is still on the bed until the z axis move up 0.5mm.

markniu, 2024-01-02:
if the there is no version string after M102 S-1, that means there is a communication problem and check the connection.

gabr42, 2024-01-02:
Connection is definitely OK since I'm reusing the BLTouch cabling and I did not unplug the Octopus side at all.

gabr42, 2024-01-02:
Also tried using PD5/PD6 (no change). Then I changed to the cable that came with the BD Sensor and I connected something incorrectly and burned out the sensor. So I'm giving up on this experiment and returning back to the BLTouch.

closed 2024-01-02 by gabr42