markniu/Bed_Distance_sensor #68: Ebb36 installation

joejoe317, 2024-01-01:
Hello I just purchased this and should receive in the next few weeks for a new printer build. I did order a monster 8 by mks, and was going to use an ebb 36 with USB. 

Do I only patch the ebb 36? Or do I need to patch both the mainboard and toolhead board?

cpp0xc0ffeeee, 2024-01-02:
I'm using ebb42 and fly gemini v3 with canbus bridge mode, bdsensor is connected to ebb42 and I only update bdsensor firmware to ebb42, gemini still using regular klipper firmware.

joejoe317, 2024-01-02:
> I'm using ebb42 and fly gemini v3 with canbus bridge mode, bdsensor is connected to ebb42 and I only update bdsensor firmware to ebb42, gemini still using regular klipper firmware.

Thanks for the info. Were you able to use the standard bltouch port on it?

cpp0xc0ffeeee, 2024-01-02:
> > I'm using ebb42 and fly gemini v3 with canbus bridge mode, bdsensor is connected to ebb42 and I only update bdsensor firmware to ebb42, gemini still using regular klipper firmware.
> 
> Thanks for the info. Were you able to use the standard bltouch port on it?

Yes, my bdsensor is connected to bltouch port on ebb42.

drupi79, 2025-04-13:
also using a EBB42 and when I am configuring the sensor in my printer.cfg it shows it open even thought the sensor is showing the trigger LED lit. showing my pin setup below. do I need a ^ at the sda_pin?

[BDsensor] 
# Don't use aliases for the board pins
sda_pin: EBBCan:PB8 
scl_pin: EBBCan:PB9