markniu/Bed_Distance_sensor #69: !! Bed Distance Sensor data error:10.24

haqbany, 2024-01-02:
Hi, I tried to install this sensor, but it is not working, as you guys said! please help 

this is what I get :
02:32:10 
$ G28
02:32:19 
// Real time leveling height:0.000000
02:32:20 
!! Bed Distance Sensor data error:10.24
02:55:17 
// Real time leveling height:0.000000
02:56:01 
// current z:0.100000
02:56:03 
// current z:0.200000
02:56:03 
// current z:0.300000
02:56:04 
// current z:0.400000
02:56:05 
// current z:0.500000
02:56:06 
// current z:0.600000
02:56:07 
// current z:0.700000
02:56:07 
// current z:0.800000
02:56:09 
// current z:0.900000
02:56:09 
// current z:1.000000
02:56:10 
// current z:1.100000
02:56:11 
// current z:1.200000
02:56:12 
// current z:1.300000
02:56:12 
// current z:1.400000
02:56:13 
// current z:1.500000
02:56:14 
// current z:1.600000
02:56:15 
// current z:1.700000
02:56:16 
// current z:1.800000
02:56:16 
// current z:1.900000
02:56:17 
// current z:2.000000
02:56:18 
// current z:2.100000
02:56:19 
// current z:2.200000
02:56:19 
// current z:2.300000
02:56:21 
// current z:2.400000
02:56:21 
// current z:2.500000
02:56:22 
// current z:2.600000
02:56:23 
// current z:2.700000
02:56:24 
// current z:2.800000
02:56:24 
// current z:2.900000
02:56:25 
// current z:3.000000
02:56:26 
// current z:3.100000
02:56:27 
// current z:3.200000
02:56:28 
// current z:3.300000
02:56:28 
// current z:3.400000
02:56:30 
// current z:3.500000
02:56:30 
// current z:3.600000
02:56:31 
// current z:3.700000
02:56:32 
// current z:3.800000
02:56:33 
// current z:3.900000
02:56:33 
// current z:4.000000
02:56:34 
// Calibrate Finished!
02:56:34 
// You can send command BDSENSOR_READ_CALIBRATION to check the calibration data
02:56:50 
$ G28
02:57:02 
// Klipper state: Shutdown
02:57:02 
!! Homing failed due to printer shutdown
02:57:09 
$ FIRMWARE_RESTART
02:57:09 
// Klipper state: Disconnect
02:57:14 
$ G28
02:57:24 
// Real time leveling height:0.000000
02:57:24 
!! Bed Distance Sensor data error:10.24
02:57:39 
$ G28 Y
02:57:44 
$ G28 X
02:57:49 
$ G28
02:57:53 
// Real time leveling height:0.000000
02:57:53 
!! Bed Distance Sensor data error:10.24
02:58:11 
$ M102 S-1
02:58:11 
   
02:58:24 
$ M102 S-2
02:58:24 
$ M102 S-2
02:58:24 
BDsensor: Connection Error

[klippy-2.log](https://github.com/markniu/Bed_Distance_sensor/files/13806366/klippy-2.log)

markniu, 2024-01-02:
the Connection Error means the MCU cannot communicate with the BDsensor.
Need to check the hardware connection and if the pins of your mainboard can be used as GPIO. which mainboard do you use?

haqbany, 2024-01-02:
Hi thank you for replying, I'm using BIGTREETECH Manta E3EZ V1.0with CB1 Sent from my iPhone

markniu, 2024-01-02:
Just check that board, and the probe pin cannot be used as GPIO, because of the capacitor that will filter the signal.
[image]
[image]

you can use the pin in EXP1

kodipa, 2024-01-30:
i receive the same error "Bed Distance Sensor data error:10.24"...... someone can help me ? thanks