markniu/Bed_Distance_sensor #71: Auto leveling (G29) with marlin

ibrahimhakkicandan, 2024-01-02:
When I try to auto-level with G29, it scans the 1st  line except last point and then freezes Marlin. I have to cycle the power to get the printer back. Also no error is reported. I have ender 3 s1 pro with the f4 chip and the firmware is https://github.com/ThomasToka/MarlinFirmware/releases/tag/MARLIN-E3S1PROFORK-BYTT-v023

markniu, 2024-01-03:
not sure what cause that, but you can try without #define BD_SENSOR_PROBE_NO_STOP first.

ibrahimhakkicandan, 2024-01-03:
Nope, it didnt work

markniu, 2024-01-04:
more details about without BD_SENSOR_PROBE_NO_STOP, did the toolhead move to the right point? or freezed like it with  BD_SENSOR_PROBE_NO_STOP

ibrahimhakkicandan, 2024-01-04:
almost same behaviour except this time toolhead tried 4 times, at 4. time it stucked. nothing changed. i cant leveling from machine, i have to use pronterface to send gcode commands. when i send g29 first it directly start somehow when it reaches the edge then doing auto home after that it start leveling and stucks at the edge with BD_SENSOR_PROBE_NO_STOP defined. if not defined same sceniario but this time tries 4 times after auto home then stucks.

markniu, 2024-01-31:
Here is one temp solution. 
https://github.com/MarlinFirmware/Marlin/issues/26738#issuecomment-1916999940

ibrahimhakkicandan, 2024-02-03:
tried this and same behaviour. but tried another approuch that you mentioned in the other post. almost same behaviour it scanned first line then did auto home in second try this time scanned two lines then did auto home, third time scanned and freezed but not halted i can control the lcd and here is the pronterface output since cannot bed level from lcd

echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
//action:notification BD:1.04mm
//action:notification BD:1.04mm
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
//action:notification BD:1.01mm
//action:notification BD:1.02mm
//action:notification BD:0.97mm
//action:notification BD:0.90mm
//action:notification BD:0.96mm
//action:notification BD:1.00mm
echo:busy: processing
//action:notification BD:1.04mm
//action:notification BD:1.03mm
//action:notification BD:1.04mm
//action:notification BD:1.01mm
//action:notification BD:1.03mm
echo:busy: processing
//action:notification BD:1.04mm
//action:notification BD:1.04mm
//action:notification BD:1.05mm
//action:notification BD:1.03mm
//action:notification BD:1.04mm
//action:notification BD:1.03mm
//action:notification BD:1.04mm
//action:notification BD:1.00mm
//action:notification BD:1.04mm
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
 some times it reads the distance but generally busy processing.
 toolhead stucked at the second line's last point probe