markniu/Bed_Distance_sensor #72: Error: Rescheduled timer in the past

NexGen-3D-Printing, 2024-01-03:
After updating the Bed Sensor Software, recompiling the new firmware for the control board and flashing it, I now cant complete homing.

Below is the error:

MCU 'tool_board' shutdown: Rescheduled timer in the past
This generally occurs when the micro-controller has been
requested to step at a rate higher than it is capable of
obtaining.

I have re adjusted the sensor to 2mm and run the M102 S-6 command several times, but I now have this error everytime the machine tries to move the Z Access during homing.

markniu, 2024-01-03:
how about change the value of delay in section BDsensor from 40 to 20?

NexGen-3D-Printing, 2024-01-03:
Yep that fixed it, strange, I was using those settings before the update.

markniu, 2024-01-03:
the new update of the code uses the timer to process the home and can handle higher home speed quicker. so if the mcu use much time that will cause this "Rescheduled timer" error.

NexGen-3D-Printing, 2024-01-03:
Did you update your Wiki? I did give it a quick read through, I didnt see anything mentioned there.

markniu, 2024-01-03:
in order to safer and make sure there is no other problem, I will wait one more day to update the wiki.

NexGen-3D-Printing, 2024-01-03:
Yes its best too, as I just found another bug: **Move out of range: 7.000 -13.000 0.600 [0.000]**

It appears that all offsets have been inverted, I seen in the notes you made changes tot he Z offset direction, but it looks like the Y axis as well, the sensor is mounted directly behind the nozzle, 14mm back, I was running a 14mm Y offset, but I just had to set it to -14mm.

markniu, 2024-01-03:
I recently fixed one bug about the XY offset that is not used before, https://github.com/markniu/Bed_Distance_sensor/issues/66#issuecomment-1872481588

markniu, 2024-01-03:
> Yes its best too, as I just found another bug: **Move out of range: 7.000 -13.000 0.600 [0.000]**
>
> It appears that all offsets have been inverted, I seen in the notes you made changes tot he Z [...]

I recently fixed one bug about the XY offset that is not used before, https://github.com/markniu/Bed_Distance_sensor/issues/66#issuecomment-1872481588

NexGen-3D-Printing, 2024-01-03:
Okay so I need to ask a silly question, is the folling Y offset, the the BD sensor from the nozzle, or the nozzle from the sensor?

[BDsensor] 
scl_pin: tool_board:gpio20
sda_pin: tool_board:gpio11
delay: 20
z_offset: 0
z_adjust: 0
x_offset: 0
y_offset: -14
no_stop_probe: true

markniu, 2024-01-03:
> Okay so I need to ask a silly question, is the folling Y offset, the the BD sensor from the nozzle, or the nozzle from the sensor?
> 
> [BDsensor] scl_pin: tool_board:gpio20 sda_pin: tool_board:gpio11 delay: 20 z_offset: 0 z_adjust: 0 x_offset: 0 y_offset: -14 no_stop_probe: true

It is the same as normal probe sensor, the nozzle is the 0 position.

NexGen-3D-Printing, 2024-01-03:
Okay then the Y-offset should be 14mm not -14mm, so something is not right, the sensor is mounted 14mm behind the nozzle, 0/home is at the front of the machine.

This only changed since the update.

closed 2024-01-16 by markniu