markniu/Bed_Distance_sensor #74: Communication timeout during homing probe Z

shlo-mix, 2024-01-11:
Hi 

So this wired bug is only happens while doing a probing operations with BDsensor not mesh or calibrations.

HW:
Voron trident 350mm
BTT M8P Version: v0.12.0-61-gb50d6669
BTT EBB36 1.2, Version: v0.12.0-61-gb50d6669, CAN
M102 s-1 V1.1 pandapi3d.com

Calibration data look good:
977
964
...
290
266
245
also M102 S-2 Sensor measures distance look accurate

Homing:
Homing of X, Y is successful ,but when It get to Z I get "Communication timeout during homing Z" :

8:58 PM
Z axis triggered at 0.000 mm
8:58 PM
warning:triggered at 0mm, Please slow down the homing z speed and the position_endstop in BDsensor >=0.5
8:58 PM
triggered at 10.150 mm !
8:58 PM
Communication timeout during homing z

After this failure:
- klipper on mcu EBBCan is busy MCU load going up from 2% to 20% then back to 2%
- "bytes retransmit state" going up from zero to about 170 
- the communication with probe is very slow (M102 sS-5 take  more then a minute).
- if I try do to second home it fails instantly on x axis: "Communication timeout during homing x"

So its look like a bug that overload MCU preventing communication.
[klippy (14).log](https://github.com/markniu/Bed_Distance_sensor/files/13907488/klippy.14.log)

if I disable the virtual endstop and use the mechanical z endstop the homing is successful. 
also I can do Mesh via Hightmap tab and it's really nice to see results ,communication is still good and fast, (bytes_retransmit =0, bytes_invalid = 0.)

But then when I do Z_TILT I get:
9:38 PM
Z axis triggered at 0.900 mm
9:38 PM
"Communication timeout during homing probe",

"bytes_retransmit" is about 150, comunication is very slow with EBBcan/BD sensor indicating communication issues

markniu, 2024-01-12:
what's the CAN bus speed while make menuconfig? if it below 1000000 you can try to increase it.

shlo-mix, 2024-01-13:
the Can is 1000000, no errors:

biqu@BTT-CB1:~$ ip -details -statistics link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 128
    link/can  promiscuity 0 minmtu 0 maxmtu 0
    can state ERROR-ACTIVE restart-ms 0
          bitrate 1000000 sample-point 0.750
          tq 62 prop-seg 5 phase-seg1 6 phase-seg2 4 sjw 1
          gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp-inc 1
          clock 48000000
          re-started bus-errors arbit-lost error-warn error-pass bus-off
          0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
    RX: bytes  packets  errors  dropped missed  mcast
    55436      7622     0       0       0       0
    TX: bytes  packets  errors  dropped carrier collsns
    17862      2836     0       0       0       0

shlo-mix, 2024-01-13:
Found a issue in the code that was part of the problem under BDsensor.py:     
  # Probe position
  try:
      if ((self.mcu_probe.bd_sensor is not None) and 
              (( "BED_MESH_CALIBRATE" in gcmd.get_command()) or  
              ("QUAD_GANTRY_LEVEL" in gcmd.get_command()))):

was missing Z_TILT_ADJUST, I added it:

  # Probe position
  try:
      if ((self.mcu_probe.bd_sensor is not None) and 
              (( "BED_MESH_CALIBRATE" in gcmd.get_command()) or 
              ( "Z_TILT_ADJUST" in gcmd.get_command()) or  
              ("QUAD_GANTRY_LEVEL" in gcmd.get_command()))):

This enabled me to do a  Z_TILT_ADJUST "scan" :) then I needed to set:
[z_tilt]
horizontal_move_z: 2 
and now Z_TILT_ADJUST working very good.
...

Its not a complete solution, still must use a mechanical Z for homing, but the main important features for me are working.

shlo-mix, 2024-01-13:
BTW, for I2C reading, I so that the code use while loop for waiting, is it passible to change it?  maybe use timer interrupt.

markniu, 2024-01-13:
Not sure, I need to think of that to see if it is possible. 

> BTW, for I2C reading, I see that the code use while loop for waiting, is it passible to change it? maybe use timer interrupt.

shlo-mix, 2024-01-13:
Thank you for your great work!

sarpel, 2024-01-14:
```
19:15:46
1015
19:15:46
1015
19:15:46
956
19:15:46
956
19:15:46
956
19:15:45
956
19:15:45
949
19:15:45
937
19:15:45
925
19:15:45
909
19:15:45
894
19:15:45
879
19:15:45
864
19:15:45
845
19:15:45
827
19:15:44
810
19:15:44
791
19:15:44
768
19:15:44
748
19:15:44
724
19:15:44
701
19:15:44
678
19:15:44
652
19:15:44
625
19:15:44
599
19:15:44
571
19:15:44
544
19:15:44
513
19:15:44
485
19:15:44
454
19:15:44
423
19:15:44
392
19:15:44
362
19:15:44
330
19:15:44
300
19:15:44
269
19:15:44
240
19:15:44
211
19:15:44
182
19:15:44
157
19:15:44
BDSENSOR_READ_CALIBRATION

19:18:44
Z axis triggered at 0.000 mm
19:18:44
warning:triggered at 0mm, Please slow down the homing z speed and the position_endstop in BDsensor >=0.5
19:18:43
triggered at 10.150 mm !
19:18:42
Communication timeout during homing z
19:18:31
G28

```

```
[BDsensor] 
#sda_pin: PB1
#scl_pin: PB0 
scl_pin: EBBCan: PB9
sda_pin: EBBCan: PB8
delay: 20 ###Tried Variations 20-30###
z_offset:0
z_adjust: 0
x_offset: 0
y_offset: 0 ###Tried Variations, My actual was 20. even tho unrelated###
no_stop_probe: true ###Tried Variations true or false or just '#' in front###
position_endstop: 1.0 ###Tried Variations 0.5-2.0### 
speed:0.8 ###Tried Variations 0.3-1###

[safe_z_home]
home_xy_position: 150, 115
speed: 200.0
z_hop: 1 ###Tried variations from 1 to 5###
z_hop_speed: 15.0 ###Tried Variations 1-15 even not related i guess###
#move_to_previous: False ###Tried Variations or just '#' in front, unrelated###

[bed_mesh]
speed: 200
horizontal_move_z: 1 ###Tried variations from 1 to 5###
mesh_min: 40, 40
mesh_max: 260,260
#probe_count: 25,25 ###Tried variations, but meshing is working the problem is homing###
algorithm: bicubic
fade_start: 0.6 ###Tried Variations or just '#' in front, unrelated###
fade_end: 10.0 ###Tried Variations or just '#' in front, unrelated###
bicubic_tension: 0.2 ###Tried Variations or just '#' in front, unrelated###
zero_reference_position: 150,115 ###Tried Variations or just '#' in front, unrelated###

[quad_gantry_level]
gantry_corners:
	-60,-10
	360,370
points:
    70,40
	70,210
	240,210
	240,40
speed: 200
horizontal_move_z: 1 ###Tried variations from 1 to 5###
retries: 5
retry_tolerance: 0.05
max_adjust: 10

[force_move]
enable_force_move: True

[stepper_z]
step_pin: PD14
dir_pin: !PD13
enable_pin: !PD15
rotation_distance: 40
gear_ratio: 80:16
microsteps: 16
endstop_pin: probe:z_virtual_endstop
position_max: 260
position_min: -10 ###Tried variations -10 to +1###
#position_endstop: 0
#homing_speed: 10 #pcb klicky
homing_speed: 3 ###Tried variations 3 to 0.5###
#second_homing_speed: 2 #pcb klicky
second_homing_speed: 0.8 ###Tried variations 0.8 to 0.3###
homing_retract_speed: 3.5 ###Tried variations also disabled in some variations###
#homing_retract_dist: 3 #klicky
homing_retract_dist: 1 ###Tried variations also disabled in some variations###
```

Then it all comes to this. Somehow ebb36 cant keep up with incoming data. My CAN speed is 1,000,000 and also CGL and bed meshing works which i assume involves more data than homing. 

```
Stats 3245.8: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000005 mcu_task_stddev=0.000004 bytes_write=5418 bytes_read=10476 bytes_retransmit=9 bytes_invalid=0 send_seq=411 receive_seq=411 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=180002956 EBBCan: mcu_awake=0.002 mcu_task_avg=0.000014 mcu_task_stddev=0.000011 bytes_write=1512 bytes_read=6577 bytes_retransmit=0 bytes_invalid=0 send_seq=162 receive_seq=162 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=63999352 adj=63998161  EBBCan: temp=26.1 heater_bed: target=0 temp=19.5 pwm=0.000 Raspberry_Pi: temp=34.6 Spider: temp=26.1 chamber: temp=21.0 sysload=0.91 cputime=71.510 memavail=1321232 print_time=25.474 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.9 pwm=0.000
Communication timeout during homing z
Stats 3246.9: gcodein=0  mcu: mcu_awake=0.004 mcu_task_avg=0.000012 mcu_task_stddev=0.000012 bytes_write=7238 bytes_read=12359 bytes_retransmit=9 bytes_invalid=0 send_seq=516 receive_seq=516 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=16 freq=180002920 EBBCan: mcu_awake=0.002 mcu_task_avg=0.000014 mcu_task_stddev=0.000011 bytes_write=1679 bytes_read=6936 bytes_retransmit=81 bytes_invalid=0 send_seq=177 receive_seq=176 retransmit_seq=175 srtt=0.004 rttvar=0.004 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=63999320 adj=63998134  EBBCan: temp=26.3 heater_bed: target=0 temp=19.5 pwm=0.000 Raspberry_Pi: temp=34.6 Spider: temp=26.0 chamber: temp=21.0 sysload=0.91 cputime=71.618 memavail=1316412 print_time=26.065 buffer_time=0.043 print_stall=0 extruder: target=0 temp=18.9 pwm=0.000
triggered at 10.150 mm !
warning:triggered at 0mm, Please slow down the homing z speed and the position_endstop in BDsensor >=0.5 
Z axis triggered at 0.000 mm 
Stats 3248.6: gcodein=0  mcu: mcu_awake=0.004 mcu_task_avg=0.000012 mcu_task_stddev=0.000012 bytes_write=7373 bytes_read=12736 bytes_retransmit=9 bytes_invalid=0 send_seq=526 receive_seq=526 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=180002920 EBBCan: mcu_awake=0.002 mcu_task_avg=0.000014 mcu_task_stddev=0.000011 bytes_write=1719 bytes_read=7148 bytes_retransmit=122 bytes_invalid=0 send_seq=181 receive_seq=181 retransmit_seq=181 srtt=0.003 rttvar=0.003 rto=0.050 ready_bytes=0 upcoming_bytes=0 freq=63999320 adj=63998075  EBBCan: temp=26.3 heater_bed: target=0 temp=19.4 pwm=0.000 Raspberry_Pi: temp=34.1 Spider: temp=25.9 chamber: temp=20.9 sysload=1.00 cputime=71.639 memavail=1313976 print_time=26.065 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.9 pwm=0.000
```

Hardware: Voron 2.4
Rpi4
Fystec Spider 2.3
EBB36 
U2C
2x Mellow 5160 HV @48V
4x Fysetc 2209 @ 24V
Goliath + vz-hex cnc + moons 8t

sarpel, 2024-01-14:
additional info,

I tried beta version and recompile reflash ebb. then it worked. then after firmware restart it started to fail again. on the same session at putty, i reflashed already existing klipper firmware(with bdsensor), and it worked again. 

I am sure after a firmware restart it will fail again. 

Does this kind of an error ring a bell?

markniu, 2024-01-15:
how about decrease the delay value from 20 to 19 18 17 ? but need to make sure the data from bdsensor is right e.g M102 S-1

markniu, 2024-01-15:
BTW, I also have met this timeout problem with EBB36 in random even without BDsensor before. I do believe this problem is caused by the busy of process by the application software in the PI or in the MCU

shlo-mix, 2024-01-16:
What is the reason that EBB36  keeping up in mesh but not while probing? 
Is it possible to relax the BD sensor probing CPU usage?  (or use it the same way  as in "Mesh")

davidsiaw, 2024-01-16:
I am also receiving communication timeout from homing probe. Even if I adjust the trsync_timeout to a higher value it still times out.

sarpel, 2024-01-16:
> how about decrease the delay value from 20 to 19 18 17 ? but need to make sure the data from bdsensor is right e.g M102 S-1

Decreasing values makes no difference. They are v1.2 panda something even at 10 delay. But whatever i tried, i couldnt make it to home besides new upload over katapult session i mentioned.

I looked for this error on klipper discourse forums. There were some type of consensus about it being caused by non 32bit OS, timer timeout to 0.05 and some managed to change SOC  and worked. The last guy says he owns a printer farm and pi4s give a lot of timeout errors, so he changed them with some kind of intel soc or laptop and them all got fixed. He also mentioned after unplugging usbs (cam, adxl etc) also fixes the problem. Conclusion was pi4 cant handle usb buses or something. 

Anyway, i tried usb unplugging, minimalizing components and timeout thing. Them all are impossible for this sensor to not timeout. Because just in the first command, ebb36 yields. Sudden 170ish retransmits. Clock sync disrupts waaayy above threshold of 0.05 or similar timeout registers. 

I mean probe works bed meshing and qgl but it just cant home even once without timeouting. 

Changed back to klicky pcb. 

@markniu isnt there some way to slow homeing procedure with somekind of delay between those messages? I am sure more and more will come for this problem. I just dont know what differs us from already working people setup wise

sarpel, 2024-01-17:
@markniu there is a klipper update about multi mcu homing and time sync issues and timeouts. Can you check if it is tailored for us. the exact moment we need. Did koconnor saw us somehow?

markniu, 2024-01-17:
Thank you let me know that update, I think that can fix the timeout problem with EBB36, 
I also sync that code into the BDsensor.py from klipper. please update the BDsensor.py and klipper then try again.

> @markniu there is a klipper update about multi mcu homing and time sync issues and timeouts. Can you check if it is tailored for us. the exact moment we need. Did koconnor saw us somehow?

markniu, 2024-01-17:
For homing with BDsensor, the `TRSYNC_TIMEOUT = 0.025` is setted in BDsensor.py not mcu.py.

> I looked for this error on klipper discourse forums. There were some type of consensus about it being caused by non 32bit OS, timer timeout to 0.05 and some managed to change SOC and worked.

shlo-mix, 2024-01-17:
Setting the delay to 13 helped a bit (12 didn't work)  but most of the time I get the same timeouts.

For Sarpel and me "mesh" is working, it there a possibility to have temporary workaround for probing by changing the "probe reading" to "mesh reading"? 
PROBE accuracy is not important  as long we do a mesh afterword

BTW
Even if I use my solution of Z mech end-stop, doing QGL/TILT must be from safe distance (>10mm), sometimes the bed is tilted by gravity, if doing meshing/Tilt from 2mm,  I may crash into bed. learned it in the hard way.

markniu, 2024-01-17:
you can try to update the BDsensor.py and the klipper, I believe that can fix the timeout problem,  no need to reflash the mcu.
if still has random timeout you can increase the `TRSYNC_TIMEOUT ` from 0.025 to 0.05 in the  BDsensor.py not mcy.py 

> Setting the delay to 13 helped a bit (12 didn't work) but most of the time I get the same timeouts.
>

sarpel, 2024-01-17:
And it works. @markniu the klipper update did it all.

shlo-mix, 2024-01-18:
- I updated the klipper on main MCU (CB1)
-  updated the BD rep and run ./install_BDsensor.sh
-  Updated the EBB36 and MANTA main board with the new updated Klipper
-  under BDsensor.py Changed the TRSYNC_TIMEOUT  from 0.025 to 0.05 

Now the probing works, ZTILT seems to probe and adjust tilt correctly , but at the end it fails:
[image]
With delay of 15 and 18

shlo-mix, 2024-01-18:
Sometimes when doing TILT  the MCU crushes and I get 

"MCU 'EBBCan' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap..."

markniu, 2024-01-22:
Seems that the speed of mcu of EBBcan is a little slow as a CAN module. could you try to increase the `sample_time =.03 `to 0.05 in BDsensor.py ?
> Sometimes when doing TILT the MCU crashes and I get red screen with following massage
> 
> "MCU 'EBBCan' shutdown: Timer too close This often indicates the host computer is overloaded. Check for other processes consuming excessive CPU time, high swap..."

shlo-mix, 2024-01-22:
Hi
I changed value to 0.05 and still get this error

shlo-mix, 2024-01-22:
Observations:
- sample_time =.01  same Red crash "MCU 'EBBCan' shutdown: ..."
- sample_time =.02  get homing timeouts
- suggest to set homing speed and Z tilit speed need to be 3.0 else you will risk crashing into bed at least with my setup.
- In order to be as far as I can from bed I set the "position_endstop: 1.99" , for accuracy I run mesh every print at 1mm distance 
- There is unharmful bug after successful  Z-Tilt,  the bed actual position is 10mm but the SW set it to zero, we need to run 
Second Z homing to bypass, this is unharmful because it is better to o Z home  after tilt anyway

shlo-mix, 2024-01-22:
new issue : 
[image]

markniu, 2024-01-23:
This is very useful. 

> 
> by increasing the txqueuelen from 128 to 1024 the problem solved with 0.03:
> 
> sudo nano /etc/network/interfaces.d/can0
> 
> allow-hotplug can0 iface can0 can static bitrate 1000000 up ifconfig $IFACE txqueuelen 1024

markniu, 2024-01-23:
seems this log file didn't upload successfully. 
> Uploading klippy (15).log…
> 
> but this is rare issue, the top one is the more frustrating (Timer too close ..)

shlo-mix, 2024-01-23:
sorry here it is:
[klippy (15).log](https://github.com/markniu/Bed_Distance_sensor/files/14021117/klippy.15.log)

shlo-mix, 2024-01-23:
> This is very helpful.
> 
> > by increasing the txqueuelen from 128 to 1024 the problem solved with 0.03:
> > sudo nano /etc/network/interfaces.d/can0
> > allow-hotplug can0
> > iface can0 can static
> > bitrate 1000000
> > up ifconfig $IFACE txqueuelen 1024

No, unfortunately the problem of "Timer too close .."  still exist, maybe it help a bit but didn't solve. this is the critical problem between the two

markniu, 2024-01-23:
Maybe this will be helpful. 
https://github.com/Dids/klipper-priority-fix

> Unfortunately the problem of "Timer too close .." still exist, maybe it help a bit but didn't solve. this is the more critical problem between the two, as it happened more frequently.

shlo-mix, 2024-01-23:
The same
Klipper reports: SHUTDOWN
MCU 'EBBCan' shutdown: Timer too close

shlo-mix, 2024-01-23:
With 100ms sample time
95% of the times it doesn't crashes

markniu, 2024-01-24:
Let the sda_pin as a switch gpio while homing that should be perfect solution without increase the sample time and slow down the home speed.  I will try to do that. but that needs to flash the firmware into the BDsenosr with external hardware like stlink or usbtouart module. 

> one way to make it work while probing, instead of I2C communication, BD need to toggle the GPIO 1 or 0 like a switch, only after that the switch toggles send a "read command" this will increase performance, reduce CPU consumption and will be safer.

shlo-mix, 2024-01-24:
And maybe we can also increase the sample time, to 5ms or more
I have the ST-Link/v2 (STM8 and STM32) will it be good? what type of MCU?

markniu, 2024-01-24:
The MCU on V1.1(hardware version) is C51 that needs a USB to UART module to flash, the MCU on V1.3 is STM32 that needs st-linkv2.
> That is great news, maybe we can also increase the samples rates - to 5ms or less. I have the ST-Link/v2 (STM8 and STM32) will it be good? what type of MCU? Also I think I have the USB to UART.

markniu, 2024-02-05:
right, I am trying to add one more feature that can support working as a strain gauge.

shlo-mix, 2024-02-05:
About the implementation, after that  do you identify that GPIO is triggers, do you stop Z travel and preform stationary reading operation to have the exact position?
This will give as better accuracy while doing fast probing, both for homing and Z-TILT.
I have RT embedded skills, let me know if I can help.

markniu, 2024-02-05:
in switch mode, it will use the gpio to trigger and then switch to I2C mode to read the exact postion.
I have updated the hex file for the V1.1

shlo-mix, 2024-02-06:
Look like I have successfully flash the 1.1 code:
[FW_Flash_log.txt](https://github.com/markniu/Bed_Distance_sensor/files/14183251/FW_Flash_log.txt)

Do I need to change something else in the Klipper config?

markniu, 2024-02-07:
No need to set other config.

you can check if it is flashed success from the serial port. there is a serial port window on the STC-ISP tool, choose the right comport and power on the BDsensor, you can see some message from the bdsensor, like the version, and the raw data of the distance.

[image]

> -I have successfully flash the 1.1 code: [FW_Flash_log.txt](https://github.com/markniu/Bed_Distance_sensor/files/14183691/FW_Flash_log.txt)
> 
> * updated the latest BDsensor and Klipper repo.
> * run install_BDsensor.sh and updated firmware on the EBB36.
> 
> I can see the led turning on when the probe is near the metal build plate. but M102 S-1 return empty line, so it's look like I don't have communication.
> 
> Do I need to set something under config?

markniu, 2024-02-07:
> No need to set other config.
> 
> you can check if it is flashed success from the serial port. there is a serial port window on the STC-ISP tool, choose the right comport and power on the BDsensor, you can see some message from the bdsensor, like the version(V1.1b), and the raw data of the distance.
> 
> ![image](https://private-user-images.githubusercontent.com/13879340/302823869-21022888-c1f4-4626-bc0b-7e84b0f6522e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcyNjk2MzMsIm5iZiI6MTcwNzI2OTMzMywicGF0aCI6Ii8xMzg3OTM0MC8zMDI4MjM4NjktMjEwMjI4ODgtYzFmNC00NjI2LWJjMGItN2U4NGIwZjY1MjJlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA3VDAxMjg1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg1ZGQ0MDk2YmM5ZmI1NTM3MzJjZjdkODJkNGU3YzM5MDk0YWQ4NTA2NDcwYjQ1Y2MyZmFmNGU1NjM5ZTQ0NzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.AUFrbPJB2bzmkgyeADzu7HtmRHroHy-W8B48O2-nC0w)
> 
> > -I have successfully flash the 1.1 code: [FW_Flash_log.txt](https://github.com/markniu/Bed_Distance_sensor/files/14183691/FW_Flash_log.txt)
> > 
> > * updated the latest BDsensor and Klipper repo.
> > * run install_BDsensor.sh and updated firmware on the EBB36.
> > 
> > I can see the led turning on when the probe is near the metal build plate. but M102 S-1 return empty line, so it's look like I don't have communication.
> > Do I need to set something under config?

shlo-mix, 2024-02-07:
When I open both files I get the file is over the validscope, and the exceeding part has been moved to the EEPROM area.

markniu, 2024-02-07:
both of the files are hex, it will transfer to bin after it is opened with download tools and you can see they are below 6Kbyte that does not exceed the flash memory.

> When I open both files I get the file is over the validscope, and the exceeding part has been moved to the EEPROM area.

shlo-mix, 2024-02-07:
Check MCU is working, flashing is working, but cannot see a communication under UART Helper.

shlo-mix, 2024-02-07:
UART TTL is 3.3 or 5V?

shlo-mix, 2024-02-07:
I tried with another new BD sensor that I have, under UART Helper tab I could see the same "pandapi3d" communication response after reset .

But soon as I flash it  with new FW or old  its stop to communicate, now I have 2 sensors that stopped working :)

markniu, 2024-02-07:
that's strange, I will download that version STC ISP and check

shlo-mix, 2024-02-07:
I found the issue, I think that the file was damaged with direct download.

I disabled the realtime scan of windows and downloaded all the rep as a ZIP (not just the file), now I can see communication

shlo-mix, 2024-02-07:
I see that you send the ADC data via UART, what is the sample rate of ADC?
It would be great to have at least 200Hz (5ms) ADC sample and GPIO update to have fast Z probe response.

markniu, 2024-02-08:
the data will be output via uart only if the data is changed.the rate is about 100~200Hz, depends on the distance,the shorter distance the rate will be higher.
> I see that you send the ADC data via UART, what is the sample rate of ADC? It would be great to have at least 200Hz (5ms) ADC sample and GPIO update to have fast Z "GPIO probe" response.

markniu, 2024-02-08:
try to decrease the value of position_endstop in BDsensor section to about 1mm, and delete the position_endstop in` [stepper_z]`

> Back to the topic unfortunately it faild to do homing: Internal Error on WebRequest: gcode/script [klippy (18).log](https://github.com/markniu/Bed_Distance_sensor/files/14199708/klippy.18.log)

shlo-mix, 2024-02-08:
I changed the position_endstop in BD sensor from 1.99 to 1mm, under stepper_z is  was already commented out.

Important to say that when Z home start its going to the wrong direction, away from tool-head, (distance from plate to BD is increasing )

markniu, 2024-02-08:
this new feature of home for the V1.1 seems need more test, I will test more.

> I changed the position_endstop in BD sensor from 1.99 to 1mm, under stepper_z is was already commented out, I got same result
> 
> Important to say that when Z home start its going to the wrong direction, away from tool-head, (distance from plate to BD is increasing )

shlo-mix, 2024-02-11:
Flashed FW V1_1BDsensor20240210.hex, with latest repo, my test results feedback:

1. The homing is working :)
2. The TILT is almost working: It's doing the calibration successfully (the TILT button is blue), but at the end it report error:
"Home z failed! the triggered z position is 10.150 mm"
"triggered at 10.150 mm !"

Also, if bed not completely leveled before,  TILT will fail (triggered at 0mm or something like that).

shlo-mix, 2024-02-12:
3. mesh sometimes fails: "Bed Distance Sensor, out of range.:10.15"
[klippy (19).log](https://github.com/markniu/Bed_Distance_sensor/files/14238611/klippy.19.log)

markniu, 2024-02-12:
you can manual move the nozzle to the failed point and checked it by M102 S-2.
BTW, please send me your email address via the email, and I would like to send you the V1.3 hardware to test.
> 3. mesh most of the time fails: "Bed Distance Sensor, out of range.:10.15"
>    [klippy (19).log](https://github.com/markniu/Bed_Distance_sensor/files/14238611/klippy.19.log)
>    before update mesh never failed.

shlo-mix, 2024-02-12:
M102 S-2 = 0.99mm
-It failed in 2 different  places, after few retries I can have successful Mesh without any magor dips or peaks, so its mostly random reading error.
- If disable "no_stop_probe" it still fail

shlo-mix, 2024-02-12:
updated to new e821238 commit and V1_1BDsensor20240212.hex
With latest code klipper report syntax error on line 1212:
[image]
I fixed this typo and now TILT ending successfully without errors.
so both issue 1 and 2 are now working very good without any issues or timeout. 
Great job!

Mesh still report "Bed Distance Sensor, out of range.:10.15", actual value M102 S-2 =0.98mm

shlo-mix, 2024-02-14:
With latest ver issue 3 with the mesh was fixed.

markniu, 2024-02-14:
if you have time you can test this new feature of collision sensing.  here is the tutorial on how to use the Collision sensing: https://github.com/markniu/Bed_Distance_sensor/wiki/Collision-sensing

[9 more replies]