markniu/Bed_Distance_sensor #75: M102 S-1 no communication

mighur, 2024-01-14:
Hello.
Just installed BD sensor in BTT SRK 1.4 Turbo.
Connections:
SDA: (Green) Servo pin 2.0
GND: (Yellow) Servo GND pin
CLK: (Orange) Probe pin 0.10
5V: (Red) NPWR servo pin (+5V)
GND: (Brown) GND servo pin (GND)

I followed all github instructions and double checked with Chris Basement video.
I commented/activated all indicated options.

At the end of pins_BTT_SKR_V1_4.h I'v added:

#define I2C_BD_SDA_PIN 2.0 // Please change to the actual number which the SDA wire is connected to your mainboard
#define I2C_BD_SCL_PIN 0.10 // Please change to the actual number which the SLK wire is connected to your mainboard
#define I2C_BD_DELAY 20 // default value is 20, should be in the range [20,50].

Compiled marlin with no errors.
Everything seems to work ok (temperatures, thermistors, movement X & Y, M119 gives expected results...)

Connected printer via terminal:
M102 S-1 outputs:

recv: bd sensor version:\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
recv: read error. check connection and delay.

BTT SKR 1.4 Turbo board is brand new. Just buyed.
Exact same behabiour with BTT SKR 1.3. Same output message.
I fact I bought SKR1.4 because I thought It was a board issue.

M102 S-6: Does de thing and z axis goes up but after that, S102 S-5 outputs:

Send: M102 S-5
Recv: ok P15 B3
Recv: Calibrate data:0,1023
Recv: Read Error.
Recv: Calibrate data:1,1023
Recv: Read Error.
Recv: Calibrate data:2,1023
Recv: Read Error.
Recv: Calibrate data:3,1023
Recv: Read Error.
Recv: Calibrate data:4,1023
Recv: Read Error.
Recv: Calibrate data:5,1023
Recv: Read Error.
Recv: Calibrate data:6,1023
Recv: Read Error.
Recv: Calibrate data:7,1023
Recv: Read Error.
Recv: Calibrate data:8,1023
Recv: Read Error.
Recv: Calibrate data:9,1023
Recv: Read Error.
Recv: Calibrate data:10,1023
Recv: Read Error.
Recv: Calibrate data:11,1023
Recv: Read Error.
Recv: Calibrate data:12,1023
Recv: Read Error.
Recv: Calibrate data:13,1023
Recv: Read Error.
Recv: Calibrate data:14,1023
Recv: Read Error.
Recv: Calibrate data:15,1023
Recv: Read Error.
Recv: Calibrate data:16,1023
Recv: Read Error.
Recv: Calibrate data:17,1023
Recv: Read Error.
Recv: Calibrate data:18,1023
Recv: Read Error.
Recv: Calibrate data:19,1023
Recv: Read Error.
Recv: Calibrate data:20,1023
Recv: Read Error.
Recv: Calibrate data:21,1023
Recv: Read Error.
Recv: Calibrate data:22,1023
Recv: Read Error.
Recv: Calibrate data:23,1023
Recv: Read Error.
Recv: Calibrate data:24,1023
Recv: Read Error.
Recv: Calibrate data:25,1023
Recv: Read Error.
Recv: Calibrate data:26,1023
Recv: Read Error.
Recv: Calibrate data:27,1023
Recv: Read Error.
Recv: Calibrate data:28,1023
Recv: Read Error.
Recv: Calibrate data:29,1023
Recv: Read Error.
Recv: Calibrate data:30,1023
Recv: Read Error.
Recv: Calibrate data:31,1023
Recv: Read Error.
Recv: Calibrate data:32,1023
Recv: Read Error.
Recv: Calibrate data:33,1023
Recv: Read Error.
Recv: Calibrate data:34,1023
Recv: Read Error.
Recv: Calibrate data:35,1023
Recv: Read Error.
Recv: Calibrate data:36,1023
Recv: Read Error.
Recv: Calibrate data:37,1023
Recv: Read Error.
Recv: Calibrate data:38,1023
Recv: Read Error.
Recv: Calibrate data:39,1023
Recv: Read Error.

If I put a bit of steel near the sensor, the white LED lights.
If I try to home Z:

Recv: echo:BDsensor connect Err!
Recv: //action:notification BeltSKeR 1.4 bugfix Lista
Recv: //action:notification BDsensor connect Err!:
Recv: Error:Printer halted. kill() called!

¿What can I do?

markniu, 2024-01-15:
the motherboard can't communicate with the bdsensor. for testing, you can use the EXP1 pins for the bdsensor, also try to increase the delay value from 20 to 40.

mighur, 2024-01-15:
Finally solved. It was a mistake with the naming of pins.
Thank you

markniu, 2024-01-16:
Glad to know.

closed 2024-01-16 by markniu

mylife4aiurr, 2024-07-16:
> Finally solved. It was a mistake with the naming of pins. Thank you

Can you please post the pin naming you used. I have same problem on skr1. 4turbo

mighur, 2024-07-17:
This is what I have for BTT_SKR_V1_4
#define  I2C_BD_SDA_PIN    P2_00   
#define  I2C_BD_SCL_PIN    P0_10   
#define  I2C_BD_DELAY  20

mylife4aiurr, 2024-07-17:
Thank you. Ive tried those exact pins named exactly how you now outlined it. I will try actually cutting and pasting what you posted here. Thank you for responding.