markniu/Bed_Distance_sensor #77: QUAD_GANTRY_LEVEL

davidsiaw, 2024-01-16:
I am having the same issue as https://github.com/markniu/Bed_Distance_sensor/issues/37  but Z_TILT adjust cannot be done because I am using a flying gantry.

[image]

The gantry can sink more than 2-3mm , so the BD sensor needs to hop. Setting the horizontal movement z is too close and will damage the bed.

markniu, 2024-01-17:
Hello, there is a update from klipper yesterday that may fix your timeout problem. please update the BDsensor.py and klipper, try again.

if you want ton increase the homing timeout `TRSYNC_TIMEOUT `please modify the `TRSYNC_TIMEOUT `in the BDsensor.py.

davidsiaw, 2024-01-17:
@markniu thank you for your reply! Can you tell me how to make QGL hop like the inductive sensor? the horizontal move is too close to the bed

markniu, 2024-01-18:
https://github.com/markniu/Bed_Distance_sensor/issues/74#issuecomment-1895468696

> @markniu thank you for your reply! Can you tell me how to make QGL hop like the inductive sensor? the horizontal move is too close to the bed

davidsiaw, 2024-01-19:
i will try it later i put my inductor probe back because i need the printer to work. Thank you 🙏

closed 2024-01-19 by davidsiaw